diederikdehaas/rtl8812AU #1: compile fails with cfg80211 enabled

MrEngman, 2015-03-22:
Compiling on Pi B 2 with gcc (Raspbian 4.8.3-13) 4.8.3 fails. Output from compile is:

```
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/3.18.9-v7+/build M=/home/pi/src1/rtl8812AU  modules
make[1]: Entering directory '/home/pi/src/linux'
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_cmd.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_security.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_debug.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_io.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_ioctl_query.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_ioctl_set.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_ieee80211.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_mlme.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_mlme_ext.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_wlan_util.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_vht.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_pwrctrl.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_rf.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_recv.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_sta_mgt.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_ap.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_xmit.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_p2p.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_tdls.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_br_ext.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_iol.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_sreset.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_btcoex.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_beamforming.o
  CC [M]  /home/pi/src1/rtl8812AU/core/rtw_odm.o
  CC [M]  /home/pi/src1/rtl8812AU/core/efuse/rtw_efuse.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/osdep_service.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/linux/os_intfs.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/linux/usb_intf.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/linux/ioctl_linux.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/linux/xmit_linux.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/linux/mlme_linux.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/linux/recv_linux.o
  CC [M]  /home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.o
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_add_key’:
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:1574:3: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
   _rtw_memcpy(param->u.crypt.seq, params->seq, params->seq_len);
   ^
In file included from /home/pi/src1/rtl8812AU/include/drv_types.h:32:0,
                 from /home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:22:
/home/pi/src1/rtl8812AU/include/osdep_service.h:259:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
 extern void _rtw_memcpy(void* dec, void* sour, u32 sz);
             ^
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:1580:3: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
   _rtw_memcpy(param->u.crypt.key, params->key, params->key_len);
   ^
In file included from /home/pi/src1/rtl8812AU/include/drv_types.h:32:0,
                 from /home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:22:
/home/pi/src1/rtl8812AU/include/osdep_service.h:259:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
 extern void _rtw_memcpy(void* dec, void* sour, u32 sz);
             ^
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_join_ibss’:
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:2934:2: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
  _rtw_memcpy(ndis_ssid.Ssid, params->ssid, params->ssid_len);
  ^
In file included from /home/pi/src1/rtl8812AU/include/drv_types.h:32:0,
                 from /home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:22:
/home/pi/src1/rtl8812AU/include/osdep_service.h:259:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
 extern void _rtw_memcpy(void* dec, void* sour, u32 sz);
             ^
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_set_pmksa’:
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:3370:2: warning: passing argument 1 of ‘_rtw_memcmp’ discards ‘const’ qualifier from pointer target type [enabled by default]
  if ( _rtw_memcmp( pmksa->bssid, strZeroMacAddress, ETH_ALEN ) == _TRUE )
  ^
[... 79 lines cut ...]
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c: At top level:
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6080:2: warning: initialization from incompatible pointer type [enabled by default]
  .scan = cfg80211_rtw_scan,
  ^
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6080:2: warning: (near initialization for ‘rtw_cfg80211_ops.get_station’) [enabled by default]
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6108:2: warning: initialization from incompatible pointer type [enabled by default]
  .del_station = cfg80211_rtw_del_station,
  ^
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6108:2: warning: (near initialization for ‘rtw_cfg80211_ops.add_station’) [enabled by default]
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6109:2: warning: initialization from incompatible pointer type [enabled by default]
  .change_station = cfg80211_rtw_change_station,
  ^
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6109:2: warning: (near initialization for ‘rtw_cfg80211_ops.del_station’) [enabled by default]
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6110:2: warning: initialization from incompatible pointer type [enabled by default]
  .dump_station = cfg80211_rtw_dump_station,
  ^
/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6110:2: warning: (near initialization for ‘rtw_cfg80211_ops.change_station’) [enabled by default]
scripts/Makefile.build:257: recipe for target '/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/home/pi/src1/rtl8812AU/os_dep/linux/ioctl_cfg80211.o] Error 1
Makefile:1381: recipe for target '_module_/home/pi/src1/rtl8812AU' failed
make[1]: *** [_module_/home/pi/src1/rtl8812AU] Error 2
make[1]: Leaving directory '/home/pi/src/linux'
Makefile:1466: recipe for target 'modules' failed
make: *** [modules] Error 2
```

MrEngman, 2015-03-23:
Hi,

Came across this post on the internet http://marc.info/?l=fedora-list&m=142282197701463&w=2 and thought you will find it useful.

A link on that page, https://github.com/pld-linux/rtl8812au, contains a number of patches. Out of curiosity I applied the linux-3.18.patch and was then able to compile your source without it failing. I then applied the warnings.patch and that reduces the number of warnings conciderably as below.

```
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/3.18.9-v7+/build M=/home/pi/wifi-src/diederikdehaas-rtl8812au  modules
make[1]: Entering directory '/home/pi/src/linux'
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_cmd.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_security.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_debug.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_io.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_mlme.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_vht.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_rf.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_recv.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_ap.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_xmit.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_p2p.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_tdls.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_br_ext.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_iol.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_sreset.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_btcoex.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_beamforming.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_odm.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/osdep_service.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.o
/home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.c:6080:2: warning: initialization from incompatible pointer type [enabled by default]
  .get_station = cfg80211_rtw_get_station,
  ^
/home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.c:6080:2: warning: (near initialization for ‘rtw_cfg80211_ops.get_station’) [enabled by default]
/home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.c:6108:2: warning: initialization from incompatible pointer type [enabled by default]
  .add_station = cfg80211_rtw_add_station,
  ^
/home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.c:6108:2: warning: (near initialization for ‘rtw_cfg80211_ops.add_station’) [enabled by default]
/home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.c:6109:2: warning: initialization from incompatible pointer type [enabled by default]
  .del_station = cfg80211_rtw_del_station,
  ^
/home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.c:6109:2: warning: (near initialization for ‘rtw_cfg80211_ops.del_station’) [enabled by default]
/home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.c:6110:2: warning: initialization from incompatible pointer type [enabled by default]
  .change_station = cfg80211_rtw_change_station,
  ^
/home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/ioctl_cfg80211.c:6110:2: warning: (near initialization for ‘rtw_cfg80211_ops.change_station’) [enabled by default]
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/wifi_regd.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/rtw_android.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/os_dep/linux/rtw_proc.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/hal/hal_intf.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/hal/hal_com.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/hal/hal_com_phycfg.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/hal/hal_phy.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/hal/hal_dm.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/hal/hal_btcoex.o
[... 28 lines cut ...]
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/hal/OUTSRC/phydm_RaInfo.o
[... cut ...]
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/platform/platform_ops.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_mp.o
  CC [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/core/rtw_mp_ioctl.o
  LD [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/8812au.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/pi/wifi-src/diederikdehaas-rtl8812au/8812au.mod.o
  LD [M]  /home/pi/wifi-src/diederikdehaas-rtl8812au/8812au.ko
make[1]: Leaving directory '/home/pi/src/linux'
```

MrEngman, 2015-03-23:
Hi,

I have been able to remove the final 4 warnings by applying 4 items from the update-cfg80211-support.patch from https://github.com/pld-linux/rtl8812au.

I have create a complee patch that I have posted to Dropbox. The link is  https://dl.dropboxusercontent.com/u/80256631/rtl8812AU.patch

All I need to do now is work out how to add the code for 8821ua devices to get my EW-7811UTC device working  :) :)

diederikdehaas, 2015-03-23:
> A link on that page, https://github.com/pld-linux/rtl8812au, contains a number of patches.

That is indeed quite a good repo :+1:  

> how to add the code for 8821ua devices to get my EW-7811UTC device working

Maybe it's just adding your device ID to `os_dep/linux/usb_intf.c`?

MrEngman, 2015-03-24:
Unfortunately it's not as simple as just adding the device id to  `os_dep/linux/usb_intf.c` 

The EW-7811UTC is an 8821au device so needs the 8821au code adding in to your current code.

diederikdehaas, 2015-03-24:
Ah ok, I thought they shared the same code base. Ah well, it was worth a shot :wink: 
It's been a while since I looked at it, but I thought that gnab's code and mine weren't all that different.

diederikdehaas, 2015-05-08:
> Unfortunately it's not as simple as just adding the device id to os_dep/linux/usb_intf.c
The EW-7811UTC is an 8821au device so needs the 8821au code adding in to your current code.

Just playing with the code atm ... and came across the following line in `Makefile`:
`CONFIG_RTL8821A = n`
Did you change that to `y` when you tried?

MrEngman, 2015-05-08:
Yes, I tried that and the compile fails. I haven't looked at the source for a while but I think if you look at gnab's code you will see a number of files either having 8821au in the name or in a directory 8821au.
e.g  in hal/OUTSRC/rtl8821a

diederikdehaas, 2015-05-08:
Bummer.
Trying to get the module to compile (and work) on `Linux rasppi-2b 3.18.0-trunk-rpi2 #1 SMP PREEMPT Debian 3.18.5-1~exp1+rpi16 (2015-03-28) armv7l GNU/Linux` right now.

Also got 2 questions to which you may know the answer:
- When I just did a `git clone`, `cd rtl8812AU` and then `make` I got this line:
`make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/3.18.0-trunk-rpi2/build M=/home/diederik/rtl8812AU  modules`
When I then changed `CONFIG_PLATFORM_I386_PC = n` and `CONFIG_PLATFORM_ARM_RPI = y` in `Makefile` I got this line:
`make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.18.0-trunk-rpi2/build M=/home/diederik/rtl8812AU  modules`
Is the former better then the latter?
- do you have any experience with getting a driver DKMS 'ready'?  
My hope is to get a bunch of not-officially supported drivers with a `-dkms` package (probably under https://github.com/debian-pi) so that when a new kernel is released the drivers would be recompiled automatically.

MrEngman, 2015-05-08:
OK, just done a new git clone and compile and I had to comment out `#define define CONFIG_IOCTL_CFG80211 1` to get the compile to complete. I didn't add any patches that can get it to compile with cfg80211 enabled.

I tend to use make clean && make (or make -j6 on a Pi 2B) which gives `make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/3.18.12-v7+/build M=/home/pi/src/rtl8812AU  modules`

Once upon a time I used to edit the make file as you have but I don't bother these days. As to which way is better I have no idea. However, the drivers I compile all generally appear to work OK, although I had problems with a mt7601 wifi module. Compiling on the Pi 2B for a Pi 2B the wifi module would cause the kernel to crash when booting, but cross-compiling on a Ubuntu 14.0.2 system and it worked fine. The same code compiled on a Pi 2B for a Pi 1 seemed to work fine. I have not been able to figure out what the problem was with the driver on the Pi 2B. I now tend to cross-compile the wifi modules I use, including the 8812au, because of the issue with the mt7601 driver, and anyway it is somewhat faster than compiling on the Pi.

As for DKMS I'm not sure I even know what it is :)  What you want to do sounds useful, though.

diederikdehaas, 2015-05-08:
DKMS: https://en.wikipedia.org/wiki/Dynamic_Kernel_Module_Support
A list of packages in Debian which have DKMS: https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=-dkms

I have on my host sytem the `nvidia-kernel-dkms` and `virtualbox-dkms` packages installed, which make sure that when a new kernel gets installed (and the headers) the kernel modules for nvidia and virtualbox are automatically compiled so they work whenever I boot into that new kernel.

MrEngman, 2015-05-08:
OK, must take a look at it.

diederikdehaas, 2015-05-09:
Wooohooo \o/

With the current code, I got it to compile completely **without warnings** and **with cfg80211 enabled** on both my main PC (amd64) and on the Raspberry Pi 2!
```
diederik@rasppi-2b:~/rtl8812AU$ make
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/3.18.0-trunk-rpi2/build M=/home/diederik/rtl8812AU  modules
make[1]: Entering directory '/usr/src/linux-headers-3.18.0-trunk-rpi2'
Makefile:10: *** mixed implicit and normal rules: deprecated syntax
  CC [M]  /home/diederik/rtl8812AU/core/rtw_cmd.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_security.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_debug.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_io.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_ioctl_query.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_ioctl_set.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_ieee80211.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_mlme.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_mlme_ext.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_wlan_util.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_vht.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_pwrctrl.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_rf.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_recv.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_sta_mgt.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_ap.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_xmit.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_p2p.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_tdls.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_br_ext.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_iol.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_sreset.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_btcoex.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_beamforming.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_odm.o
  CC [M]  /home/diederik/rtl8812AU/core/efuse/rtw_efuse.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/osdep_service.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/os_intfs.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/usb_intf.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/ioctl_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/xmit_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/mlme_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/recv_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/wifi_regd.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/rtw_android.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/rtw_proc.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_intf.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_com.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_com_phycfg.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_phy.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_dm.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_btcoex.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_hci/hal_usb.o
  CC [M]  /home/diederik/rtl8812AU/hal/led/hal_usb_led.o
  CC [M]  /home/diederik/rtl8812AU/hal/HalPwrSeqCmd.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/Hal8812PwrSeq.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/Hal8821APwrSeq.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_xmit.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_sreset.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_hal_init.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_phycfg.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_rf6052.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_dm.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_rxdesc.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_cmd.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/usb/usb_halinit.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/usb/rtl8812au_led.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/usb/rtl8812au_xmit.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/usb/rtl8812au_recv.o
[... 31 lines cut ...]
  MODPOST 1 modules
  CC      /home/diederik/rtl8812AU/8812au.mod.o
  LD [M]  /home/diederik/rtl8812AU/8812au.ko
make[1]: Leaving directory '/usr/src/linux-headers-3.18.0-trunk-rpi2'
```
```
root@rasppi-2b:/home/diederik/rtl8812AU# make install
install -p -m 644 8812au.ko  /lib/modules/3.18.0-trunk-rpi2/kernel/drivers/net/wireless/
/sbin/depmod -a 3.18.0-trunk-rpi2
```
Then I unplugged my device (`Bus 001 Device 006: ID 7392:a822 Edimax Technology Co., Ltd`), removed `/etc/udev/rules.d/70-persistent-net.rules` and plugged it back in:
```
root@rasppi-2b:/home/diederik/rtl8812AU# iwconfig
wlan0     IEEE 802.11AC  ESSID:"RTIRACE"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.18 GHz  Access Point: 50:46:5D:5E:2E:A5   
          Bit Rate:867 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=100/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

Wooohooo \o/

diederikdehaas, 2015-05-09:
> The EW-7811UTC is an 8821au device so needs the 8821au code adding in to your current code.

While there's certainly overlap with this driver, a diff between my driver and https://github.com/Braklet/rtl8811AU_rtl8821A-linux also shows quite some difference.
Since my C/C++ knowledge is **VERY** limited, it is beyond my capabilities to 'merge' the 2 code bases. 
But the Braklet repo sounds like a good starting point for your device (also check the various forks from there).

diederikdehaas, 2015-05-09:
Also compiles without errors/warnings on a Pi 1B+ ... and works :smile:

MrEngman, 2015-05-09:
Whether it compiles or not I think may depend on the kernel version being compiled for and also the version of gcc being used.

MrEngman, 2015-05-09:
Just changed Pi's I compile on, cloned your source and now it compiles without errors. Using a newer version of gcc. 

Went back to the other Pi, re-configured linux with `make mrproper && make bcm2709_defconfig && make modules_prepare && cp ../firmware/extra/Module7.symvers Module.symers` and recompiled your source and it now compiled with warnings. 

Think I may have messed up the linux source on the first Pi doing a git pull to update the source and forgot I did not re-configure it after updating the linux source.

diederikdehaas, 2015-05-09:
> Whether it compiles or not I think may depend on the kernel version being compiled for and also the version of gcc being used.

That could very well be true. 
On both RPi 1B+ and RPi 2B I tested it on are running jessie and my host system is Debian Sid/Unstable. I'll also try it on wheezy and on my server (Debian Jessie amd64) and report back.
The tests I've done were with either the normal Debian kernel or the kernel provided by raspbian.org (with appropriate `linux-header` packages), not the kernels provided by the Raspberry Pi Foundation.

diederikdehaas, 2015-05-09:
I have just installed a fresh wheezy system on a RPi 1B (with my [Net Installer](https://github.com/debian-pi/raspbian-ua-netinst)).
Then I installed the kernel headers package (`linux-headers-rpi-rpfv`), `git`, `build-essential`, `wpasupplicant`, `wireless-tools` and ... `sudo`. I then created a normal user (diederik) and added that user to the `netdev` and `sudo` groups. 
Added the following stanza to `/etc/network/interfaces`:
```
# Wireless interface
allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp
```
and copied a `wpa_supplicant.conf` from one of my other systems.

Rebooted (although not needed) and logged in as myself over SSH.
Did a `git clone https://github.com/diederikdehaas/rtl8812AU`, `cd rtl8812AU`, `make clean` and then `make`:
```
diederik@rasppi-1b:~/rtl8812AU$ make
make ARCH=armv6l CROSS_COMPILE= -C /lib/modules/3.18.0-trunk-rpi/build M=/home/diederik/rtl8812AU  modules
make[1]: Entering directory '/usr/src/linux-headers-3.18.0-trunk-rpi'
Makefile:10: *** mixed implicit and normal rules: deprecated syntax
  CC [M]  /home/diederik/rtl8812AU/core/rtw_cmd.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_security.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_debug.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_io.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_ioctl_query.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_ioctl_set.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_ieee80211.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_mlme.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_mlme_ext.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_wlan_util.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_vht.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_pwrctrl.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_rf.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_recv.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_sta_mgt.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_ap.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_xmit.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_p2p.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_tdls.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_br_ext.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_iol.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_sreset.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_btcoex.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_beamforming.o
  CC [M]  /home/diederik/rtl8812AU/core/rtw_odm.o
  CC [M]  /home/diederik/rtl8812AU/core/efuse/rtw_efuse.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/osdep_service.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/os_intfs.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/usb_intf.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/ioctl_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/xmit_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/mlme_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/recv_linux.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/wifi_regd.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/rtw_android.o
  CC [M]  /home/diederik/rtl8812AU/os_dep/linux/rtw_proc.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_intf.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_com.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_com_phycfg.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_phy.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_dm.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_btcoex.o
  CC [M]  /home/diederik/rtl8812AU/hal/hal_hci/hal_usb.o
  CC [M]  /home/diederik/rtl8812AU/hal/led/hal_usb_led.o
  CC [M]  /home/diederik/rtl8812AU/hal/HalPwrSeqCmd.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/Hal8812PwrSeq.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/Hal8821APwrSeq.o
  CC [M]  /home/diederik/rtl8812AU/hal/rtl8812a/rtl8812a_xmit.o
[... 71 lines cut ...]
Since `lsmod` didn't show `8812au` I did a `modprobe 8812au` and got the message 
`libkmod: ERROR ../libkmod/libkmod.c:554 kmod_search_moddep: could not open moddep file '/lib/modules/3.18.0-trunk-rpi/modules.dep.bin'`
So I checked that file:
```
diederik@rasppi-1b:~$ ls -l /lib/modules/3.18.0-trunk-rpi/
total 220
lrwxrwxrwx 1 root root     39 Apr  2 06:18 build -> /usr/src/linux-headers-3.18.0-trunk-rpi
drwxr-xr-x 9 root root   4096 May  9 16:08 kernel
-rw-r--r-- 1 root root      0 May  9 17:06 modules.alias
-rw-r--r-- 1 root root      0 May  9 17:06 modules.alias.bin
-rw-r--r-- 1 root root   4078 Apr  2 06:18 modules.builtin
-rw-r--r-- 1 root root      0 May  9 17:06 modules.builtin.bin
-rw-r--r-- 1 root root 148332 May  9 17:06 modules.dep
-rw-r--r-- 1 root root      0 May  9 17:06 modules.dep.bin
-rw-r--r-- 1 root root      0 May  9 17:06 modules.devname
-rw-r--r-- 1 root root  63234 Apr  2 06:18 modules.order
-rw-r--r-- 1 root root      0 May  9 17:06 modules.softdep
-rw-r--r-- 1 root root      0 May  9 17:06 modules.symbols
-rw-r--r-- 1 root root      0 May  9 17:06 modules.symbols.bin
lrwxrwxrwx 1 root root     42 Apr  2 06:18 source -> /usr/src/linux-headers-3.18.0-trunk-common
```
And saw there were a whole bunch of files with file size 0!
Earlier in this post I mentioned and emphasized `sudo` since that's a departure from what I normally do. My normal way of operating is login as `root` when I need to do root-stuff.
So now I logged in as root and did `make install` again and then the file sizes were as they should be.
Rebooted the Pi and then my wireless device did work normally.

diederikdehaas, 2015-05-18:
@MrEngman I just found out why my driver won't work on your device: 
I don't have an `hal/OUTSRC/rtl8821a` directory.
See https://github.com/gnab/rtl8812au/issues/1#issuecomment-55927477

diederikdehaas, 2015-05-19:
I tried to integrate support for 8821a in a new branch: https://github.com/diederikdehaas/rtl8812AU/tree/add-rtl8821a-support
It compiles cleanly on my machines, but since I don't have a rtl8821a device I can't test if it works or not. It would be great if you could test it.
The code for 8821a is based on https://github.com/Braklet/rtl8811AU_rtl8821A-linux/

diederikdehaas, 2015-09-21:
@MrEngman 

I have merged the `add-rtl8821a-support` branch into master and I'd really appreciate it if you could test it and tell whether it works for you now. 
If it doesn't work, I have a small change to `Makefile` which might help:
```diff
diff --git a/Makefile b/Makefile
index 3c3445f..4808bea 100755
--- a/Makefile
+++ b/Makefile
@@ -1455,10 +1455,12 @@ $(MODULE_NAME)-$(CONFIG_MP_INCLUDED)+= core/rtw_bt_mp.o
 endif
 
 obj-$(CONFIG_RTL8812AU) := $(MODULE_NAME).o
+obj-$(CONFIG_RTL8821AU) := $(MODULE_NAME).o
 
 else
 
 export CONFIG_RTL8812AU = m
+export CONFIG_RTL8821AU = m
 
 all: modules
```
That change is from https://github.com/saiarcot895/rtl8811AU_rtl8821A-linux which is a repo which is (slightly) further developed then the Braklet one.
It would be really great if this repo would also support the EW-7811UTC, but as I don't have it, I can't test it for myself.

diederikdehaas, 2015-09-21:
Found a new repo with **VERY** recent code from Realtek: https://github.com/Grawp/rtl8812au_rtl8821au \o/

MrEngman, 2015-09-21:
That's the version I'm using :)  I was sent it to me by someone who got it after talking to Edimax support I think. However, good to see it on github now.

I've tried your update and after adding the USB ID for the EW-7811UTC it compiles and runs OK. No need to edit the Makefile to compile.

FYI here's a patch with a few more USB ID's for 8812au and 8821au devices
```
--- rtl8812AU1/os_dep/linux/usb_intf.c  2015-09-10 08:54:24.153108058 +0100
+++ 8812au/os_dep/linux/usb_intf.c      2015-09-21 21:22:07.986448051 +0100
@@ -281,6 +281,7 @@
        {USB_DEVICE(USB_VENDER_ID_REALTEK, 0x881C),.driver_info = RTL8812},/* Default ID */
        /*=== Customer ID ===*/
        {USB_DEVICE(0x050D, 0x1106),.driver_info = RTL8812}, /* Belkin - sercomm */
+       {USB_DEVICE(0x050D, 0x1109),.driver_info = RTL8812}, /* Belkin F9L1109 - SerComm */
        {USB_DEVICE(0x2001, 0x330E),.driver_info = RTL8812}, /* D-Link - ALPHA */
        {USB_DEVICE(0x7392, 0xA822),.driver_info = RTL8812}, /* Edimax - Edimax */
        {USB_DEVICE(0x0DF6, 0x0074),.driver_info = RTL8812}, /* Sitecom - Edimax */
@@ -297,7 +298,11 @@
        {USB_DEVICE(0x07B8, 0x8812),.driver_info = RTL8812}, /* Abocom - Abocom */
        {USB_DEVICE(0x2001, 0x3315),.driver_info = RTL8812}, /* D-Link - Cameo */
        {USB_DEVICE(0x2001, 0x3316),.driver_info = RTL8812}, /* D-Link - Cameo */
-       {USB_DEVICE(0x2357, 0x0101),.driver_info = RTL8812}, /* TP-Link - TP-Link */
+       {USB_DEVICE(0x2357, 0x0101),.driver_info = RTL8812}, /* TP-Link - Archer T4U */
+       {USB_DEVICE(0x20f4, 0x805b),.driver_info = RTL8812}, /* TRENDnet - */
+       {USB_DEVICE(0x13B1, 0x003F),.driver_info = RTL8812}, /* Linksys WUSB6300 */
+       {USB_DEVICE(0x148F, 0x9097),.driver_info = RTL8812}, /* Amped Wireless ACA1 */
+       {USB_DEVICE(0x2357, 0x0103),.driver_info = RTL8812}, /* TP-Link - T4UH */
 #endif

 #ifdef CONFIG_RTL8821A
@@ -306,12 +311,19 @@
        {USB_DEVICE(USB_VENDER_ID_REALTEK, 0x0821),.driver_info = RTL8821},/* Default ID */
        {USB_DEVICE(USB_VENDER_ID_REALTEK, 0x8822),.driver_info = RTL8821},/* Default ID */
        {USB_DEVICE_AND_INTERFACE_INFO(USB_VENDER_ID_REALTEK, 0x0820,0xff,0xff,0xff),.driver_info = RTL8821}, /* 8821AU */
+       {USB_DEVICE_AND_INTERFACE_INFO(USB_VENDER_ID_REALTEK, 0x0823,0xff,0xff,0xff),.driver_info = RTL8821}, /* 8821AU */
        /*=== Customer ID ===*/
        {USB_DEVICE(0x7392, 0xA811),.driver_info = RTL8821}, /* Edimax - Edimax */
        {USB_DEVICE(0x04BB, 0x0953),.driver_info = RTL8821}, /* I-O DATA - Edimax */
+       {USB_DEVICE(0x0BDA, 0xA811),.driver_info = RTL8821}, /* OUTLINK - Edimax */
+       {USB_DEVICE(0x7392, 0xA812),.driver_info = RTL8821}, /* Edimax - EW-7811UTC */
+       {USB_DEVICE(0x7392, 0xA813),.driver_info = RTL8821}, /* Edimax - EW-7811UAC */
        {USB_DEVICE(0x2001, 0x3314),.driver_info = RTL8821}, /* D-Link - Cameo */
+       {USB_DEVICE(0x0846, 0x9052),.driver_info = RTL8821}, /* Netgear - A6100 */
        {USB_DEVICE(0x2001, 0x3318),.driver_info = RTL8821}, /* D-Link - Cameo */
        {USB_DEVICE(0x0E66, 0x0023),.driver_info = RTL8821}, /* HAWKING - Edimax */
+       {USB_DEVICE(0x0411, 0x0242),.driver_info = RTL8821}, /* BUFFALO - Edimax */
+       {USB_DEVICE(0x2019, 0xAB32),.driver_info = RTL8821}, /* Planex - GW-450S */
 #endif

 #ifdef CONFIG_RTL8192E

```

diederikdehaas, 2015-09-21:
Excellent, thanks :+1: 
And with that good news it's time to close this issue.

I probably won't maintain this driver much any more since I found the other one with much newer code. I'd likely 'steal' code from the other one ... and that seems like a waste of effort. I already submitted a pull request to the other repo :wink:

closed 2015-09-21 by diederikdehaas

samurdhilbk, 2017-11-14:
@MrEngman @diederikdehaas Can you please clarify how to compile the driver with cfg80211 support? I couldn't find any option in the Makefile pointing towards that.