diederikdehaas/rtl8812AU #3: compile fails

qwertzdenek, 2015-09-09:
compiling on ArchLinux x86_64 against 4.1.6 kernel with GCC 5.2.0. It looks like that function `cfg80211_vendor_event_alloc changed`.

module use this call:
`cfg80211_vendor_event_alloc(struct wiphy *wiphy, int len, int event_id, gfp_t gfp)`

but in the kernel tree is:
`cfg80211_vendor_event_alloc(struct wiphy *wiphy, struct wireless_dev *wdev, int approxlen, int event_idx, gfp_t gfp)`

```bash
  CC [M]  /usr/src/rtl8812AU-1.0/os_dep/linux/ioctl_cfg80211.o
/usr/src/rtl8812AU-1.0/os_dep/linux/ioctl_cfg80211.c:6245:22: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^
/usr/src/rtl8812AU-1.0/os_dep/linux/ioctl_cfg80211.c:6245:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
  CC [M]  /usr/src/rtl8812AU-1.0/os_dep/linux/rtw_cfgvendor.o
/usr/src/rtl8812AU-1.0/os_dep/linux/rtw_cfgvendor.c: In function ‘rtw_cfgvendor_send_async_event’:
/usr/src/rtl8812AU-1.0/os_dep/linux/rtw_cfgvendor.c:170:47: warning: passing argument 2 of ‘cfg80211_vendor_event_alloc’ makes pointer from integer without a cast [-Wint-conversion]
  skb = rtw_cfg80211_vendor_event_alloc(wiphy, len, event_id, kflags);
                                               ^
/usr/src/rtl8812AU-1.0/os_dep/linux/rtw_cfgvendor.c:143:37: note: in definition of macro ‘rtw_cfg80211_vendor_event_alloc’
  cfg80211_vendor_event_alloc(wiphy, len, event_id, gfp)
                                     ^
In file included from /usr/src/rtl8812AU-1.0/include/osdep_service_linux.h:75:0,
                 from /usr/src/rtl8812AU-1.0/include/osdep_service.h:41,
                 from /usr/src/rtl8812AU-1.0/include/drv_types.h:32,
                 from /usr/src/rtl8812AU-1.0/os_dep/linux/rtw_cfgvendor.c:21:
include/net/cfg80211.h:4371:1: note: expected ‘struct wireless_dev *’ but argument is of type ‘int’
 cfg80211_vendor_event_alloc(struct wiphy *wiphy, struct wireless_dev *wdev,
 ^
/usr/src/rtl8812AU-1.0/os_dep/linux/rtw_cfgvendor.c:143:2: error: too few arguments to function ‘cfg80211_vendor_event_alloc’
  cfg80211_vendor_event_alloc(wiphy, len, event_id, gfp)
  ^
/usr/src/rtl8812AU-1.0/os_dep/linux/rtw_cfgvendor.c:170:8: note: in expansion of macro ‘rtw_cfg80211_vendor_event_alloc’
  skb = rtw_cfg80211_vendor_event_alloc(wiphy, len, event_id, kflags);
        ^
In file included from /usr/src/rtl8812AU-1.0/include/osdep_service_linux.h:75:0,
                 from /usr/src/rtl8812AU-1.0/include/osdep_service.h:41,
                 from /usr/src/rtl8812AU-1.0/include/drv_types.h:32,
                 from /usr/src/rtl8812AU-1.0/os_dep/linux/rtw_cfgvendor.c:21:
include/net/cfg80211.h:4371:1: note: declared here
 cfg80211_vendor_event_alloc(struct wiphy *wiphy, struct wireless_dev *wdev,
 ^
```

MrEngman, 2015-09-09:
I found the same problem myself a couple of days ago compiling for the Raspberry Pi and I appeared to fix the error with this patch although I still got a warning regarding '.add_virtual_intf'
```
diff -Naur '--exclude=.git' rtl8812AU/os_dep/linux/rtw_cfgvendor.c rtl8812AU0/os_dep/linux/rtw_cfgvendor.c
--- rtl8812AU/os_dep/linux/rtw_cfgvendor.c      2015-09-09 21:47:41.506822692 +0100
+++ rtl8812AU0/os_dep/linux/rtw_cfgvendor.c     2015-09-06 02:48:44.004861936 +0100
@@ -50,7 +50,11 @@
        struct sk_buff *skb;
        unsigned int truesize = 0;

+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 1, 0))
+       skb = cfg80211_vendor_event_alloc(wiphy, NULL, len, event_id, gfp);
+#else
        skb = cfg80211_vendor_event_alloc(wiphy, len, event_id, gfp);
+#endif

        if(skb)
                truesize = skb->truesize;
@@ -139,8 +143,13 @@
 #define rtw_cfg80211_vendor_cmd_reply(skb) \
                dbg_rtw_cfg80211_vendor_cmd_reply(skb, MSTAT_FUNC_CFG_VENDOR|MSTAT_TYPE_SKB, __FUNCTION__, __LINE__)
 #else
-#define rtw_cfg80211_vendor_event_alloc(wiphy, len, event_id, gfp) \
-       cfg80211_vendor_event_alloc(wiphy, len, event_id, gfp)
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 1, 0))
+       #define rtw_cfg80211_vendor_event_alloc(wiphy, len, event_id, gfp) \
+               cfg80211_vendor_event_alloc(wiphy, NULL, len, event_id, gfp)
+#else
+       #define rtw_cfg80211_vendor_event_alloc(wiphy, len, event_id, gfp) \
+               cfg80211_vendor_event_alloc(wiphy, len, event_id, gfp)
+#endif

 #define rtw_cfg80211_vendor_event(skb, gfp) \
        cfg80211_vendor_event(skb, gfp)
```

I generated the fix after searching the Linux source for the Raspberry Pi and finding an example of this update in the file `linux/drivers/net/wireless/ti/wl18xx/event.c`

Googled around the internet and found no reference to the error but the patch appears to work OK.

diederikdehaas, 2015-09-10:
I think both the error and the warning should be gone now :smile: 
Thanks to @rasmorthil for the patch :+1: 

Can you verify whether it also fixes it for you?

qwertzdenek, 2015-09-10:
Thanks, it compiles now, but it seems that we have broken something. I getting packet loss and the device is disconnected. I use NetworkManager but it should not change anything.

```
[  199.537500] RTL871X: linked_status_chk(wlp10s0u2) disconnect or roaming
[  213.490277] RTL871X: There are some pkts to transmit
[  213.490287] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 30
[  220.457744] RTL871X: There are some pkts to transmit
[  220.457753] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
```

Still repeating..

diederikdehaas, 2015-09-10:
Is that `dmesg` output? Because this is what I'm getting when plugging in my adapter (EW-7822UAC) on my Debian Sid  x86_64 against 4.1.6-1 kernel with GCC 5.2.1-16.
I'm using `/etc/network/interfaces` combined with wpa-supplicant and no NM though.
```
[  488.125735] usb 3-4: new high-speed USB device number 3 using ehci-pci
[  488.258557] usb 3-4: New USB device found, idVendor=7392, idProduct=a822
[  488.258568] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  488.258574] usb 3-4: Product: 802.11n NIC
[  488.258580] usb 3-4: Manufacturer: Realtek
[  488.258585] usb 3-4: SerialNumber: 123456
[  488.309220] cfg80211: Calling CRDA to update world regulatory domain
[  488.379149] RTL871X: module init start
[  488.379154] RTL871X: rtl8812au v4.3.8_12175.20140902
[  488.382899] cfg80211: World regulatory domain updated:
[  488.382903] cfg80211:  DFS Master region: unset
[  488.382905] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[  488.382908] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  488.382911] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[  488.382912] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  488.382915] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  488.382917] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  488.382919] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  488.382921] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  488.382923] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[  488.710011] RTL871X: rtw_ndev_init(wlan0)
[  488.710855] usbcore: registered new interface driver rtl8812au
[  488.710860] RTL871X: module init ret=0
[  489.428830] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  493.717989] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[  493.913072] RTL871X: start auth
[  493.915299] RTL871X: auth success, start assoc
[  493.917593] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[  493.917636] RTL871X: assoc success
[  493.917746] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  493.920234] RTL871X: send eapol packet
[  493.929154] RTL871X: send eapol packet
[  493.933131] RTL871X: set pairwise key camid:4, addr:50:46:5d:5e:2e:a5, kid:0, type:AES
[  493.936128] RTL871X: set group key camid:5, addr:50:46:5d:5e:2e:a5, kid:2, type:AES
```

I've found some [more 4.1 related fixes here](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/commit/fae1e8ccae427bd056ed30135176dd70e6dc697e) but I doubt it will improve your situation. With me the I had no compiler warnings at all.

qwertzdenek, 2015-09-10:
Yes it was `dmesg` output. It looks like some conflict with the internal pci wifi card. I would try to dig in when I had more time. I tried it on another computer without NM and the module works flawlessly.

One offtopic question: Any idea what `Calling CRDA to update world regulatory domain` warning means?

closed 2015-09-10 by qwertzdenek

diederikdehaas, 2015-09-10:
I don't think the `Calling CRDA` message is a warning, just an informational message.
I'm not sure what it means, but I think it has to do with the wireless frequencies allowed in the various regions.
This seems relevant: http://linuxwireless.org/en/developers/Regulatory/CRDA/

qwertzdenek, 2015-09-10:
Thanks. I asked you before googling it. My mistake ;). Just paranoia about frequency regulations.