diederikdehaas/rtl8812AU #5: Any chances of a git tag release?

DoctorWho11, 2015-10-21:
Just as the title says, any chances of tagging a release sometime soon? I'm looking to include this driver for solus (solus-project.com) but our packaging guidelines doesn't really allow for anything except a tarball.

diederikdehaas, 2015-10-21:
Short answer is: probably not (soon).

I don't think it's wise to base your driver off of this repo as I am not proficient in C/C++. That means that when bugs are in this code, which is not unlikely as it's based on a code drop of driver version 4.3.22-**beta**, I'm probably not capable of fixing that by myself. 
I also plan to move the code from 4.3.22-beta (back) into a separate branch and update the master branch to 4.3.14. The question is when I get to that as I'm currently focused on [another project](https://github.com/debian-pi/raspbian-ua-netinst).

Having said that, I should start tagging various code points/driver versions :wink:

DoctorWho11, 2015-10-22:
Would be nice, this is atm one of the more active forks of this driver, and it's also a driver some users are wanting to use.

diederikdehaas, 2015-11-12:
I've now pushed several tags for the 4.3.8 version of the driver (with which I started out).
I hope that suffices for your purpose.

I haven't figure out fully how to set up the whole repo and how to deal with other driver versions, but I'll leave the driver-4.3.8 branch with its tags intact.

diederikdehaas, 2015-11-26:
Now there is also a driver-4.3.14 and a driver-4.3.22-beta branch with tags, so I consider this issue fixed.

closed 2015-11-26 by diederikdehaas