diederikdehaas/rtl8812AU #9: Can't change MAC address

tranaconda, 2016-02-03:
I'm trying to use this driver for an access point, which works very well - thank you for this driver! The only thing not working is changing the IP address to get multiple SSIDs with hostapd working. I tried it with `ifconfig hw ether` and with `macchanger`, which said `Network driver didn't actually change to the new MAC!!`. Is it possible to enable modifying the MAC address?

fumantsu, 2016-10-09:
up to this issue. Basically changing the MAC this is one of the suggested ways to use hostapd with simultaneous dual band. 
So in general this driver doesn't allow dual band configuration with hostapd

controlledentropy, 2016-11-22:
that's not sure. 
I have RTL8187, SMSC9512, and a RTL 8192su. RTL8187, as we all know, good for wireless penetrate testing. and my RTL8192su's mac addr is available to change on kali\debian8\centos linux.

So it's possible the problem of raspi linux, not the problem of dongle

someone help please!