diederikdehaas/rtl8812AU #10: rtl8814au?

DanMan, 2016-03-31:
I see mention of that chipset in the sources, but if I plug in a [D-Link DWA-192](https://wikidevi.com/wiki/D-Link_DWA-192), the driver currently doesn't seem to support it, unfortunatly.

What's missing?

diederikdehaas, 2016-06-19:
It looks like Realtek uses much of the same code across chipsets and only uses the parts which are set by conditionals.
It is very likely that you'd need a different driver then this repo provides.

DanMan, 2016-06-23:
Ok, thanks.

weedy, 2016-10-10:
https://aur.archlinux.org/packages/rtl8812au_rtl8821au-dkms-git/

According to this, copies of 4.3.20 should have very complete rtl8812 driver and 4.3.21 has been stripped down and tuned for rtl8814. 
Can we either combine these trees or is anyone interested in cleaning up the 4.3.21 driver to work with new kernels?

weedy, 2016-10-10:
Follow up.
1. Found https://github.com/uminokoe/rtl8812AU/commits/driver-4.3.21
2. Reverted 9260f77aab06f0d4ec8efc9028f221c624619746
3. Reverted 8d3310013bbf98d45d5c60d19d154975fd91f987

I have a WPA2 connection right now to my home router and it seems stable. 
And I can connect to my across the street neighbour.
```
wlan1     IEEE 802.11an  ESSID:"mine"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.745 GHz  
          Bit Rate:300 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=86/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
```
wlan1     IEEE 802.11bgn  ESSID:"neighbour"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.412 GHz 
          Bit Rate:130 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=99/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
ath9k card for comparison
```
wlan0     IEEE 802.11  ESSID:"mine"  
          Mode:Managed  Frequency:5.745 GHz  
          Bit Rate=243 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=55/70  Signal level=-55 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:31   Missed beacon:0
```

weedy, 2016-10-10:
ath9k
```
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=1 ttl=64 time=1.28 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=2 ttl=64 time=1.26 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=3 ttl=64 time=1.31 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=4 ttl=64 time=1.25 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=5 ttl=64 time=1.29 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=6 ttl=64 time=1.28 ms
```
8814au D-Link DeathStar DWA-192
```
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=1 ttl=64 time=3.75 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=2 ttl=64 time=2.17 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=3 ttl=64 time=2.23 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=4 ttl=64 time=2.21 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=5 ttl=64 time=2.23 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=6 ttl=64 time=2.17 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=7 ttl=64 time=2.11 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=8 ttl=64 time=2.15 ms
64 bytes from OpenWrt.lan (192.168.8.1): icmp_seq=9 ttl=64 time=3.21 ms
```

diederikdehaas, 2016-10-10:
@weedy **for testing purposes only** you could try https://github.com/diederikdehaas/rtl8812AU/tree/driver-4.3.20-add-vht-capabilities-wip which is the original 4.3.20 driver to which I added the patches to work up to kernel 4.8.

> Can we either combine these trees or is anyone interested in cleaning up the 4.3.21 driver to work with new kernels?

My intention is to find/create a patch(-set) so that VHT capabilities will be added to the 4.3.20 driver, which are present in 4.3.21 and 4.3.22-beta, but not in 4.3.20.
But this entails comparing 3 different driver archives so it's a bit of work.
I was already aware of Grawp's work, but I hadn't visited his AUR page in a while which now contains some very useful info regarding that and especially the 4.3.21 driver. So thanks for that :+1:

weedy, 2016-10-10:
Doing a quick diff you're completely missing the rtl8814 folder in 4.3.20. So I can't see any way of it working.

It's a bit strange, I'm scrolling through meld and 4.3.20 or 4.3.21 will trade places on refinements.
I think you should use 4.3.21 as the base if you were going to try and merge drivers into a unified tree. It seems to have much fuller beamforming code for example.

weedy, 2016-10-11:
Should I open a new issue or can you reopen this one to keep track of things?

diederikdehaas, 2016-10-11:
I can reopen this issue, but I don't see the need for it.

I just did a quick test by adding the rtl8814au folders into this driver's tree and enabled the setting in the `Makefile`, but running `make` quickly bailed out because of 
`error: nested redefinition of ‘enum _TX_PWR_PERCENTAGE’` 
and a bunch of similar errors.

While it *may* be possible to make those compilation errors go away and thereby giving the appearance of it supporting both rtl8812au and rtl8814au, I don't think it's worth it and on top of that I don't have a rtl8814au device so I won't be able to test it. 

I (therefor) still think my initial response is correct and you should consider rtl8814au a different driver. The 4.3.21 driver is an excellent starting point for that and there's a real good chance that the patches applied to this repo to support newer kernels can be applied simply/cleanly to the rtl8814au driver and I would have no problem *at all* if you'd grab them straight from this repo.

As the code base is indeed rather similar, I do intend to use the 4.3.21 (and 4.3.22-beta) code as **inspiration** to add VHT (80mhz channels) capabilities to the 4.3.20 driver for the rtl8812au chipset.

weedy, 2016-10-11:
@uminokoe has already made a tree with it and used it with his hardware. I had to revert changes that he made to swap it back to a rtl8814au driver. 
So is already a combined driver tree with VHT and 4.3.22-beta stuff pulled in.

uminokoe, 2016-10-12:
@weedy @diederikdehaas 
The [branch driver-4.3.21 in my repository](https://github.com/uminokoe/rtl8812AU/tree/driver-4.3.21) is mainly for RTL8811AU/RTL8812AU/RTL8821AU. I [downloaded the driver here from EDIMAX](http://www.edimax.co.uk/edimax/download/download/data/edimax/uk/download/for_home/wireless_adapters/wireless_adapters_ac1750_dual-band/ew-7833uac) (which is originally targeted for an adapter with RTL8814AU chipset), then made the necessary changes to support newer kernels and 802.11ac operation.

I do not have any RTL8814AU chipset to test, but I am using an adapter with the RTL8811AU/RTL8812AU/RTL8821AU series chipset, operating it in AP mode (hostapd) every day with my 4.3.21 driver. The device generally works fine. I have yet to carry out any STA mode tests. Still, bugs **may** be present and please use the driver at your risk and with caution.

If you are using it on a device with RTL8814AU chipset, you may try reverting the relevant commits, compiling the driver and test it. If you think that the driver is working properly, you can fork / re-upload the code of driver-4.3.21 on GitHub (or Bitbucket, etc.), then use separate branches (e.g. driver-4.3.21_8811_8812_8821 and driver-4.3.21_8814) to store the codes, such that other users will not have to modify the code manually / revert the commits to change the target chipset to RTL8814AU.

TL,DR:
> THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.

P.S. I guess 4.3.21 is similar to 4.3.22-beta ("similar" means less difference when 4.3.21 is compared to other versions, such as 4.3.20)

weedy, 2016-10-12:
@uminokoe I do use your tree and I have reverted 9260f77 and 8d33100, my RTL8814AU works perfect.
I guess I'm trying to push for @diederikdehaas  to adopt your work so we can keep all the fixes here.
This is the most active RTL8812AU repository on github and I don't want to split the "community" if I don't have to. Also I don't know C so I would have to blindly pull things from you and uminokoe and hope they keep working.

uminokoe, 2016-10-12:
Well, the repository owner has the choice as to whether my work will be adopted/integrated. In the meantime, he may also create a new repository (maybe something like `diederikdehaas/rtl8814AU`) and put my work in it (with that two commits reverted). The ease of management will need to be taken into account - in the future, there will be additional commits to support newer kernels, and the same changes/patches will have to be applied to both.
And if in the end the work is adopted, you may want to consider making a package (e.g. AUR) for your Linux distribution, making a blog post to spread the driver, or sharing it with your friends, in hope that some hobbyists/professionals will work on this/these driver(s) and improve it/them.
(I do not know if this is the case for RTL8814AU, but at least for RTL8812AU driver, there are many many versions, making it a challenge to concentrate (put all the effort) on a single version.)

weedy, 2016-10-15:
I just want to comment that I've had 0 problems with the combined driver and it's connection to my router.
And apparently @uminokoe has a stable AP running on it too.

diederikdehaas, 2016-10-16:
> I *guess* 4.3.21 is similar to 4.3.22-beta

What is that *guess* based on?
I've spend way more time that I'd have liked trying to do a 3-way compare on the original downloads, whereby I converted line-endings on the 4.3.20 driver to unix to not make it **really** impossible (4.3.21 and 4.3.22-beta were already unix) :-/

And now I'm thoroughly confused and annoyed by Realtek('s (lack of (proper)) development procedure). :-1: 
Some things are equal between 4.3.20 and 4.3.21, but different on 4.3.22-beta.
Some things are equal between 4.3.21 and 4.3.22-beta, but different on 4.3.20.
Some things are equal between 4.3.20 and 4.3.22-beta, but different on 4.3.21.

And it is not clear to me which of those is 'better' at any given point.
I'm even inclined to think that the 4.3.22-beta driver is the most advanced as several *to do* items in 4.3.20 and 4.3.21 have been replaced with an actual implementation and several things seem to be refactored into separate functions. But its code is older then the 4.3.20 and 4.3.21 drivers based on timestamp and it's also marked as beta.

And as I'm not a C developer, let alone an (very) experienced one, I find it **way** to risky to try to integrate those driver versions of which 4.3.21 is also made for a different chipset.
And 'conclusions' like *it compiles* and/or *it seems to work* aren't good enough for me.
If there was an extensive unit test suite to run, that would be another matter, but there isn't one. And I don't have the skills or inclination to (learn to) write one. 
But it would be truly awesome if someone else would.

diederikdehaas, 2016-10-16:
The above reply may have come across more hostile then I intended; sorry about that.
But I do like to know what your reasons are/were for thinking that 4.3.21 and 4.3.22-beta are similar.

I'm planning to abandon my attempts to add vht capabilities to the 4.3.20 driver as I'm not comfortable with the (imo) substantial changes that I'd need to apply to that driver version in order to support that and I also prefer to stay close to 'upstream' and wait till Realtek releases a driver for rtl8812AU with vht and monitor mode and upload that then.

@uminokoe I'll also add your 4.3.21 driver to this repo, but with a 'disclaimer' that it is based on a driver for the rtl8814au chipset and therefor 'experimental'.

I'll also add the instructions in what to do to make it work on a(n actual) rtl8814au device.
But I'm really uncomfortable with giving users of this repo the impression that *I* (can) actually support this device while I don't have one to verify that even the basic functionality works, let alone the more advanced things. Creating a separate repo for rtl8814au would give the same impression.

> I don't want to split the "community"

I like that, I really do. 
I have thought about that many times before and was even on the verge of creating an organization in the hope of creating such a community where we'd all work together to support and improve these out-of-tree kernel drivers. But I'm afraid that would give an impression I wouldn't be able to live up to and/or be able to sustain such an effort, so I didn't.

After all, I consider myself just some random user on the internet who happen to *find* a more recent driver for my device and uploaded it to GitHub. Just like so many others :-/
The only difference may be that I didn't abandon it after the initial upload.
But the popularity it has gotten since still surprises me.

uminokoe, 2016-10-17:
I appreciate your efforts and I accept your conclusion that there are changes and "todo"s between the three versions. Originally, I made the guess since I found that version 4.3.20's implementation of 802.11ac seemed to be inferior to 4.3.22-beta, and I assumed that this might be why the VHT capabilities could not be added.

I agree that "it seems to be working" may not be enough. If you wish, you can add a disclaimer (just like most pieces of libre software). This way, users have been warned that they are using it on their own risks.

diederikdehaas, 2016-10-17:
> I found that version 4.3.20's implementation of 802.11ac seemed to be inferior to 4.3.22-beta, and I assumed that this might be why the VHT capabilities could not be added.

I'm **quite** sure you're right about that. 
IIRC it actually says somewhere in the comments that the 4.3.20 driver (explicitly) doesn't support *80+80* for example. 
But to me 802.11ac is primarily the higher speed, and the reason I bought [my ac device](http://www.edimax.nl/edimax/merchandise/merchandise_detail/data/edimax/nl/wireless_adapters_ac1200_dual-band/ew-7822uac/), and as I consistently get a connection speed of 867 Mbps (vs 54/144 Mbps for *n*) I've accomplished that goal.
But contributions like yours does make me want to achieve more of ac then just that :smile: 

The main reason that I'm a bit hesitant about 4.3.21 and where the *it compiles and/or it seems to work aren't good enough for me* comes from, is that on at least 2 occasions I saw in the **comments** sth like *on rtl8814au we should do X* and yet Realtek didn't enclose those sections in `#ifdef` structures or sth like that actually testing for that device type. 
And that gives me the creeps. 

And that *could* also mean that the 4.3.21 driver really shouldn't be applied/run on anything else then a rtl8814au device. While I actually don't expect that to be the case, this whole diff-ing process did not make me happy, to put it mildly :wink: 
It really annoyed me that 4.3.19 is for rtl8821au devices, 4.3.20 is for rtl8812au devices and 4.3.21 is for rtl8814au devices and too many diffs are (to me) just different instead of natural progressions/improvements as I would expect from sequential version numbers.
It's like each release was branched of at a semi random point, various fixes were applied to make that work for a (very) specific chipset and then released, but they never bothered to merge those fixes back into mainline/master. 

Addendum:
I just compared the product pages for [EW-7822UAC](http://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/global/wireless_adapters_ac1200_dual-band/ew-7822uac/) with rtl8812au chipset against the [EW-7833UAC](http://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/global/wireless_adapters_ac1750_dual-band/ew-7833uac/) with rtl8814au chipset and made the following observations:

- EW-7822UAC supports 802.11ac **draft** while the EW-7833UAC supports 802.11ac (final)
- EW-7833UAC explicitly states support for *Beamforming* and *3T3R MIMO*, while EW-7822UAC doesn't

:thinking:

diederikdehaas, 2016-10-17:
Ah, fsck it. I'll create a repo for the rtl8814au chipset even though I don't have the device (and it's not available in NL)

weedy, 2016-10-17:
I'll do my best to check any changes you commit to the tree for 14au. If you do want to make an org I'll join it. Our readme can just be "This is a best effort project. None of us know what we are doing, and Realtek sucks at releasing drivers. This will probably kill your pet, we're mostly sorry about that. We did warn you".

Aliexpress has devices for 25usd if you're interested.

diederikdehaas, 2016-10-17:
@weedy, @DanMan:
I've created the https://github.com/diederikdehaas/rtl8814AU repository and added the initial code and on my system it now compiles on kernels up to and including 4.8

weedy, 2016-10-17:
As an FYI since I have access to a higher end AC AP today.
```
wlan1     IEEE 802.11AC  ESSID:"friends"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.745 GHz
          Bit Rate:1.3 Gb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=96/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
```
PING 192.168.2.1 (192.168.2.1) 56(84) bytes of data.
64 bytes from 192.168.2.1: icmp_seq=1 ttl=64 time=2.47 ms
64 bytes from 192.168.2.1: icmp_seq=2 ttl=64 time=2.58 ms
64 bytes from 192.168.2.1: icmp_seq=3 ttl=64 time=2.21 ms
64 bytes from 192.168.2.1: icmp_seq=4 ttl=64 time=4.36 ms
64 bytes from 192.168.2.1: icmp_seq=5 ttl=64 time=2.64 ms
```
```
Oct 17 17:02:42 localhost NetworkManager[8443]: <info>  [1476738162.3996] device (wlan0): supplicant interface state: disconnected -> inactive 

==> /var/log/kern.log <== 
Oct 17 17:02:45 localhost kernel: [1344234.453207] RTL871X: rtw_set_802_11_connect(wlan1)  fw_state=0x00000008 
Oct 17 17:02:45 localhost kernel: [1344234.453221] RTL871X: [by_bssid:1][assoc_ssid:coco][to_roam:0] new candidate: friend(40:65:a3:b0:5c:ff, ch149) rssi:-41 
Oct 17 17:02:45 localhost kernel: [1344234.453233] RTL871X: [HT] HAL Support LDPC = 0x03 
Oct 17 17:02:45 localhost kernel: [1344234.453235] RTL871X: [HT] HAL Support STBC = 0x03 
Oct 17 17:02:45 localhost kernel: [1344234.453236] RTL871X: [HT] HAL Support Beamformer 
Oct 17 17:02:45 localhost kernel: [1344234.453238] RTL871X: [HT] Declare supporting RX LDPC 
Oct 17 17:02:45 localhost kernel: [1344234.453239] RTL871X: [HT] Declare supporting TX STBC 
Oct 17 17:02:45 localhost kernel: [1344234.453241] RTL871X: [HT] Declare supporting RX STBC = 1 
Oct 17 17:02:45 localhost kernel: [1344234.453244] RTL871X: [VHT] Support LDPC = 0x03 
Oct 17 17:02:45 localhost kernel: [1344234.453245] RTL871X: [VHT] Support STBC = 0x03 
Oct 17 17:02:45 localhost kernel: [1344234.453246] RTL871X: [VHT] Support Beamformer 
Oct 17 17:02:45 localhost kernel: [1344234.453247] RTL871X: [VHT] Support Beamformee 

==> /var/log/daemon.log <== 
Oct 17 17:02:45 localhost NetworkManager[8443]: <info>  [1476738165.0629] device (wlan1): supplicant interface state: scanning -> associating 

==> /var/log/kern.log <== 
Oct 17 17:02:45 localhost kernel: [1344234.888586] RTL871X: start auth 
Oct 17 17:02:45 localhost kernel: [1344234.896279] RTL871X: auth success, start assoc 
Oct 17 17:02:45 localhost kernel: [1344234.901529] RTL871X: Current VHT LDPC Setting = 0A 
Oct 17 17:02:45 localhost kernel: [1344234.901531] RTL871X: Current VHT STBC Setting = 0A 
Oct 17 17:02:45 localhost kernel: [1344234.901556] RTL871X: RateBitmap=000000003fffffff , pVHTRate[0]=ea] pVHTRate[1]=ff] 
Oct 17 17:02:45 localhost kernel: [1344234.901583] RTL871X: rtw_cfg80211_indicate_connect(wlan1) BSS not found !! 
Oct 17 17:02:45 localhost kernel: [1344234.901589] RTL871X: assoc success 
Oct 17 17:02:45 localhost kernel: [1344234.901614] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready 
Oct 17 17:02:45 localhost kernel: [1344234.903145] RTL871X: recv eapol packet 
Oct 17 17:02:45 localhost kernel: [1344234.904961] RTL871X: send eapol packet 
Oct 17 17:02:45 localhost kernel: [1344234.907384] ath: EEPROM regdomain: 0x807c 
Oct 17 17:02:45 localhost kernel: [1344234.907388] ath: EEPROM indicates we should expect a country code 
Oct 17 17:02:45 localhost kernel: [1344234.907390] ath: doing EEPROM country->regdmn map search 
Oct 17 17:02:45 localhost kernel: [1344234.907392] ath: country maps to regdmn code: 0x3a 
Oct 17 17:02:45 localhost kernel: [1344234.907394] ath: Country alpha2 being used: CA 
Oct 17 17:02:45 localhost kernel: [1344234.907397] ath: Regpair used: 0x3a 
Oct 17 17:02:45 localhost kernel: [1344234.907399] ath: regdomain 0x807c dynamically updated by country IE 
Oct 17 17:02:45 localhost kernel: [1344234.911022] RTL871X: recv eapol packet 
Oct 17 17:02:45 localhost kernel: [1344234.911711] RTL871X: send eapol packet 

==> /var/log/daemon.log <== 
Oct 17 17:02:45 localhost NetworkManager[8443]: <info>  [1476738165.4687] device (wlan1): supplicant interface state: associating -> 4-way handshake 
Oct 17 17:02:45 localhost NetworkManager[8443]: <info>  [1476738165.4774] device (wlan1): supplicant interface state: 4-way handshake -> completed 
```

diederikdehaas, 2016-10-17:
That looks quite good :smile: 
Is that with uminokoe's driver or with mine? It likely will be the same, but still good to have verification whether my driver works properly (too).

weedy, 2016-10-17:
uhh still his tree. if you just cloned his repo then it would be the same.

In case you have something diff I'll clone yours now.

diederikdehaas, 2016-10-17:
> if you just cloned his repo

No, I didn't. I took the original download of the rtl8814AU drivers and added some patches to support kernels up to 4.8 (which are the same as used for this repo).
See https://github.com/diederikdehaas/rtl8814AU/commits/driver-4.3.21 for details/'changelog'.

weedy, 2016-10-17:
Here we go.
```
rm -f /lib/modules/4.7.4-zen+/kernel/drivers/net/wireless//8814au.ko
/sbin/depmod -a 4.7.4-zen+
install -p -m 644 8814au.ko  /lib/modules/4.7.4-zen+/kernel/drivers/net/wireless/
/sbin/depmod -a 4.7.4-zen+
HAF diederik # modprobe -r 8814au; modprobe 8814au
HAF diederik # iwconfig 
wlan1     IEEE 802.11AC  ESSID:"friends"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.745 GHz
          Bit Rate:1.3 Gb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=92/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
```
# ping 192.168.2.1
PING 192.168.2.1 (192.168.2.1) 56(84) bytes of data.
64 bytes from 192.168.2.1: icmp_seq=1 ttl=64 time=2.22 ms
64 bytes from 192.168.2.1: icmp_seq=2 ttl=64 time=2.32 ms
64 bytes from 192.168.2.1: icmp_seq=3 ttl=64 time=2.50 ms
64 bytes from 192.168.2.1: icmp_seq=4 ttl=64 time=2.33 ms
64 bytes from 192.168.2.1: icmp_seq=5 ttl=64 time=2.50 ms
```
```
Oct 17 18:59:21 localhost NetworkManager[8443]: <info>  [1476745161.7857] sup-iface[0x1f11f70,wlan1] config: set interface ap_scan to 1 
Oct 17 18:59:21 localhost NetworkManager[8443]: <info>  [1476745161.7992] device (wlan1): supplicant interface state: disconnected -> inactive 

==> /var/log/kern.log <== 
Oct 17 18:59:21 localhost kernel: [1351231.244714] RTL871X: =>cfg80211_rtw_connect(wlan1) - Start to Connection 
Oct 17 18:59:21 localhost kernel: [1351231.244718] RTL871X: privacy=1, key=          (null), key_len=0, key_idx=0, auth_type=0 
Oct 17 18:59:21 localhost kernel: [1351231.244723] RTL871X: _rtw_pwr_wakeup call ips_leave.... 
Oct 17 18:59:21 localhost kernel: [1351231.244725] RTL871X: ==>ips_leave cnts:5 
Oct 17 18:59:21 localhost kernel: [1351231.244726] RTL871X: ===>  rtw_ips_pwr_up.............. 
Oct 17 18:59:21 localhost kernel: [1351231.244731] RTL871X: ===> ips_netdrv_open......... 
Oct 17 18:59:21 localhost kernel: [1351231.251866] RTL871X:  power-on :REG_SYS_CLKR 0x09=0xb0. REG_CR 0x100=0xea. 
Oct 17 18:59:21 localhost kernel: [1351231.251868] RTL871X:  MAC has not been powered on yet. 
Oct 17 18:59:21 localhost kernel: [1351231.273361] RTL871X: ===>_InitQueueReservedPage_8814AUsb() 
Oct 17 18:59:21 localhost kernel: [1351231.277610] RTL871X: <===_InitQueueReservedPage_8814AUsb() 
Oct 17 18:59:21 localhost kernel: [1351231.279108] RTL871X: FirmwareDownload8814A fw source from Header 
Oct 17 18:59:21 localhost kernel: [1351231.279111] RTL871X: FirmwareDownload8814A fw:NIC, size: 66904 
Oct 17 18:59:21 localhost kernel: [1351231.279114] RTL871X: FirmwareDownload8814A: fw_ver=25 fw_subver=0 sig=0x8814 
Oct 17 18:59:21 localhost kernel: [1351231.282859] RTL871X: HalROMDownloadFWRSVDPage8814A(): There is an Adapter is sending beacon. 
Oct 17 18:59:21 localhost kernel: [1351231.283984] RTL871X: dmem_pkt_size = 5120, iram_pkt_size = 61720,FWHeaderSize = 64, Len = 66904!! 
Oct 17 18:59:21 localhost kernel: [1351231.283986] RTL871X: MaxRsvdPageBufSize 1488,  Total len 66904 
Oct 17 18:59:21 localhost kernel: [1351231.295355] RTL871X: Check sum OK 
Oct 17 18:59:21 localhost kernel: [1351231.295730] RTL871X: dmem check sum tmp 33 
Oct 17 18:59:21 localhost kernel: [1351231.414151] RTL871X: Check sum OK 
Oct 17 18:59:21 localhost kernel: [1351231.414512] RTL871X: imem check sum tmp 105 
Oct 17 18:59:22 localhost kernel: [1351231.469378] RTL871X: FWDL success. write_fw:0, 190ms 
Oct 17 18:59:22 localhost kernel: [1351231.469382] RTL871X: rtl8814au_hal_init: Download Firmware Success!! 
Oct 17 18:59:22 localhost kernel: [1351231.557621] RTL871X: EEEPROMRegulatory 1 
Oct 17 18:59:23 localhost kernel: [1351233.097498] RTL871X: ==>PHY_SwitchWirelessBand8814() 2.4G 
Oct 17 18:59:23 localhost kernel: [1351233.110900] RTL871X: <==PHY_SwitchWirelessBand8814():Switch Band OK. 
Oct 17 18:59:23 localhost kernel: [1351233.258348] RTL871X: pDM_Odm TxPowerTrackControl = 1 
Oct 17 18:59:23 localhost kernel: [1351233.258352] RTL871X: pDM_Odm TxPowerTrackControl = 1 
Oct 17 18:59:23 localhost kernel: [1351233.267605] RTL871X: MAC Address from REG_MACID = c4:12:f5:5f:da:97 
Oct 17 18:59:23 localhost kernel: [1351233.267610] RTL871X: rtl8814au_hal_init in 2023ms 
Oct 17 18:59:23 localhost kernel: [1351233.267987] RTL871X: ==>PHY_SwitchWirelessBand8814() 5G 
Oct 17 18:59:23 localhost kernel: [1351233.279602] RTL871X: <==PHY_SwitchWirelessBand8814():Switch Band OK. 
Oct 17 18:59:23 localhost kernel: [1351233.393732] RTL871X: hw_var_set_opmode()-4935 mode = 2 
Oct 17 18:59:23 localhost kernel: [1351233.395202] RTL871X: Registry kfree default force disable. 
Oct 17 18:59:23 localhost kernel: [1351233.395216] RTL871X: <===  rtw_ips_pwr_up.............. in 2153ms 
Oct 17 18:59:23 localhost kernel: [1351233.395218] RTL871X: nolinked power save leave 
Oct 17 18:59:23 localhost kernel: [1351233.395573] RTL871X: ==> ips_leave.....LED(0xc0228282)... 
Oct 17 18:59:23 localhost kernel: [1351233.395578] RTL871X: ssid=friends, len=4 
Oct 17 18:59:23 localhost kernel: [1351233.395580] RTL871X: bssid=40:65:a3:b0:5c:ff 
[... 96 lines cut ...]
Oct 17 18:59:24 localhost kernel: [1351233.846751] RTL871X: seq_len=0x6 
Oct 17 18:59:24 localhost kernel: [1351233.846752] RTL871X: key_index=0 
[... cut ...]
Oct 17 18:59:24 localhost kernel: [1351233.846893] RTL871X: rtw_cfg80211_set_encryption, ret=0 
Oct 17 18:59:24 localhost kernel: [1351233.847003] RTL871X: cfg80211_rtw_change_station(wlan1) 

==> /var/log/daemon.log <== 
Oct 17 18:59:24 localhost NetworkManager[8443]: <info>  [1476745164.4042] device (wlan1): supplicant interface state: associating -> completed 
```
I need to make a PR for the couple edits I made to shut up most debug but not alllllllll debug.
I forgot how noisy this thing is stock. My poor SSD.

diederikdehaas, 2016-10-17:
> I need to make a PR for the couple edits I made to shut up most debug but not alllllllll debug.

No need for that. I'll `cherry-pick` commit https://github.com/diederikdehaas/rtl8812AU/commit/3e80ebc6bae5695b1317925497c2ead167e09a48

weedy, 2016-10-17:
It's not the same. I'll show you later.

diederikdehaas, 2016-10-17:
Ok :+1: 

Just found the following in the `Makefile`:
```
###################### MP HW TX MODE FOR VHT #######################
CONFIG_MP_VHT_HW_TX_MODE = n
```
So it looks like VHT stuff isn't enabled (by default)

weedy, 2016-10-17:
I saw that, and then I grep'ed the code. I don't know NEAR enough C (read none) to figure out if that's specifically VHT or some kind of VHT-in-the-firmware-something-something.
AND @uminokoe did this thing on his tree https://github.com/uminokoe/rtl8812AU/commit/5f75242fce32cc1d86cb80a10ffdb7da9dac5185.

diederikdehaas, 2016-10-18:
Just applied the following diff and it still compiles ... : 
```diff
diederik@bagend:~/dev/drivers/rtl8814AU$ git diff
diff --git a/Makefile b/Makefile
index addda91..d907b0e 100644
--- a/Makefile
+++ b/Makefile
@@ -74,7 +74,7 @@ CONFIG_AP_WOWLAN = n
 ######### Notify SDIO Host Keep Power During Syspend ##########
 CONFIG_RTW_SDIO_PM_KEEP_POWER = y
 ###################### MP HW TX MODE FOR VHT #######################
-CONFIG_MP_VHT_HW_TX_MODE = n
+CONFIG_MP_VHT_HW_TX_MODE = y
 ###################### Platform Related #######################
 CONFIG_PLATFORM_I386_PC = y
 CONFIG_PLATFORM_ANDROID_X86 = n
@@ -528,8 +528,8 @@ endif
 ########### HAL_RTL8814A #################################
 ifeq ($(CONFIG_RTL8814A), y)
 ## ADD NEW VHT MP HW TX MODE ##
-#EXTRA_CFLAGS += -DCONFIG_MP_VHT_HW_TX_MODE
-#CONFIG_MP_VHT_HW_TX_MODE = y
+EXTRA_CFLAGS += -DCONFIG_MP_VHT_HW_TX_MODE
+CONFIG_MP_VHT_HW_TX_MODE = y
 ##########################################
 RTL871X = rtl8814a
 ifeq ($(CONFIG_USB_HCI), y)
```

But mostly interested in the changes you saw with uminokoe's patch applied.

uminokoe, 2016-10-18:
Please forgive me for being stupid as I do not have knowledge in programming or WiFi adapter driver development.
If there is an option that can "automagically" enable the VHT stuff, I would be really happy. My question is that, besides `Makefile`, the code `CONFIG_MP_VHT_HW_TX_MODE` also appeared in `os_dep/linux/ioctl_mp.c`, `hal/hal_mp.c`, and `core/rtw_mp.c`. What does "mp" mean? I _guess_ it may mean [Mesh Point](https://wireless.wiki.kernel.org/en/users/drivers/ath10k/mesh), and [here](http://www.arubanetworks.com/techdocs/ArubaOS_63_Web_Help/Content/ArubaFrameStyles/Mesh/Mesh_Access_Points.htm) may be of help in explaining the term.
You can try enabling that and [check the iw output](https://wiki.archlinux.org/index.php/Software_access_point):
``` 	
$ iw list
(...)
Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * WDS
		 * monitor
		 * mesh point <- look for this
```
Even if mesh point is supported, I am unsure whether the implementation is good enough for use.

diederikdehaas, 2016-10-18:
> Please forgive me for being stupid as I do not have knowledge in programming or WiFi adapter driver development.

Welcome to the club! LOL

I went through the code as well and I primarily noticed how **very** few useful comments are present in the code. Iow, I have no clue (either).

weedy, 2016-10-18:
> Please forgive me for being stupid as I do not have knowledge in programming or WiFi adapter driver development.

F***ing same, so very same
```
 # grep CONFIG_MP_VHT_HW_TX_MODE Makefile 
CONFIG_MP_VHT_HW_TX_MODE = y
EXTRA_CFLAGS += -DCONFIG_MP_VHT_HW_TX_MODE
CONFIG_MP_VHT_HW_TX_MODE = y
ifeq ($(CONFIG_MP_VHT_HW_TX_MODE), y)
EXTRA_CFLAGS += -DCONFIG_MP_VHT_HW_TX_MODE
# iw phy
Wiphy phy31
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
	Band 1:
```