diederikdehaas/rtl8812AU #11: Error in command { sudo git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION} }

AntonioAlgaida, 2016-04-13:
When I run the fourth command I get this error.

```
a@a-L50:~/Downloads/rtl8812AU$ sudo git archive driver-${DRV_VERSION} | tar -x -C  usr/src/${DRV_NAME}-${DRV_VERSION}
tar: .gitignore: No se puede open: Permiso denegado
tar: Kconfig: No se puede open: Permiso denegado
tar: Makefile: No se puede open: Permiso denegado
tar: README.md: No se puede open: Permiso denegado
tar: clean: No se puede open: Permiso denegado
tar: core: No se puede mkdir: Permiso denegado
tar: core: No se puede mkdir: Permiso denegado
.
.
.
tar: platform: No se puede mkdir: Permiso denegado
tar: platform/platform_ops.h: No se puede open: No existe el archivo o el directorio
tar: platform: No se puede mkdir: Permiso denegado
tar: platform/platform_sprd_sdio.c: No se puede open: No existe el archivo o el directorio
tar: runwpa: No se puede open: Permiso denegado
tar: wlan0dhcp: No se puede open: Permiso denegado
tar: Saliendo con fallos debido a errores anteriores
```

Do not run this command.
What may be due?

diederikdehaas, 2016-04-13:
My guess is that the `tar` command is run as your normal user, while it needs to be executed as root.

AntonioAlgaida, 2016-04-15:
Ok, now it has been installed.

But now, I want to know the transmission power, but this does not appear when I iwconfig.
This is the output:

```
a@a-L50:~/Downloads/rtl8812AU$ iwconfig
wlan3     IEEE 802.11bgn  ESSID:"Free Candies"  Nickname:"<WIFI@REALTEK>" ##This is TP-Link Archer T4U with rtl8812AU
          Mode:Managed  Frequency:2.437 GHz  Access Point: AA:BB:CC:DD:EE:FF   
          Bit Rate:300 Mb/s  Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=100/100  Signal level=67/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

wlan0     IEEE 802.11abgn  ESSID:"Free Candies"  ## This is internal board
          Mode:Managed  Frequency:2.437 GHz  Access Point: AA:BB:CC:DD:EE:FF 
          Bit Rate=135 Mb/s   Tx-Power=22 dBm  
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=49/70  Signal level=-61 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:1  Invalid misc:375   Missed beacon:0

lo        no wireless extensions.

```

Why not transmit power appears?
And another thing is that it is fixed to 300MB / s

Thanks so much.
A greeting.
AntonioG.

diederikdehaas, 2016-04-15:
> Why not transmit power appears?

I don't know actually. Maybe this driver doesn't have a function for it? 

> And another thing is that it is fixed to 300MB / s

I think that's due to using 2.4Ghz. On my system, using 5Ghz, I get `867 Mb/s`

diederikdehaas, 2016-06-19:
@AntonioAlgaida could you try the change from this commit: https://github.com/yjcoshc/rtl8812AU/commit/69d4700a918e1e603cafaefae19f60efd1552bb0 ?