diederikdehaas/rtl8812AU #15: rtl8812AU-4.3.22-beta disconnecting after a while

jhonny-oliveira, 2016-05-11:
Just installed this version rtl8812AU-4.3.22-beta on Unbuntu 16.04, but it keeps on throwing errors in syslog and eventually it stops working at all (the later issue may not be related to the first). This is a common issue to all drivers I came across and installed on Ubuntu 16.04.

Installation procedure and syslog are attached. Let me know if you need further information.
[rtl8812AU-driver-4.3.22-beta.on.Ubuntu16.04.syslog.txt](https://github.com/diederikdehaas/rtl8812AU/files/260303/rtl8812AU-driver-4.3.22-beta.on.Ubuntu16.04.syslog.txt)
[rtl8812AU-driver-4.3.22-beta.on.Ubuntu16.04-Install.txt](https://github.com/diederikdehaas/rtl8812AU/files/260304/rtl8812AU-driver-4.3.22-beta.on.Ubuntu16.04-Install.txt)

diederikdehaas, 2016-06-19:
This sounds like a NetworkManager issue and not so much with this driver.
It talks a lot about `wlan0` but you have predictable interface naming and it should therefor deal with `enx801f02bd22fb`. Another thing I noticed is that it looks like its trying to use `wext` first as messages like `ioctl[SIOCGIWSCAN]: Resource temporarily unavailable` I only know in combination with `wext`. The non-deprecated wireless stack is `nl80211` and is what this driver uses.
The last thing I can think of is that it may be a hardware issue.

The first thing I'd try to do is removing NetworkManager and see whether the problem persists.

diederikdehaas, 2016-10-11:
Any updates on this?

In the `Makefile` is the following setting: `CONFIG_POWER_SAVING = y` and you could try setting it to `n` to see whether that improves things.

dumper93, 2017-02-21:
It stops after 4,x minutes airodump-ng. 
CONFIG_POWER_SAVING = n
Does not help ... .

tjphippen, 2017-08-28:
CONFIG_POWER_SAVING = n
May or may not help me. Do I simply update the make file & run make install again? or now that it's already installed do I need to change it elsewhere?