diederikdehaas/rtl8812AU #17: It doesn't work on Fedora

lucaciavatta, 2016-06-19:
I have successfully installed the Driver on Fedora 24 Beta with Kernel 4.5.7:

`Module rtl8812AU/4.3.14 already installed on kernel 4.5.7-300.fc24.x86_64/x86_64`

but it seems not working because my adapter (Alfa 036AC) is still turned off.

Here my lsusb:

`Bus 001 Device 008: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
`
Any idea to make it working?

diederikdehaas, 2016-06-19:
When you plug in your device and then do `dmesg | tail` do you see a message related to your device?

It doesn't seem to work here either on 4.6.1-1 ...

diederikdehaas, 2016-06-19:
I have just merged and pushed a PR by @napalu which fixed the issue for me.
Can you test whether it also fixes your issue?

If not, could you post the full build log?

lucaciavatta, 2016-06-20:
I investigated and my issue seems related to SecureBoot. I need to sign the kernel module with a secure key, so I'm looking how to do it.

diederikdehaas, 2016-06-20:
Then I'm closing this issue. 
If you still have issues after fixing the SecureBoot issue, feel free to reopen it.

closed 2016-06-20 by diederikdehaas

lucaciavatta, 2016-06-20:
Ok, thanks for your attention.