diederikdehaas/rtl8812AU #19: Unable to lock on channel in monitor mode

abilify01, 2016-06-26:
airodump-ng shows channel the channel is locked but channel hops or defaults to channel 1. Unable to lock on channel.
This is the best driver for the Alpha awus036ach that I have came across. Thanks for all the hard work.

well0nez, 2016-09-23:
i have similar issues with another realtek driver

diederikdehaas, 2016-09-23:
I have no experience with monitor mode, so I won't be able to offer (direct) help, but hopefully this issue attracts people who do :-)
With which driver version have you tried it? There is also a 4.3.22-beta and a 4.3.20 version is coming.

replete, 2016-09-23:
I've tried all the RTL8811AU drivers I can find, and all have problems with `aircrack`.

It would be nice to find where the drivers are actually coming from rather than wait for product customers to upload them!

diederikdehaas, 2016-10-10:
> 4.3.22-beta

Have you tried that driver? That version has a [`CONFIG_WIFI_MONITOR = y`](https://github.com/diederikdehaas/rtl8812AU/blob/driver-4.3.22-beta/Makefile#L76) which the 4.3.14 version (the current default) does not.

brennnaaannn, 2016-10-11:
I have tried monitor mode as well as injection on the 4.3.22-beta driver but does not work.

diederikdehaas, 2016-10-11:
I found these links which may be of help:
https://alfanetworkinc.blogspot.com/2016/08/awus036ac-awus036ach-awus036eac-kali.html
https://forum.aircrack-ng.org/index.php/topic,1086.0.html

pcamelo, 2017-07-17:
I'm having the same problem, it also hops randomly between different channels while I specifically set it to a fixed channel via command line.