diederikdehaas/rtl8812AU #24: DKMS Build Fails

arrmo, 2016-08-26:
Hi,

This is likely me, but in any case - for other dummies like me ... ;)

Build is working fine - but to make it run happily on RPi, I have to make as follows,
`make ARCH=arm -j4`

OK, the j4 is just for speed, but I need the ARCH=arm line, or it tries to use armv7l, and the build fails.  So I tried adding this in dkms.conf, to the make command. When I try to run a DKMS build, it errors out, as follows,

`Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make ARCH=arm -j4'....(bad exit status: 127)
Error! Bad return status for module build on kernel: 4.4.19-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.14/build/make.log for more information.`

And, if I look at the log file, I see the following,
`DKMS make.log for rtl8812AU-4.3.14 for kernel 4.4.19-v7+ (armv7l)
Fri 26 Aug 17:55:12 CDT 2016
/usr/sbin/dkms: line 74: make ARCH=arm -j4: command not found
`

Thoughts?

Thanks!

arrmo, 2016-08-27:
My apologies - my fault! Seems dkms is very particular about quote locations (even more than normal). FYI, this works,
MAKE="'make' ARCH=arm -j4"

And this doesn't,
MAKE="'make ARCH=arm -j4'"

Sorry!

closed 2016-08-27 by arrmo

diederikdehaas, 2016-08-27:
> CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_ARM_RPI = n

What if you remove the DKMS thing (instructions in README.md) and switch the y/n in the 2 lines quoted above in the `Makefile` (line 86) and then re-add it, but don't modify the `MAKE` command?

arrmo, 2016-08-28:
Sorry, trying to understand ... I am building on RPi, not PC - so why swap these?

Thanks!

diederikdehaas, 2016-08-28:
When you set `CONFIG_PLATFORM_ARM_RPI = y` (and i386 to `n`), it uses [settings optimized for the Raspberry Pi](https://github.com/diederikdehaas/rtl8812AU/blob/driver-4.3.14/Makefile#L890) :wink:

arrmo, 2016-08-28:
Yep, that works! I thought you wanted it the other way around, that's why I was confused.

I still keep the -j4 => much faster ... :)

Thanks!

diederikdehaas, 2016-10-27:
> I still keep the -j4 => much faster

On Debian, dkms version 2.2.0.3-4 automatically detects the number of CPUs and adjusts the `make` call accordingly: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=663114

:tada: