diederikdehaas/rtl8812AU #25: AC Support

arrmo, 2016-08-29:
Hi,

I really don't like posting this as an issue - as it's a question really, and likely me being dumb ... but "issue" seems to be the only option. Just please don't take this as trying to flag a problem!

Driver compiled, installed, all working great - thanks so much! Working through hostapd now, gradually figuring it out, but now an odd question (OK, really to of them ... LOL).

I have reviewed the information provided back from iw list, and it's not showing any VHT capability ... yet this device / driver should support 802.11ac - right? I would like to use this (at 5 GHz of course), to up my bitrates, but not seeing it as possible. And I admit, not sure if this is a driver issue, or rather hostapd, or even the HW (though the HW spec notes 433 MHz support, which is 1x1 MIMO, at 80 MHz bandwidth). Thoughts on this?

Also an oddity ... iw list shows the following information,
` Capabilities: 0x1862
HT20/HT40
Static SM Power Save
RX HT20 SGI
RX HT40 SGI
No RX STBC
Max AMSDU length: 7935 bytes
DSSS/CCK HT40
`

But when I add [SMPS-STATIC] to ht_capab, I get an error message that this is not supported. Again ... driver, hostapd, or HW issue? It's hard to say for sure, but you may have some experience here.

FYI, I'm running the 4.3.14 driver - it may be that the later driver helps out here. I can try that also if you recommend, no issue at all!

arrmo, 2016-08-29:
OK, digging a bit - I do see in rtw_vht.c, the "large" check for,
#ifdef CONFIG_80211AC_VHT

but not seeing this defined anywhere ... or perhaps I'm just missing it?

Thanks!

diederikdehaas, 2016-08-29:
> this device / driver should support 802.11ac - right?

Yes. And due to the ac + 5Ghz, my connection is at 867Mbps.
There is PR https://github.com/diederikdehaas/rtl8812AU/pull/22 regarding VHT, but I haven't had time to look at it yet. That PR is targeted at 4.3.22-beta, so I'd suggest to try that branch too as it generally has newer code.

I hope others will chime in as this is likely the best I can do (for now at least).

arrmo, 2016-08-29:
I like the question label ... :). Thanks!

Tried 4.3.22-beta. It builds and installs, but not seeing any VHT info (iw list) - should I? Also, the "Capabilities" section is missing using this driver. perhaps these are the PR #22 fixes?

Are you using hostapd, and getting 867 Mbps? I admit, confused by that ... :(.

Thanks again.

diederikdehaas, 2016-08-29:
The 867 Mbps is between my adapter and wireless AC router (Asus).
Until very recently I didn't have a device to test the adapter in hostapd mode, at least not in ac mode. I do now, but haven't had time to work/play with it yet. 
Furthermore, I'm not really knowledgeable about hostapd or VHT and I actually only learned about the latter through [one of the PRs](https://github.com/diederikdehaas/rtl8812AU/pulls?utf8=%E2%9C%93&q=is%3Apr) where others (@uminokoe) did the real work.

arrmo, 2016-08-29:
No worries - we'll figure it out.

More than willing to test things here. Do you recommend I try with the proposed changes (in the pull request)? I admit, not a git expert - not sure how I can do that locally (but not have you do it to your repository).

Thanks!

diederikdehaas, 2016-08-30:
> Do you recommend I try with the proposed changes (in the pull request)?

Yes. Here's a simplified procedure to accomplish that:
1. Fork my repo, so it becomes available on https://github.com/arrmo/rtl8812AU
2. Clone *your* fork locally with `git clone https://github.com/arrmo/rtl8812AU`
3. `cd` into that directory with `cd rtl8812AU`
4. Add @uminokoe's repo as a remote: `git remote add uminokoe https://github.com/uminokoe/rtl8812AU`
5. Checkout uminokoe's driver-4.3.22-beta branch into your own local branch: `git checkout -b uminoke-vht-capabilities uminokoe/driver-4.3.22-beta`

After the last step you should see:
```
Branch uminoke-vht-capabilities set up to track remote branch driver-4.3.22-beta from uminokoe.
Switched to a new branch 'uminoke-vht-capabilities'
```
And then you'd have uminokoe's PR #22 locally and can experiment all you want :smile:

arrmo, 2016-08-30:
Hi,

Actually, I took a "cheat" way for now - just cloned his repository, as it matches to PR #22. Cut the corner a bit ... ;).

Compiled, and running right now. It does get AC working, that's awesome! One thing I did notice thought ... either pre- or post- PR #22, if I try "iw list" before running hostapd - the returned information is not complete. At least "Capabilities" is missing. After running hostapd, then this information shows up (some sort of initialization issue?).

Thanks!

uminokoe, 2016-08-30:
Sorry for the confusion. Actually, some configurations for VHT capabilities are "hidden" inside the `os_dep/linux/os_intfs.c` file. If you want to test the VHT capabilities, you can edit that file. Having some knowledge about endianness and binary/hexadecimal numeral system is recommended. My repository keeps the default settings of the aforementioned file.

You may wish to change the following:
`rtw_rx_stbc`: Rx STBC enable/disable on 2.4G/5G
`rtw_ldpc_cap`: Tx/Rx LDPC enable/disable on HT/VHT
`rtw_stbc_cap`: Tx/Rx STBC enable/disable on HT/VHT

Note: SMPS seems to be unusable. I also faced the same problem and I suspect that this is a driver problem (e.g. bad implementation). Please do not use the `[SMPS-STATIC]` flag in `hostapd.conf`.
Example hostapd `vht_capab` flags:
`vht_capab=[SHORT-GI-80][MAX-A-MPDU-LEN-EXP7][RX-STBC-1][MAX-MPDU-11454][SU-BEAMFORMEE][HTC-VHT]`
Example iw version 4.7 output of VHT capabilities after modifications to `os_intfs.c`:
```
                VHT Capabilities (0x03c03122):
                        Max MPDU length: 11454
                        Supported Channel Width: neither 160 nor 80+80
                        short GI (80 MHz)
                        SU Beamformee
                        +HTC-VHT
```
If you still have problems on VHT capabilities, please post the VHT capabilities (refer to the above example).

arrmo, 2016-08-31:
Thanks - much appreciated! It's funny, as the vht_capab you list above is pretty much what I have configured.

Odd, but I purposely left hostapd running last night, using the driver noted above - and while it's still running, I no longer see the SSID ... so it's crashed? Restarted hostapd, shows "wlan1: AP-ENABLED" ... but no SSID broadcast.

Thoughts?

Thanks again!

uminokoe, 2016-08-31:
Usually, I run hostapd (**version 2.5**) and have it operate continuously for about 3 hours daily without any problem. Sometimes (maybe 5-10% chance), the dongle will not "cooperate" and the LED light keeps blinking, then I just reconnect the dongle and/or restart the service. Occasionally (maybe 20-40% chance) for Android clients, some time is needed before it can detect the presence of the network.
For me, I avoided any channel that requires radar detection (for less trouble). In addition, I am using a rolling-release Linux distribution.
I believe it is hard to "have the best of both worlds". Creating a "software" access point on Linux offers customization (and maybe security, too), but the performance (network throughput) may not be the best. Furthermore, if the driver has lots of flaws, stability and functions will be limited. Unfortunately I do not have adequate knowledge to diagnose and modify the driver, so I guess I will accept the limitations until the day a nearly perfect driver gets released.
P.S. I also do not know why I cannot get a decent speed using this driver - for a 433 Mbps device, the actual throughput is around 160-170 Mbps. If you can get a good result (250 Mbps or higher), congratulations to you.

arrmo, 2016-08-31:
Thanks for the thoughts! A few questions now,
- hostapd, v2.5: do you find this better than v2.3? v2.3 is "stock" on Raspbian (what I'm running). I can try v2.5, no issues here.
- my issue above: the fix was to rmmod and insmod, so remove and reinstall the driver. Seems to point to the driver then, no?
- speed: need to check (iperf), but I do see 433 Mbps at the interface level.

Thanks again.

arrmo, 2016-08-31:
Hi,

Perhaps a question as I try to work through this ... is there a capability to output debug information from the driver, perhaps to a specified file?

Thanks!

diederikdehaas, 2016-08-31:
> is there a capability to output debug information from the driver

Yes, see/revert this commit: https://github.com/diederikdehaas/rtl8812AU/commit/49f88fb0ec13949c78a7ec56dfe0862e3d2e948b

uminokoe, 2016-09-01:
I think it is safe to use a newer version of hostapd. Ubuntu 16.04 uses version 2.4 so I guess it is stable enough. However, that means you will have to compile hostapd by yourself if the repository does not contain the 2.4/2.5 binary package.
Regarding the speed, at the "interface level", I also get 433 Mbps. My iperf test only gives me <200 Mbps throughput though.

arrmo, 2016-09-03:
Hi,

OK, some testing here - hopefully some help, and some thoughts (and an explanation why your 200 Mbps result sounds correct ... :)). 
1) Tested a different driver / Wi-Fi adapter (Asus USB-N53): Wi-Fi rate ~ 225 Mbps, iperf (to RPi) ~ 100 Mbps. Just a baseline - seems OK for this device.
2) Tested this driver (pull request), with Edimax ew-7811utc. Wi-Fi rate = 433 Mbps, iperf (to RPi) ~ 170 Mbps. Perhaps slightly limited, but not bad.
3) Tested to a different machine on the network, with AC router. Wi-Fi rate 650 Mbps, iperf (Linux box) ~ 300 Mbps. Also seems OK.

So the observations,
a) Remember that Wi-Fi is half-duplex, so if you get 1/2 of the Wi-Fi rate, things are about right. This is why I say your numbers seem reasonable, as do the testing results I found. Perhaps a very slight bit low in case #2 above, but I'm also not sure what the RPi can handle. I didn't test across the network (LAN speed only 100 Mbps for RPi), rather to the RPi itself.
b) So the driver seems OK to me, from a speed perspective. But ... I had the driver "hang" again ... no SSID broadcast. Again had to remove and reinstall the driver. BTW, this is all with hostapd v2.5 (built it locally). To me this is the main concern ... driver exiting. Any thoughts on what to try to output to debug? Just crank up the debug level for now, log to a file?

By all means let me know your thoughts. Thanks!

uminokoe, 2016-09-03:
I am more interested in the speedtest (speedtest.net) result because sometimes I suspect that I am using iperf wrongly and cannot get the best result.
Anyway, I searched the Web for the speed. Here, I list the results that I found:
```
Result      Client      Router/AP      Speed      Link
1           433Mbps     1.3Gbps        ~150Mbps   https://mrncciew.com/2014/01/10/802-11ac-with-cisco-3700-ap/
2           867Mbps     433Mbps        ~230Mbps   https://www.youtube.com/watch?v=qsVs0rWPKxc
3           433Mbps     Unknown        ~279Mbps   https://github.com/gnab/rtl8812au/issues/6#issuecomment-225077548
```
That means maybe the quality of the 802.11ac card matters. If it is not the driver problem, then I guess the only reason is that the chipset / dongle / card is poor. With 802.11n 150Mbps (theoretical), I can get around 90Mbps actual throughput. However the so-called "433Mbps" (theoretical, which is approximately three times the 802.11n, only gives double throughput.

Have you tried the device on a x86-64 platform? I suspect that ARM platform may be causing the issue. Nonetheless, I have never tried testing the network overnight / for many hours, and I don't have any ARM device to test, so I can't tell. What I have experienced so far is that 3 hours of continuous operation on x86-64 platform seems OK.

arrmo, 2016-09-03:
Hi,

I think your internet connection may be faster than mine ... ;). I only have 85 Mbps to the internet, and 100 Mbps from the RPi - that's why I was using iperf, to test to the box itself only. I would like to set up an internal speedtest (like speedtest.net), but not sure that's possible ... is it? That said, without that I'm limited to iperf, or by my internet connection.

I was at the gym this morning, was thinking the same thing - plan to try the driver on an Intel x86-64 box today. Will let you know.

Thanks!

arrmo, 2016-09-03:
FYI, tested on an Intel machine, Ubuntu 16.04 -> driver build, hostapd running ... exactly the same as 2) above! Wi-Fi rate at 433 Mbps, then the output from iperf (tried iperf2 and iperf3, iperf3 below),

[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-10.11  sec   201 MBytes   167 Mbits/sec                  sender
[  5]   0.00-10.11  sec   200 MBytes   166 Mbits/sec                  receiver

To me, overall this looks very good - your proposed updates seem to be working. I just need to see if the "crash" (for lack of a better term) is the driver, or hostapd.

Thanks!

diederikdehaas, 2016-09-03:
Just a quick msg from my side: I hope to be able to look into this and @uminokoe's PR next week, but the week thereafter is more likely. But I surely haven't forgotten this, just short on time atm.

arrmo, 2016-09-03:
Not a worry here at least - we'll keep working through things. Thanks!

uminokoe, 2016-09-04:
My situation is as follows:
My Internet: 300Mbps
My Ethernet controller: 10/100Mbps (Fast Ethernet) (Bottleneck)
If I am going to buy a Gigabit Ethernet adapter, I will be able to test speedtest.net but it takes _at least a week_ for me to receive the product. Please be patient and stay tuned.

arrmo, 2016-09-04:
No worries! FYI, here is an interesting test option, http://www.speedtest.net/mini.php

I did install it on my local box, it works fine ... so then you can test without the limitation of the Ethernet link out of the box.

uminokoe, 2016-09-05:
I have tried that "Speedtest mini" before. It is useless because the test duration is too short. The "full" speedtest lasts around 10 seconds (per upload/download test), but the mini test lasts 1 or 2 seconds only. (I guess the size of the test file is too small, so it only works for slower Internet connection which will take some time to transferring the file.) Therefore the result may not be accurate.

arrmo, 2016-09-05:
Yep, tried it here - same conclusion, agree with you! I think a mix of Speedtest and iperf makes sense - iperf because you can isolate the link between the client and server (alone).

Thanks!

diederikdehaas, 2016-10-05:
> Example iw version 4.7 output of VHT capabilities after modifications to os_intfs.c

I tried `iw list` ([iw version = 3.17](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=816457)) and didn't see those VHT capabilities. 
Therefor:

1. If the driver doesn't report those capabilities, it seems likely that it also won't be used, right?
2. Can you share a patch of the needed changes to `os_ints.c` so that I can also see them (on 4.3.22-beta)?

uminokoe, 2016-10-05:
Assuming that the capabilities are all implemented correctly (i.e. the features such as STBC, Beamformee, etc. work well), you must enable them in the code and then compile to use the features in AP or STA mode. In AP mode, hostapd should refuse to start if you enabled the flag(s) in `hostapd.conf`, but have the respective feature(s) disabled in the driver.
You should see a hexadecimal value after the words "VHT capabilities", as shown in the example below:
```
                VHT Capabilities (0x03c03122):
                        Max MPDU length: 11454
                        Supported Channel Width: neither 160 nor 80+80
                        short GI (80 MHz)
                        SU Beamformee
                        +HTC-VHT
```
The hexadecimal value is the representation of the VHT capabilities. If you convert it into binary form (should be 4 bytes in length), you will get the complete info. See [http://chimera.labs.oreilly.com/books/1234000001739/ch03.html](http://chimera.labs.oreilly.com/books/1234000001739/ch03.html) for more information. Notice that some capabilities may not be displayed by iw (such as VHT STBC), and you will have to depend on the hexadecimal value to  determine whether it is enabled.

In version 4.3.22 beta, you will have to refer to [my comment above](https://github.com/diederikdehaas/rtl8812AU/issues/25#issuecomment-243328604), edit the file and find the variables, then change the assigned value. Here is an example: for line 114 `int rtw_rx_stbc = 1;`, if you wish to use Rx STBC in 5GHz, you change the assigned value from 1 to 3. (1 means 01 (bit 0 = 1, bit 1= 0), 3 means 11 (bit 0 = 1, bit 1= 1)
In version 4.3.21, Realtek **enabled the capabilities by default** so it is unlikely that you will have to edit the code to get the features working.