diederikdehaas/rtl8812AU #26: 4.3.22-beta Monitoring and Injection

brennnaaannn, 2016-08-30:
Firstly, thanks for the beta driver. I have it all installed and working successfully.

However, I am only able to get into monitor mode by using - _iwconfig wlan0 mode monitor_

If I try and run - _airmon-ng start wlan0_

I get the error - Warning: Unable to start monitor mode, please run "airmon-ng check kill"

After running 'check kill' it still errors out with the same message. All attempts to inject packets using aireplay-ng fail also. The card always appears to be looking for the BSSID or ESSID on channel 1, I am under the impression this is because I am not able to lock the card to a channel using airmon-ng.

Has anyone achieved either or both of the above on this chipset?

replete, 2016-09-23:
I'm having the same problems with this driver and configuration.

I've read somewhere that users have worked around this by manually setting the channel.

diederikdehaas, 2016-09-24:
As I don't have experience with monitor mode I won't be able to contribute much (if any).
What you could try is revert commit https://github.com/diederikdehaas/rtl8812AU/commit/49f88fb0ec13949c78a7ec56dfe0862e3d2e948b to re-enable debugging output and maybe that would give you a clue as to why it isn't working.

brennnaaannn, 2016-09-28:
Replete - I did manage to get the channel to lock but no matter what the injection will never work. I actually don't think monitor mode was working correctly in these instances either, it just appears to be...

diederikdehaas - Thanks, I will give that a go and see what output I can get. Is there anything newer than 4.3.22-beta that you or anyone else is working on?

diederikdehaas, 2016-09-28:
> Is there anything newer than 4.3.22-beta that you or anyone else is working on?

I do have 4.3.20 (final) locally that I plan to add to this repository. Even though the version number is/seems lower, the date it was published is later then that of 4.3.22-beta.

brennnaaannn, 2016-09-28:
I would be very interested in giving that a shot with what I am trying to achieve. Then enabling debug output on it as a comparison to the 4.3.22.

iamtrying-127times, 2016-10-28:
Someone reports that https://drive.google.com/file/d/0B5CVb15vbMjCeXJsNUFRUUJPZEE/view works with injection and monitor mode. But i cant compile it. In fact, your repository is the best one. No trouble to install and get it work. May you look it?

diederikdehaas, 2016-10-28:
I'm quite sure that's the exact same archive that the 4.3.22-beta branch started with here as filename, timestamps and files are/seem identical.
If you could share how that "someone" got it to work, I'm sure it would be appreciated :wink:

iamtrying-127times, 2016-10-28:
He did it on Ubuntu 12.10. I've tryed also, but I got a lot of problems using aircrack-ng because it seems uses some dependencies that I cant install it anymore. I will try it in an old version of Ubuntu.
Maybe it is the same archive. I grab it from a forum. The last post reported that worked.
https://forum.aircrack-ng.org/index.php?topic=1086.0

diederikdehaas, 2016-10-28:
I got 4.3.22-beta via that forum thread :wink:

cristi28, 2017-01-25:
take a look here https://github.com/astsam/rtl8812au

osenbruggen, 2017-02-05:
But that one doesn't seem to have Raspberry Pi included?

osenbruggen, 2017-02-06:
Oh got that one working on a Pi with:
make ARCH="arm" CROSS_COMPILE=arm-linux-gnueabihf- KSRC=/usr/src/linux-headers-4.4.45+/
Maybe it helps someone else as well.

capturepi, 2017-02-12:
@osenbruggen What Raspberry Pi OS did you use? I've tried the latest Raspbian image with 4.4.34-v7+ and it would not compile. I've installed the headers using 'sudo apt-get install raspberrypi-kernel-headers' but no luck.

capturepi, 2017-02-12:
I've now managed to get it to compile however it does not appear.
Using 'sudo make install' produces:

install -p -m 644 881au.ko /lib/modules/4.4.38-v7+/kernel/drivers/net/wireless/
/sbin/depmod -a 4.4.38-v7+

But nothing happens when I connect the Alfa card. I've also followed the process on an older PwnPi install using 4.4.19 but the same happens. Astsam works with the WiFi card on Kali VM.

Is there another step to load the driver afterwards?

MrEngman, 2017-02-12:
@capturepi , what is your Alfa device? What is the USB ID. 

Is the ID for your card in the driver source file `os_dep/linux/usb_intf.c`?

capturepi, 2017-02-12:
@MrEngman The Alfa adapter is AWUS036AC. 0bda:8812
I can see the VID and PID in usb_intf.c

MiscellaneousMemphian, 2017-05-16:
can someone tell me which 8812au driver to install that has frame injection and monitor mode work with fluxion? or a driver that i can use to successfully execute WPS cracking using the terminal. its atleast 3 versions i know and i kind of know about compiling drivers but i dont want to do it if it will be no difference than installing another one.