diederikdehaas/rtl8812AU #27: Performance Differences Between Pie and Labtop

danmartinj, 2016-09-08:
Hello,
First of all I would not consider this a true issue but more an inconsistency between the platforms.  I am using an Alpha AWUS036ACH USB WiFi dongle with a realtek 8812 chipset.  My only purpose for using this card is monitoring WiFi associations between designated WiFi stations and WiFi access points.  Long story short I can see fully all associations with the x86 laptop but with the Pie I am only getting a subnet of associations.  It might be worth noting for the Pie I am using an external USB Power Hub.  Other things I have noticed is on the laptop the USB drivers are able to pickup more identification information of the card such as idProduct.  Below is some of that information.  Perhaps you might have an idea of what is going on?

`
lsusb -v
Bus 002 Device 027: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8812 RTL8812AU 802.11a/b/g/n/ac WLAN Adapter  ## laptop 

lsusb -v
Bus 001 Device 019: ID 0bda:8812 Realtek Semiconductor Corp.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8812  ## Pie
`

diederikdehaas, 2016-09-24:
My guess is that the kernel (version) is different and/or the USB system differs. The former seems a more likely explanation.

I did a compare between the `lsusb` output on my laptop and a RPi 1B+:
```diff
$ git diff lsusb-wifi-ac-laptop.txt lsusb-wifi-ac-pi1B+.txt
diff --git a/lsusb-wifi-ac-laptop.txt b/lsusb-wifi-ac-pi1B+.txt
index 6499a85..6e8c8bb 100644
--- a/lsusb-wifi-ac-laptop.txt
+++ b/lsusb-wifi-ac-pi1B+.txt
@@ -1,5 +1,4 @@
-Bus 001 Device 003: ID 7392:a822 Edimax Technology Co., Ltd 
-Couldn't open device, some information will be missing
+Bus 001 Device 004: ID 7392:a822 Edimax Technology Co., Ltd 
 Device Descriptor:
   bLength                18
   bDescriptorType         1
@@ -11,9 +10,9 @@ Device Descriptor:
   idVendor           0x7392 Edimax Technology Co., Ltd
   idProduct          0xa822 
   bcdDevice            0.00
-  iManufacturer           1 
-  iProduct                2 
-  iSerial                 3 
+  iManufacturer           1 Realtek
+  iProduct                2 802.11n NIC
+  iSerial                 3 123456
   bNumConfigurations      1
   Configuration Descriptor:
     bLength                 9
@@ -85,3 +84,15 @@ Device Descriptor:
           Usage Type               Data
         wMaxPacketSize     0x0040  1x 64 bytes
         bInterval               1
+Device Qualifier (for other device speed):
+  bLength                10
+  bDescriptorType         6
+  bcdUSB               2.00
+  bDeviceClass            0 (Defined at Interface level)
+  bDeviceSubClass         0 
+  bDeviceProtocol         0 
+  bMaxPacketSize0        64
+  bNumConfigurations      1
+Device Status:     0x0002                                                                                                               
+  (Bus Powered)                                                                                                                         
+  Remote Wakeup Enabled
```
As you can see is that on the RPi I get **more** output then on my laptop.
My laptop is a brand new model running Debian Sid with kernel 4.7 and my RPi 1B+ is running Jessie.
My PC is not so new, but is also running Debian Sid, but with kernel 4.8-rc5 and it's output is identical to that of my laptop (except Bus/Device number)

In your title you used 'Performance', but don't you mean 'Capabilities'?
It is still very possible that the RPi is slower then your laptop, given the USB speed/throughput of the RPi isn't that great and your laptop may be using (far) more capable/speedy components.

danmartinj, 2016-09-25:
Hey Diederik,

I really appreciate your reply.  You are for the most part right I should of used capabilities as it seems a more appropriate word.  This whole problem I have been working on for a long time and I believe it comes down to a USB speed/throughput issue and at this time I am wondering if as a next step it might be wise for me to try and configure the kernel of the pie.  I noticed it uses primarily simple USB drivers such as OTG.  When my labtop uses EHCI and several other high speed USB kernel modules.  Do you think it would be a good idea to try and get the pie to load and use some of these newer and more capable USB drivers or do you think this whole problem might be a limitation of pie hardware?

diederikdehaas, 2016-09-25:
> do you think this whole problem might be a limitation of pie hardware

Yes. The USB part has always been the weak link of the Pi as it is takes care of both USB2.0 and the networking part and IMO is just not of good quality. It is sufficient for the (official) goal of the Pi, but it lacks for various other uses.
Thus building your own kernel likely won't help.

danmartinj, 2016-10-11:
Diederik,

Thanks for your help again.  I received a ODROID-XU4 a few weeks ago and this device is about the same size as the Pi and is arm based and it has 2xUSB 3.0 and in general a more powerful device.  Anyhow, I am having similar issues with this device as the Pi.  I am starting to wonder if the problem is in the aircrack-ng suite, if it is a driver issue, or again some type of hardware problem.

I even compiled a custom kernel on the device and tried to enable as many kernel modules as I could which are being used on the laptop.

Anyhow, I was wondering for the XU4 do I enable the same config option in the Makefile as the Pi?  I have been using the below option.

`
CONFIG_PLATFORM_ARM_RPI = y
`

Also, when my device loads the drivers, I set the driver name to be wlan12

`
~/alfa3/rtl8812AU-driver-4.3.22-beta# lsmod | grep wlan12
wlan12               1252404  0
usbcore               180393  10 r8152,ehci_exynos,ohci_hcd,ehci_hcd,xhci_plat_hcd,usbnet,wlan12,ohci_exynos,xhci_hcd,cdc_ether
cfg80211              240798  2 mac80211,wlan12
`

In addition this time the lsusb more resembles output from my laptop so this to me is definitely strange.

`
Bus 003 Device 003: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8812 RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
`
Does that all look right to you?  Just out of curiosity do you have any more ideas I could try?

Thanks again for your help.

diederikdehaas, 2016-10-11:
> if it is a driver issue

This is most likely. In the `Makefile` is the following setting `CONFIG_WIFI_MONITOR = n` and that's likely a problem for aircracker-ng. You *may* get some better results when setting it to `y`, but I think the implementation is lacking if not completely absent.
See also https://github.com/diederikdehaas/rtl8812AU/issues/26 and https://github.com/diederikdehaas/rtl8812AU/issues/19

> for the XU4 do I enable the same config option in the Makefile as the Pi?

I *think* that should do it, yes.

> do you have any more ideas I could try?

I have no experience with monitor mode at all. The earlier linked issues may be of help though.