diederikdehaas/rtl8812AU #28: Compiling and running on arm mainline 4.5.4

paulashford1975, 2016-09-09:
Hi, I have cross compiled for arm using our toolchain, everything compiles module installs however on reboot I get the following errors, I can see wlan0 but iw list does not show the device correctly;

[    5.980912] RTL871X: rtl8821au v4.3.14
[    5.986059] RTL871X: rtl8821au BT-Coex version = BTCOEX20150128-51

[    6.087757] =============================================
[    6.093168] [ INFO: possible recursive locking detected ]
[    6.098580] 4.5.4 #1 Tainted: G           O   
[    6.103033] ---------------------------------------------
[    6.108443] systemd-udevd/210 is trying to acquire lock:
[    6.113765]  (&(plock)->rlock){+.....}, at: [<bf1497ec>] rtw_alloc_macid+0x80/0x2c8 [8812au]
[    6.122447] 
               but task is already holding lock:
[    6.128293]  (&(plock)->rlock){+.....}, at: [<bf150198>] rtw_alloc_stainfo+0x30/0x198 [8812au]
[    6.137091] 
               other info that might help us debug this:
[    6.143633]  Possible unsafe locking scenario:

[    6.149564]        CPU0
[    6.152014]        ----
[    6.154462]   lock(&(plock)->rlock);
[    6.158064]   lock(&(plock)->rlock);
[    6.161666] 
                *** DEADLOCK ***

[    6.167598]  May be due to missing lock nesting notation

[    6.174401] 3 locks held by systemd-udevd/210:
[    6.178852]  #0:  (&dev->mutex){......}, at: [<c0397830>] __driver_attach+0x50/0xa0
[    6.186583]  #1:  (&dev->mutex){......}, at: [<c0397840>] __driver_attach+0x60/0xa0
[    6.194309]  #2:  (&(plock)->rlock){+.....}, at: [<bf150198>] rtw_alloc_stainfo+0x30/0x198 [8812au]
[    6.203543] 
               stack backtrace:
[    6.207911] CPU: 0 PID: 210 Comm: systemd-udevd Tainted: G           O    4.5.4 #1
[    6.215497] Hardware name: Marvell Armada 380/385 (Device Tree)
[    6.221428] Backtrace: 
[    6.223896] [<c00142fc>] (dump_backtrace) from [<c00144f8>] (show_stack+0x18/0x1c)
[    6.231481]  r6:600e0093 r5:00000000 r4:c08ff920 r3:dc8ba704
[    6.237200] [<c00144e0>] (show_stack) from [<c02a2fa8>] (dump_stack+0xb8/0xe4)
[    6.244442] [<c02a2ef0>] (dump_stack) from [<c0074b10>] (__lock_acquire+0x161c/0x2150)
[    6.252375]  r8:00000028 r7:eebf3d80 r6:eebf3d40 r5:00000000 r4:c0a8ad50 r3:00000002
[    6.260194] [<c00734f4>] (__lock_acquire) from [<c0075eec>] (lock_acquire+0x78/0x98)
[    6.267953]  r10:f1028004 r9:00000000 r8:f1028208 r7:00000001 r6:00000000 r5:600e0013
[    6.275853]  r4:00000000
[    6.278404] [<c0075e74>] (lock_acquire) from [<c0689974>] (_raw_spin_lock_bh+0x4c/0x5c)
[    6.286424]  r7:f1028004 r6:f0fca000 r5:bf1497ec r4:ee9271f4
[    6.292252] [<c0689928>] (_raw_spin_lock_bh) from [<bf1497ec>] (rtw_alloc_macid+0x80/0x2c8 [8812au])
[    6.301404]  r5:ee927000 r4:ee9271f4
[    6.305230] [<bf14976c>] (rtw_alloc_macid [8812au]) from [<bf1502ec>] (rtw_alloc_stainfo+0x184/0x198 [8812au])
[    6.315253]  r10:f1028004 r9:00000000 r8:f1028afc r7:f1028024 r6:f0fcc1f0 r5:ffffffff
[    6.323154]  r4:f1028afc
[    6.325924] [<bf150168>] (rtw_alloc_stainfo [8812au]) from [<bf1507f0>] (rtw_init_bcmc_stainfo+0x38/0x44 [8812au])
[    6.336295]  r10:eeb76000 r9:eeb76068 r8:eeb76068 r7:ee921800 r6:eeab8000 r5:00000001
[    6.344197]  r4:f0fca000
[    6.346973] [<bf1507b8>] (rtw_init_bcmc_stainfo [8812au]) from [<bf163e74>] (rtw_init_drv_sw+0x128/0x160 [8812au])
[    6.357577] [<bf163d4c>] (rtw_init_drv_sw [8812au]) from [<bf1651b0>] (rtw_usb_if1_init+0x110/0x1e4 [8812au])
[    6.367513]  r5:ee927000 r4:f0fca000
[    6.371348] [<bf1650a0>] (rtw_usb_if1_init [8812au]) from [<bf165480>] (rtw_drv_init+0x1fc/0x314 [8812au])
[    6.381023]  r7:eeab8000 r6:bf209ca0 r5:ee927000 r4:00000006
[    6.386861] [<bf165284>] (rtw_drv_init [8812au]) from [<c042018c>] (usb_probe_interface+0x190/0x278)
[    6.396013]  r10:eeab8020 r9:bf209ca0 r8:eeab8000 r7:bf209860 r6:eeb76000 r5:eeb76068
[    6.403915]  r4:00000000
[    6.406465] [<c041fffc>] (usb_probe_interface) from [<c0397734>] (driver_probe_device+0x21c/0x2c8)
[    6.415443]  r10:c08e74c8 r9:0000001c r8:bf209860 r7:00000000 r6:c116cd1c r5:c116cd14
[    6.423344]  r4:eeab8020
[    6.425894] [<c0397518>] (driver_probe_device) from [<c039787c>] (__driver_attach+0x9c/0xa0)
[    6.434349]  r9:bf262c00 r8:bf201088 r7:00000000 r6:eeab8054 r5:bf209860 r4:eeab8020
[    6.442169] [<c03977e0>] (__driver_attach) from [<c039581c>] (bus_for_each_dev+0x74/0xa8)
[    6.450363]  r6:c03977e0 r5:bf209860 r4:00000000 r3:edb91f5c
[    6.456081] [<c03957a8>] (bus_for_each_dev) from [<c03970a4>] (driver_attach+0x24/0x28)
[    6.464101]  r6:c0912d0c r5:ed8dae80 r4:bf209860
[    6.468761] [<c0397080>] (driver_attach) from [<c0396c50>] (bus_add_driver+0x1a8/0x220)
[    6.476784] [<c0396aa8>] (bus_add_driver) from [<c0398134>] (driver_register+0x80/0x100)
[    6.484891]  r7:bf209860 r6:00000000 r5:c0912d0c r4:bf209860
[    6.490606] [<c03980b4>] (driver_register) from [<c041ecf0>] (usb_register_driver+0x8c/0x150)
[    6.499149]  r5:c0912d0c r4:bf209810
[    6.502863] [<c041ec64>] (usb_register_driver) from [<bf2a109c>] (rtw_drv_entry+0x9c/0xec [8812au])
[    6.511929]  r9:ec49cf00 r8:c08eb990 r7:00000000 r6:bf2a1000 r5:c08eb990 r4:bf209810
[    6.519855] [<bf2a1000>] (rtw_drv_entry [8812au]) from [<c000997c>] (do_one_initcall+0xa8/0x208)
[    6.528660]  r4:bf262c00 r3:00000000
[    6.532266] [<c00098d4>] (do_one_initcall) from [<c00d9a24>] (do_init_module+0x68/0x39c)
[    6.540374]  r10:ec49ce08 r9:13b63228 r8:00000001 r7:bf262c00 r6:ec49cec0 r5:00000001
[    6.548275]  r4:bf262c00
[    6.550828] [<c00d99bc>] (do_init_module) from [<c00b579c>] (load_module+0x18e0/0x1df4)
[    6.558848]  r6:ec49ce00 r5:00000001 r4:ed851f44
[    6.563510] [<c00b3ebc>] (load_module) from [<c00b5e64>] (SyS_finit_module+0x6c/0x7c)
[    6.571356]  r10:00000000 r9:ed850000 r8:c000fec4 r7:0000017b r6:b6e7065d r5:00000007
[    6.579256]  r4:00000000
[    6.581807] [<c00b5df8>] (SyS_finit_module) from [<c000fd20>] (ret_fast_syscall+0x0/0x1c)
[    6.590002]  r6:00000000 r5:00000000 r4:00020000
[    6.597733] RTL871X: rtw_ndev_init(wlan0)
[    6.604359] usbcore: registered new interface driver rtl8821au
[    6.610464] RTL871X: module init ret=0

diederikdehaas, 2016-09-24:
On what ARM system are you trying to run it? What OS is running on it?
What happens if you compile the driver natively on the ARM system?
Where did you get your output from (ie which commands did you use)?

> iw list does not show the device correctly

What do you see? On my amd64 system the output of `iw list` is quite different from what you have posted :wink:

paulashford1975, 2016-10-04:
This is running on mainline 4.5.4 on an Arm cortex a9. It appears that overtime something needs to access the rtl8812au driver that it causes a deadlock on the system. So if I do an ifup it then complains that their is a deadlock so anything talking to the driver reports this in the dmesg. I have to change the makefile to point to our toolchain that is generated using buildroot as the standard arm options are mainly for raspberry pi;

`# Clearfog
ifeq ($(CONFIG_PLATFORM_ARM), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
ARCH := arm
KVER := 4.5.4
CROSS_COMPILE := $(CROSS_COMPILE)
KSRC := $(KDIR)
MODDESTDIR=/usr/lib/modules/$(KVER)/kernel/drivers/net/wireless/
endif`

diederikdehaas, 2016-10-04:
Have you tried whether the `driver-4.3.22-beta` branch works better?

paulashford1975, 2016-10-04:
Let me check

paulashford1975, 2016-10-04:
As their is no standard arm build are my extra_cflag settings correct? I see it now has an ARM64 config option

diederikdehaas, 2016-10-04:
Unfortunately I'm far from a Makefile expert, but I would check and/or change the following:
```
ARCH := arm
KVER := 4.5.4
CROSS_COMPILE := $(CROSS_COMPILE)
KSRC := $(KDIR)
```
Instead of `:=` I'd do `?=`. 
Do you have `CROSS_COMPILE` and `KDIR` defined elsewhere in your `Makefile`? Otherwise they may not have the desired effect.
Regarding `ARCH := arm`, you could try to experiment with different values as it *could* be significant, but I expect it would only provide further optimizations (to optionally do when everything already works properly). 
What I did for `RPI` was just copy the `I386` block and changed the `ARCH` as the Raspberry Pi would otherwise report `armv7l` (iirc) and that causes problems and those were resolved by defining `ARCH := arm`.

paulashford1975, 2016-10-05:
Tried that version and still the same problem. Once compiled and installed I get the following deadlock
`[    6.620077] RTL871X: rtl8821au v4.3.22-beta

[    6.751795] =============================================
[    6.758593] [ INFO: possible recursive locking detected ]
[    6.764005] 4.5.4 #1 Tainted: G           O   
[    6.768715] ---------------------------------------------
[    6.774124] systemd-udevd/205 is trying to acquire lock:
[    6.779446]  (&(plock)->rlock){+.....}, at: [<bf0bdf40>] rtw_alloc_macid+0x80/0x2c8 [8812au]
[    6.788120] 
               but task is already holding lock:
[    6.788253]  (&(plock)->rlock){+.....}, at: [<bf0c4f7c>] rtw_alloc_stainfo+0x30/0x194 [8812au]
[    6.788260] 
               other info that might help us debug this:
[    6.788261]  Possible unsafe locking scenario:

[    6.788262]        CPU0
[    6.788263]        ----
[    6.788265]   lock(&(plock)->rlock);
[    6.788267]   lock(&(plock)->rlock);
[    6.788268] 
                *** DEADLOCK ***

[    6.788269]  May be due to missing lock nesting notation

[    6.788272] 3 locks held by systemd-udevd/205:
[    6.788280]  #0:  (&dev->mutex){......}, at: [<c0397a7c>] __driver_attach+0x50/0xa0
[    6.788286]  #1:  (&dev->mutex){......}, at: [<c0397a8c>] __driver_attach+0x60/0xa0
[    6.788392]  #2:  (&(plock)->rlock){+.....}, at: [<bf0c4f7c>] rtw_alloc_stainfo+0x30/0x194 [8812au]
[    6.788393] 
`

Apparently this is a known issue, so I apply the following patch, which sorts the mutex locking
`diff -ur a/include/osdep_service.h b/include/osdep_service.h
--- a/include/osdep_service.h	2016-06-24 18:06:15.000000000 +0100
+++ b/include/osdep_service.h	2016-09-19 12:15:56.587935209 +0100
@@ -274,9 +274,11 @@
 extern u32	_rtw_down_sema(_sema *sema);
 extern void	_rtw_mutex_init(_mutex *pmutex);
 extern void	_rtw_mutex_free(_mutex *pmutex);
+#ifndef PLATFORM_LINUX
 #ifndef PLATFORM_FREEBSD
 extern void	_rtw_spinlock_init(_lock *plock);
 #endif //PLATFORM_FREEBSD
+#endif //PLATFORM_LINUX
 extern void	_rtw_spinlock_free(_lock *plock);
 extern void	_rtw_spinlock(_lock	*plock);
 extern void	_rtw_spinunlock(_lock	*plock);
diff -ur a/include/osdep_service_linux.h b/include/osdep_service_linux.h
--- a/include/osdep_service_linux.h	2016-06-24 18:06:15.000000000 +0100
+++ b/include/osdep_service_linux.h	2016-09-19 12:15:13.448351681 +0100
@@ -231,6 +231,11 @@
 	spin_unlock_bh(plock);
 }
 
+__inline static void _rtw_spinlock_init(_lock *plock)
+{
+	spin_lock_init(plock);
+}
+
 __inline static int _enter_critical_mutex(_mutex *pmutex, _irqL *pirqL)
 {
 	int ret = 0;
diff -ur a/os_dep/osdep_service.c b/os_dep/osdep_service.c
--- a/os_dep/osdep_service.c	2016-06-24 18:06:15.000000000 +0100
+++ b/os_dep/osdep_service.c	2016-09-19 12:15:27.662214463 +0100
@@ -1138,14 +1138,10 @@
 #endif
 }
 
[... 106 lines cut ...]
[   41.849154]  r10:00000000 r9:edc31b98 r8:00000000 r7:00000000 r6:ec672500 r5:ed75c800
[   41.857053]  r4:00000000
[   41.859603] [<c0542158>] (rtnl_newlink) from [<c054086c>] (rtnetlink_rcv_msg+0x188/0x228)
[   41.867798]  r10:ec66ce40 r9:00000000 r8:00000000 r7:00000000 r6:ec66ce40 r5:ec672500
[   41.875697]  r4:00000000
[   41.878250] [<c05406e4>] (rtnetlink_rcv_msg) from [<c055d7a8>] (netlink_rcv_skb+0xa8/0xc4)
[   41.886531]  r8:00000000 r7:ec66ce40 r6:ec66ce40 r5:c05406e4 r4:ec672500
[   41.893303] [<c055d700>] (netlink_rcv_skb) from [<c05406dc>] (rtnetlink_rcv+0x30/0x38)
[   41.901237]  r6:ec44f000 r5:00000020 r4:ec66ce40 r3:0000027d
[   41.906954] [<c05406ac>] (rtnetlink_rcv) from [<c055d10c>] (netlink_unicast+0x16c/0x1f4)
[   41.915061]  r4:ef10c000 r3:c05406ac
[   41.918667] [<c055cfa0>] (netlink_unicast) from [<c055d5c4>] (netlink_sendmsg+0x370/0x384)
[   41.926948]  r8:00000000 r7:00000020 r6:00000000 r5:ec44f000 r4:edc31f4c
[   41.933720] [<c055d254>] (netlink_sendmsg) from [<c050d96c>] (sock_sendmsg+0x1c/0x2c)
[   41.941567]  r10:edc31e64 r9:00000000 r8:eef618c0 r7:00000000 r6:00000000 r5:c08e74c8
[   41.949467]  r4:edc31f4c
[   41.952017] [<c050d950>] (sock_sendmsg) from [<c050e530>] (___sys_sendmsg+0x21c/0x228)
[   41.959954] [<c050e314>] (___sys_sendmsg) from [<c050f3c0>] (__sys_sendmsg+0x44/0x74)
[   41.967800]  r10:00000000 r9:edc30000 r8:c000fec4 r7:00000128 r6:bea66624 r5:00000000
[   41.975700]  r4:eef618c0
[   41.978251] [<c050f37c>] (__sys_sendmsg) from [<c050f400>] (SyS_sendmsg+0x10/0x14)
[   41.985836]  r6:00000000 r5:bea6e718 r4:bea66640
[   41.990499] [<c050f3f0>] (SyS_sendmsg) from [<c000fd20>] (ret_fast_syscall+0x0/0x1c)
[   42.226402] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
`

diederikdehaas, 2016-10-05:
> Apparently this is a known issue

Link?

paulashford1975, 2016-10-05:
Will need to dig out the link stubble over it, apparently realtek were aware of the lock issue but left it in place until they decided to fix it. The diff patch sorts that error out.

diederikdehaas, 2016-10-05:
Aside: use 1 backtick (\`) for inline quotes and 3 backticks for block quotes
I wasn't able to properly read it until I edited your post.