diederikdehaas/rtl8812AU #29: Found rtl8821au 4.3.19 version

aldegoeij, 2016-09-14:
Bough an Edimax AC600 (EW-7811DAC) 2.4 and 5 Ghz dongle which comes with a rtl821au 4.3.19 stamped driver here: http://www.edimax.nl/edimax/download/download/data/edimax/nl/download/for_home/wireless_adapters/wireless_adapters_ac600_dual-band/ew-7811dac

Is this useful to add to your repo? (knowing you are already working on 22-beta)

diederikdehaas, 2016-09-15:
> Is this useful to add to your repo? (knowing you are already working on 22-beta)

It would, were it not that I had recently found the [4.3.20 driver](http://www.edimax.nl/edimax/download/download/data/edimax/nl/download/for_home/wireless_adapters/wireless_adapters_ac1200_dual-band/ew-7822uac) and I will be adding that to this repo.
Thanks for sharing :+1:

uminokoe, 2016-09-15:
In the meantime, please note that additional changes are needed for newer kernels. The download page states that the supported kernel versions are 2.6.18 - 4.2. However, the great thing is that the documentation in the .zip file can be quite useful.

diederikdehaas, 2016-09-15:
Yep, it's quite likely I'll re-apply various changes we've done in the repo to the 4.3.20 driver, but that one does seem to be a good candidate to be promoted as default. Despite the (slightly) lower version number than 4.3.22-beta, the code does seem to be more recent.

uminokoe, 2016-09-16:
Grawp's repository now contains the 4.3.20 version of the driver (https://github.com/Grawp/rtl8812au_rtl8821au) and made changes to support kernel version >=4.7. It seems to me that, although 4.3.20 fixes some issues, some codes present in 4.3.22-beta have yet to be included in 4.3.20. I hope the 4.3.22-beta branch will be kept.

diederikdehaas, 2016-09-16:
> I hope the 4.3.22-beta branch will be kept.

It will. But I will change the **default** branch from 4.3.14 to 4.3.20 :wink:

uminokoe, 2016-09-24:
Using the 4.3.21 driver from EDIMAX (which is originally designed for RTL8814AU), I modified it and it can then be used for RTL8811AU / RTL8812AU / RTL8821AU. You may view the code here: [https://github.com/uminokoe/rtl8812AU/tree/driver-4.3.21](https://github.com/uminokoe/rtl8812AU/tree/driver-4.3.21)
However, please note that it is experimental and I cannot guarantee that everything works. Should you find any bugs, please do not hesitate to submit a pull request or an issue.

diederikdehaas, 2016-10-05:
> It seems to me that, although 4.3.20 fixes some issues, some codes present in 4.3.22-beta have yet to be included in 4.3.20

Indeed.
I'm working on the 4.3.20 driver and your PRs regarding VHT make it fail as apparently that part of the code didn't make it into 4.3.20, while it is present in 4.3.22-beta and I just saw also in your 4.3.21 driver. :thinking:

diederikdehaas, 2016-10-05:
I've uploaded my **WIP** branch here: https://github.com/diederikdehaas/rtl8812AU/tree/driver-4.3.20-add-vht-capabilities-wip

fedux, 2016-10-09:
I found what seems to be the original ZIP for the version @uminokoe is taling about here: http://www.edimax.co.uk/edimax/download/download/data/edimax/uk/download/for_home/wireless_adapters/wireless_adapters_ac1750_dual-band/ew-7833uac

uminokoe, 2016-10-10:
Thank you @fedux .
@diederikdehaas , I was not aware of your updated comment and I apologize for that. Indeed, I downloaded the driver for EW7833UAC (RTL8814AU chipset), unzipped that and converted all line endings to LF. Then the whole folder was uploaded and became the Initial Import.

diederikdehaas, 2016-10-10:
No worries :+1:

weedy, 2016-10-11:
@uminokoe does your wifi stick work with your driver-4.3.21 branch? 
Are you using your driver from everyday?

uminokoe, 2016-10-12:
@weedy 
Yes, I do use it every day. See [this comment](https://github.com/diederikdehaas/rtl8812AU/issues/10#issuecomment-253115897) for the full reply.

diederikdehaas, 2016-10-16:
Just as I thought I could close this issue because I uploaded the 4.3.20 driver, I noticed that the 4.3.19 driver is for the rtl88**21**au and not for the rtl88**12**au. *sigh*

I need a break.

Goga777, 2017-01-29:
> Bough an Edimax AC600 (EW-7811DAC) 2.4 and 5 Ghz dongle which comes with a rtl821au 4.3.19 >stamped driver 

with which drivers could you run the ew-7811dac which based on 8811au chip ?

uminokoe, 2017-02-02:
You may wish to download the driver for EW-7811DAC from Edimax's site, but the driver may need modification to work on newer kernels. Alternatively, try [diederikdehaas' 4.3.22-beta version](https://github.com/diederikdehaas/rtl8812AU/tree/driver-4.3.22-beta) and it should work well with the RTL8811AU chip.