diederikdehaas/rtl8812AU #34: Can't compile driver

lkw16, 2016-09-27:
I am using the 4.3.14 branch to compile the driver.  I got the headers from RP repository using this command

# apt-get install raspberrypi-kernel-headers

Everything seemed to be working OK except when I tried to dkms build.  I received error message.  Here's make.log file.

DKMS make.log for rtl8812AU-4.3.14 for kernel 4.4.13-v7+ (armv7l)
Tue Sep 27 10:53:39 PDT 2016
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.4.13-v7+/build M=/var/lib/dkms/rtl8812AU/4.3.14/b$
make[1]: Entering directory '/usr/src/linux-headers-4.4.13-v7+'
Makefile:606: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.4.13-v7+'
Makefile:1576: recipe for target 'modules' failed
make: *** [modules] Error 2

It looks like it is looking for armv7l directory but the headers just has arm and arm64 folders.  It doesn't have the armv7l folder there under arch.  The same problem with the 4.3.22-beta branch as well.  Am I using the wrong headers?  Any help would be appreciated!  Thanks!

Regards,
KWL

uminokoe, 2016-09-28:
Could you please run `uname -a` and post the output here? Knowing the architecture of the OS may be of help.

diederikdehaas, 2016-09-28:
If you do these steps, what do you get then?

- Follow the procedure in the README to remove the dkms stuff again. 
- Change the `Makefile` and [set `CONFIG_PLATFORM_I386_PC` to `n` and `CONFIG_PLATFORM_ARM_RPI` to `y`](https://github.com/diederikdehaas/rtl8812AU/blob/driver-4.3.14/Makefile#L86)
- Install the dkms thing again.

lkw16, 2016-09-28:
Here's the "uname -a" output.
4.4.21-v7+

Originally, it was 4.4.13-v7+ but I just did an dist-upgrade.  Right now, it is the above version.

After made the changes suggested by diederikdehaas, it works OK now.  I should have checked that in the first place before posted here.  Thanks!

closed 2016-09-28 by lkw16