diederikdehaas/rtl8812AU #35: Some channels are disabled regardless of the regulatory domain

uminokoe, 2016-09-29:
I have my regulatory domain set to a country. This is what I get from `iw reg get`:
```
country <Country_code_here>: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
```
For the same country, the matching record in `db.txt` from [https://www.kernel.org/pub/software/network/wireless-regdb/](https://www.kernel.org/pub/software/network/wireless-regdb/) is shown below:
```
country <Country_code_here>: DFS-ETSI
	(2402 - 2482 @ 40), (20)
	(5170 - 5250 @ 80), (17), AUTO-BW
	(5250 - 5330 @ 80), (24), DFS, AUTO-BW
	(5490 - 5710 @ 160), (24), DFS
	(5735 - 5835 @ 80), (30)
```
However, when I run `iw list`, the output shows that some channels are disabled. Some channels have the flag "(no IR)", so those channels cannot be used, too.
```
		Frequencies:
			* 5170 MHz [34] (disabled)
			* 5180 MHz [36] (17.0 dBm)
			* 5190 MHz [38] (disabled)
			* 5200 MHz [40] (17.0 dBm)
			* 5210 MHz [42] (disabled)
			* 5220 MHz [44] (17.0 dBm)
			* 5230 MHz [46] (disabled)
			* 5240 MHz [48] (17.0 dBm)
			* 5260 MHz [52] (24.0 dBm) (no IR)
			* 5280 MHz [56] (24.0 dBm) (no IR)
			* 5300 MHz [60] (24.0 dBm) (no IR)
			* 5320 MHz [64] (24.0 dBm) (no IR)
			* 5500 MHz [100] (24.0 dBm) (no IR)
			* 5520 MHz [104] (24.0 dBm) (no IR)
			* 5540 MHz [108] (24.0 dBm) (no IR)
			* 5560 MHz [112] (24.0 dBm) (no IR)
			* 5580 MHz [116] (24.0 dBm) (no IR)
			* 5600 MHz [120] (24.0 dBm) (no IR)
			* 5620 MHz [124] (24.0 dBm) (no IR)
			* 5640 MHz [128] (24.0 dBm) (no IR)
			* 5660 MHz [132] (24.0 dBm) (no IR)
			* 5680 MHz [136] (24.0 dBm) (no IR)
			* 5700 MHz [140] (24.0 dBm) (no IR)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5920 MHz [184] (disabled)
			* 5940 MHz [188] (disabled)
			* 5960 MHz [192] (disabled)
			* 5980 MHz [196] (disabled)
			* 6000 MHz [200] (disabled)
			* 6020 MHz [204] (disabled)
			* 6040 MHz [208] (disabled)
			* 6060 MHz [212] (disabled)
			* 6080 MHz [216] (disabled)
```
In addition, if I interpreted the regulatory domain regulation (records in `db.txt`) correctly, channels 12 and 13 in 2.4GHz should be available for use. However, this is what I get from `iw list`:
```
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (disabled)
			* 2472 MHz [13] (disabled)
			* 2484 MHz [14] (disabled)
```
As a result, if the adapter attempts to use the disabled / no-IR channels (no matter in station mode or AP mode), the operation will fail.

diederikdehaas, 2016-09-30:
> I have my regulatory domain set to a country. 

How/where can/should you do that? 

> This is what I get from iw reg get

Do you happen to know some good tutorials on the use of `iw`? I'm actually pretty clueless about it and the help info is quite daunting (to me) with loads of things in it that don't ring a bell and the `man` page seems to be pretty useless.
I've been having some issues (maybe related to `wpa_supplicant`) and getting (and/or setting) the right info with the right tools (`iw` and `ip`) seems like something I should learn.
Things work fine on my PC which uses a wired network connection, but the issues are on my laptop which has a wireless device and plugging in my rtl8812AU device doesn't make things easier :wink:

uminokoe, 2016-10-01:
Using version 4.7 of `iw`, the command `iw help` should help you. (Yeah, `man iw` is not really helpful and I think the sentence "`iw help` will print all supported commands" should be highlighted.)

I have tried "analyzing" the code, and it seems that Realtek is trying to apply the channel flags according to the regulatory domain (here, "regulatory domain" means FCC, ETSI, etc. In the `db.txt`, you will see things like "DFS-ETSI", "DFS-FCC", "DFS-JP".) I guess that the "regulatory domain" here is just some rough "standards" of the allowed/disallowed channels. Just like in the world of Linux distributions, Fedora, Debian, openSUSE, etc. are the major distributions, but at the same time there are many derivatives (e.g. Ubuntu is Debian-based). Therefore this regulatory domain issue may only affect those countries which have varied sets of rules.

olivluca, 2016-10-12:
I'm not sure that the driver reacts correctly to iw reg set, but it has an "iwpriv channel_plan" command (see include/rtw_mlme_ext.h and core/rtw_mlme_ext.c) though I'm not sure it does the "right" thing either.

uminokoe, 2016-10-12:
@olivluca 
Thank you for your interest in this issue. I just did a quick search in the files you mentioned, but no pieces of useful information were returned. Lines which contain `iwpriv` are not relevant to the command or "channel_plan", and lines with `channel_plan` seem to be only some variables used in the driver. Could you please provide some more information? Thank you.
P.S. In my case, the country code used in `iw reg set <country_code>` seems to be effective and `iw reg get` will return the corresponding country code and data. Since I am by no means a developer/programmer/professional, I am not able to understand the "internal operation" of the driver.

olivluca, 2016-10-12:
If you look at os_dep/linux/ioctl_linux.c you'll see that "iwpriv channel_plan" is implemented by rtw_wx_set_channel_plan which in turn calls rtw_set_chplan_cmd.
This is in core/rtw_cmd.c and it checks that the parameter is valid by using the macro rtw_is_channel_plan_valid, defined in include/rtw_mlme_ext.h as
`#define rtw_is_channel_plan_valid(chplan) (chplan<RT_CHANNEL_DOMAIN_MAX || chplan == RT_CHANNEL_DOMAIN_REALTEK_DEFINE)`

RT_CHANNEL_DOMAIN_MAX is defined in include/rtw_mlme_ext.h, where you can see the valid values for the "iwpriv channel_plan" parameter while the definitions of the various plans are in core/rtw_mlme_ext.c

I investigated because I see with iw list that various channels are disabled while they shouldn't in my regdomain (ES). Using the iwpriv I see some of them enabled (at least for x seconds, I suppose for DFS but I'm not an expert on wifi issues).
BTW, I'm currently using your repository (the 4.3.14 branch, the 4.3.21 doesn't work with my adapter) but there should be no significant difference wrt regdomains to  diederikdehaas' one

olivluca, 2016-10-12:
Oh, and I investigated just because I couldn't see my access point, which had the channel set to auto and it selected one of the disabled one.
I just set it manually and now I can connect, but I wonder why the driver doesn't allow me to use channels that are available here.

uminokoe, 2016-10-12:
@olivluca 
First of all, if you wish to use version 4.3.21, you need to make sure that your adapter's vendor ID and product ID is listed in os_dep/linux/usb_intf.c under the corresponding `#ifdef` block of your chipset. If not, add the ID yourself and compile again. In addition, if you are using it for RTL8814AU, you will need to revert two relevant commits to change the targeted chipset.
Regarding the regulatory domain issue, the codes you mentioned have been changed in newer versions (such as in version 4.3.21), but the problem of some channels being disabled still exists. If you absolutely need to use the channels, you can try commenting out `_rtw_regd_init_wiphy(NULL, wiphy);` (around line 543) in os_dep/linux/wifi_regd.c. This is not a good way to actually fix the problem, however.

olivluca, 2016-10-12:
@uminokoe yes, it is listed (0bda:881a)  but somehow it fails during initialization and it enters in a loop, but I didn't investigate too much. I don't absolutely need to see all channels but it bothers me that it doesn't use all legally available channels.
The change I see in 4.3.21 is that RT_CHANNEL_DOMAIN* is now RTW_CHPLAN* but the rest seems to be the same (just had a cursory glance).

olivluca, 2016-10-12:
I commented out that line, iw list shows all channels up to 140 as available (52-140 with radar detection).
Put the AP in auto mode, it selected channel 104 but the wifi stick cannot see it (and it is there: my mobile connects to it just fine). I tried channel 52 (also with radar detection) and it doesn't work either.

uminokoe, 2016-10-13:
This may be the problem of radar detection. I remember that when I commented out that line, I could use the channels which has no "DFS" or "radar detection". For example, referring to the first post, channels 34 to 48 can be used (with the label "auto-bw", and this label does not affect the normal usage), but are disabled by the driver. Then when the line is commented out, channels 34 to 48 can be used. I always avoid (and I have not tried) DFS/radar detection channels for less problems.
At this moment, we can only accept the "fate" and use the limited channels, try to fix the problem, or give up using Linux and switch to probably the most widely used operating system.

CableCatDK, 2017-02-21:
I could not get 5Ghz to work. After reading this post I change the channel to 36, and it now works. Thanks.