diederikdehaas/rtl8812AU #36: modules unstable

yborges, 2016-10-07:
Hallo Diederick! Alvast heel erg bedankt voor de tijd die je in dit project steekt!

I have bought a rapsberry pi 3 and an alfa AWUS036AC. It has chipset 8812au.

On the raspberry pi 3 I have the latest version of the [raspbian image(2016-09-23) from raspberrypi.org.](https://www.raspberrypi.org/downloads/raspbian/)

[From the release notes:](http://downloads.raspberrypi.org/raspbian/release_notes.txt)
> Updated firmware and kernel (https://github.com/raspberrypi/firmware/commit/ad8608c08b122b2c228dba0ff5070d6e9519faf5)

This [firmware points to](https://github.com/raspberrypi/firmware/blob/ad8608c08b122b2c228dba0ff5070d6e9519faf5/extra/git_hash) commit [2d31cd571e6f2ac61a3eda273d973058e1bb1f58 of raspbian kernel source on githut ](https://github.com/raspberrypi/linux/commits/2d31cd571e6f2ac61a3eda273d973058e1bb1f58)

uname -a gives:
Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux

I have used .config from "modprobe configs", and Module7.sysvers as Module.sysvers to "make modules_prepare" and I changed the target to the ARM architecture in the Makefile of the module before executing "make" for the module. Furthermore I compiled the modules on debian x64 using the tools also available on the GitHub account of raspberrypi.

I tried first to compile the module from gnab. I'm not sure anymore where I compiled it, but at the end the compilation went well and the driver was the most stable. The problem is that it doesn't see the network I'm interested in. In fact is sees much few networks compared to when I'm using the device with your drivers. I tried both version 4.3.14 and version 4.3.22-beta of your drivers. I'm not very acquainted with module development or wireless in linux so now I'm speaking like a user: when I turn on the RP3 half of the times it will connect to the network I'm interested in - which is very far - otherwise the device isn't available not even for a "iw wlan0 scan" from the command line. So I have to restart the RP3 again and hope it works. 

I'm not sure what's wrong. For version 4.3.14 I got this message:

> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.854649] Internal error: Oops: 7 [#1] SMP ARM
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855067] Process RTW_CMD_THREAD (pid: 525, stack limit = 0xb6be2210)
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855073] Stack: (0xb6be3ed8 to 0xb6be4000)
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855080] 3ec0:                                                       805ba2a4 800d96e0
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855088] 3ee0: af1ec49c af1ec480 b6be3f14 af1ec480 bbb49000 bbb4a958 bbb4a978 0000c725
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855097] 3f00: bbb4a958 000048a4 b6be3f5c b6be3f18 7f140b5c 7f15ef90 b6be3f3c b6be2000
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855105] 3f20: b924e740 bbb4a968 bbb4a9a4 b8b50200 00000000 00000000 b924e740 bbb49000
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855113] 3f40: 7f1408cc 00000000 00000000 00000000 b6be3fac b6be3f60 80042754 7f1408d8
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855122] 3f60: 3972f000 00000000 b6be3f94 bbb49000 00000000 00000000 b6be3f78 b6be3f78
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855130] 3f80: 00000000 00000000 b6be3f88 b6be3f88 b924e740 80042668 00000000 00000000
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855138] 3fa0: 00000000 b6be3fb0 8000fbc8 80042674 00000000 00000000 00000000 00000000
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855146] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855154] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000 3a7fa861 3a7fac61
> 
> Message from syslogd@raspberrypi at Oct  2 20:32:20 ...
>  kernel:[  809.855904] Code: e0888005 e59f9208 e2853078 e0876003 (e7d73003) 

I'm very much willing to test whatever you tell me to test and give you the results.

diederikdehaas, 2016-10-16:
Hoi @yborges 

Did you happen to use the [instruction I wrote a while back on my 'cknow.org' repo](https://github.com/diederikdehaas/cknow.org/blob/master/rpi/compile-kernel-module-on-raspberrypi.md)?
If so, that shouldn't be needed anymore as there is **finally** a `kernel-headers` package and you should install and use that one.
After you've installed that, clone this repo (if you haven't already), `cd` into the folder and make the following change to the `Makefile`:
```
CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_ARM_RPI = n
```
to
```
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = y
```
And then it *should* be as simple as running `make` and (as root or with sudo) `make install`.

ThijsWithaar, 2016-12-22:
with current raspbian, it's kernel-headers and dkms to handle the module, I also ran into a similar kernel oops (same thread).
Any hints on where to look, how to dig?

ThijsWithaar, 2017-02-19:
Just as a few others, the driver works fine if cross-compiled from x86 to rpi3:
rpi3: gcc version 4.9.2 (Raspbian 4.9.2-10)
x86: gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12)

Details are on:
https://www.raspberrypi.org/forums/viewtopic.php?p=706261
I did also disable powersaving, since there are some claims that that's unstable.

Note: I did need to copy usr/src/linux/Module.symvers to the x86 to get a module which insmod wanted to load.