diederikdehaas/rtl8812AU #37: 4.7.0-kali-amd64

DenisSlobodyanyuk, 2016-10-12:
I am having issues installing it on Kali Rolling 2016.2 kernel:4.7.0-kali-amd64
I am new to Kali and trying to get my Linksys WUSB6300 to work. Currently I have it working sometimes but I get:

wlan1 unassociated Nickname:"<WIFI@REALTEK>"
Mode:Managed Frequency=2.412 GHz Access Point: Not-Associated 
Sensitivity:0/0 
Retryff RTS thrff Fragment thrff
Encryption keyff
Power Managementff
Link Quality:0 Signal level:0 Noise level:0
Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0
Tx excessive retries:0 Invalid misc:0 Missed beacon:0

When I get to 
"git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}" 
I get the error 
"fatal: Not a git repository"
"tar: This does not look like a tar archive"
"tar: Exiting with the failure status due to previous errors"
and when I use "cd rtl8812AU" and then "git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}" 
I get the error
"fatal: Not a valid object name"
"tar: This does not look like a tar archive"
"tar: Exiting with the failure status due to previous errors"

diederikdehaas, 2016-10-15:
Did you define `DRV_NAME` and `DRV_VERSION` before you ran the `git archive ...` command?
Also make sure you run those commands as root and if you use `sudo`, place the `sude` before the `tar -x` part as writing to `/usr/src/` isn't (normally) allowed by a normal user.
And you do indeed need to run those commands inside the `rtl8812AU` directory (or whichever directory you cloned the repo into).

DenisSlobodyanyuk, 2016-10-17:
Ran:

DRV_NAME=rtl8812AU
DRV_VERSION=4.3.14
sudo mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
git archive driver-${DRV_VERSION} | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}

still get 

"fatal: Not a valid object name"
"tar: This does not look like a tar archive"
"tar: Exiting with the failure status due to previous errors"

fumantsu, 2016-10-17:
you need to run "git archive driver-${DRV_VERSION} | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}" inside the directory that you have decompress the files from git. If you have the files on "/tmp/rtl8812AU" then you need to be inside before run the command above.
diederikdehaas is say this on his last sentence. The instructions needs to clarify this as it is not clear.

diederikdehaas, 2016-10-17:
what happens if you make the first part of the last command you used into the following:
```
git archive --format=tar driver-${DRV_VERSION}
```

DenisSlobodyanyuk, 2016-10-17:
Sorry do I actually ran:
Ran:

cd rtl8812AU/
DRV_NAME=rtl8812AU
DRV_VERSION=4.7.0
sudo mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
git archive driver-${DRV_VERSION} | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}

still get

"fatal: Not a valid object name"
"tar: This does not look like a tar archive"
"tar: Exiting with the failure status due to previous errors"

diederikdehaas, 2016-10-17:
> DRV_VERSION=4.7.0

That's your error. There is no `driver-4.7.0` branch.

fumantsu, 2016-10-17:
Second comment ->DRV_VERSION=4.3.14, last comment -> DRV_VERSION=4.7.0. ?????????????????

DenisSlobodyanyuk, 2016-10-17:
I thought the version was supposed to be kernel version. using 4.3.14 worked on installing but still get the plan as unassociated