diederikdehaas/rtl8812AU #40: KernelOops with ubuntu 14.04

neveldo, 2016-10-24:
Hello,

First, thanks a lot for releasing this driver, it helped me a lot to get my netgear AC600 (Realtek RTL8811AU) work !

The wireless connection works very fine. However, I get some kernelOops errors on startup (ubuntu 14.04, kernel 3.13.0-98-generic). I tried both v4.3.14 and 4.3.22-beta and get the same result :+1: 

BUG : unable to handle kernel paging request at (...).

Do you know if there is a way to fix it ?

Thank a lot again for this helpful driver !

diederikdehaas, 2016-10-24:
> I get some kernelOops errors on startup

But it doesn't crash the kernel? Kernel oops are normally quite serious and often fatal.
Are you sure it's related to this kernel driver/module?
Anyway, you should surely report this to the ubuntu kernel team as it is at the very least a (serious) indication of a problem somewhere in the kernel, which this module appears to trigger.

neveldo, 2016-10-25:
Thanks for your answer. It doesn't crash the kernel, everything seems to worke fine.  I will report the errors, maybe I will try to upgrade to unbuntu 16.