diederikdehaas/rtl8812AU #41: [info] Lag spikes

Id2ndR, 2016-10-30:
I'm not sure if it is a driver issue or and hardware one, but it seems that lots of people suffer from lag spikes on 5Ghz band only (2,4Ghz is fine). Using AC means using 5Ghz band, and improve lots of things.

After a long while with this issue, I try now to summarize causes : 
- Network Manager Bug: this is an old one: every 2 minutes, there was a lag due to scanning. This is know [fixed] (https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/24) 
- Channel Width: it seems that stability is affect by channel width (20Mhz, 40Mhz, 80Mhz, 160Mhz). (http://us.battle.net/forums/en/wow/topic/14278688390).

Whereas using an Intel AC adaptor integrated to my laptop give me no trouble with 80Mhz width, the connexion is not stable with my usb adaptor TP-Link T4U at 3 meters from the router. I changed the TP-Link Archer C8 channel width from auto to 40Mhz, and now it's stable.

If someone know how to set channel width at driver level (8812au), I'll try so my router will use 80Mhz width with other AC adaptors.

diederikdehaas, 2016-10-31:
> people suffer from lag spikes

How can I/one measure that?
Also, what's your DeviceID (`lsusb` should tell you that)

> Create `/etc/modprobe.d/8812au.conf` with content `options 8812au rtw_bw_mode=0x11`

Nice :+1: 

> If someone know how to set channel width at driver level (8812au)

Didn't you do just that with the contents of `/etc/modprobe.d/8812au.conf`?
Whether that could work is probably related to the driver (version) you're using. 
In the driver-4.3.22-beta branch @uminokoe added VHT capabilities and that is likely needed. Search the (open and closed) issues/pull requests for VHT and you should find some hints.
In one of those issues/PRs you should also find info on [uminokoe's 4.3.21 driver](https://github.com/uminokoe/rtl8812AU/tree/driver-4.3.21) and that may be worthwhile to test too.

HTH

Id2ndR, 2016-10-31:
To measure, I just continuously ping an other equipment on the Lan (the router itself is the simplest).
When there are lag spikes, I get between 2 and 8 ping responses at the same time. This means that the last one is somewhere between 1-1000 ms, and the others cumulates about 1s. For example: 450ms, 1458ms, 2443ms, 3451ms. This used to happened once in less that 2 minutes

But today I cannot reproduce the lags anymore (with no modprobe option).
I changed Channel width on my router, so maybe there was an impact at on the router level... I don't know.

So uminokoe's 4.3.21 driver give me good results today, but your driver too!
I will add a comment it the trouble appears again later.

uminokoe, 2016-11-02:
I have to thank @Id2ndR because one of the links he posted helped me solve the "mystery" of [this commit](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/commit/2e83034d8723442181749f2781fc9fb411380c15). I found [this](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/24#issuecomment-61385146) and followed the link to [https://github.com/austinmarton/rtl8812au_linux/issues/2](https://github.com/austinmarton/rtl8812au_linux/issues/2). You can click the link to see the full details.
ddrozdov mentioned that 
> Every 2-3 minutes I have a strange lag for some seconds.

and his fix is
> Setting SURVEY_TO to 0 was enough to reduce the lag.

If you have the same problem as what ddrozdov described, maybe you could try his fix.

weedy, 2016-11-05:
If you run the driver with the stock full debugging turned on you'll see that it's doing full spectrum scans on both bands when you feel lag. 
I feel like this is just the current driver being shit. NM has no problems partial scanning on my ath9k card but realtek just goes full blast on scanning EVERYTHING. And realtek is flipping slow to switch bands.

bootsy52, 2016-11-08:
I found that on the Raspberry PI and Rasbian the issue is related to the WLAN lxpanel Applet dhcpcdui which starts scans every 20 - 30  seconds causing lags about 5 sec. I've also disabled backgroundscanning in /etc/wpa_supplicant/wpa_supplicant.conf by adding the parameter bgscan="" to the global section. After killing the WLAN Applet the lags are gone. Further more the following module parameters could be of interest

rtw_network_mode, valid values =  0,1,2,3,4,5

where 

0 = Ndis802_11IBSS
1 = Ndis802_11Infrastructure
2 = Ndis802_11AutoUnknown
3 = Ndis802_11InfrastructureMax
4 = Ndis802_11APMode
5 = Ndis802_11Monitor

this parameter defaults to Ndis802_11IBSS. 

rtw_scan_mode, valid values 0 = passive wait for beacons and 1 = active sent out probes, defaults to 1 = active.

I've made a patch that this parameter can be set in modprobe.conf (normally this is not configurable for whatever reason).

############# CUT HERE ###############
--- os_dep/linux/os_intfs.c.orig	2016-11-08 16:36:10.975976304 +0100
+++ os_dep/linux/os_intfs.c	2016-11-08 16:35:56.606009927 +0100
@@ -50,6 +50,8 @@
 int rtw_frag_thresh = 2346;//*
 int rtw_preamble = PREAMBLE_LONG;//long, short, auto
 int rtw_scan_mode = 1;//active, passive
+module_param(rtw_ips_mode, int, 0644);
+MODULE_PARM_DESC(rtw_scan_mode,"The default scan mode, 1 = active, 0 = passive");
 int rtw_adhoc_tx_pwr = 1;
 int rtw_soft_ap = 0;
 //int smart_ps = 1;
############### END CUT ####################

diederikdehaas, 2016-11-08:
nice :+1:

bootsy52, 2016-11-08:
I'm tested the 4.3.20 driver now for quite some time with mtr the results are, when disabling the WLAN Applet

3200 Packets, AVG = 2,1 Best= 1,2 Worst = 28 StdDeviation= 1,8 on a 2,4 Ghz link with 144 Mbit, 0,0% loss

nothing else has to be changed (what is mentioned at the top in the source code to get these results) Neither have i changed SURVEY_TO or some Thresholds. This is on a RPI3 with Kernel 4.4.26-v7+. I have to say I'm satisfied, compared to the results I've got before.

trustin, 2021-01-07:
> Create `/etc/modprobe.d/8812au.conf` with content `options 8812au rtw_bw_mode=0x11`

This worked for me. Using ipTIME A2000UA USB 3.0 WiFi Adapter with Realtek 8812AU. Thanks a lot, @Id2ndR! :bow: