diederikdehaas/rtl8812AU #45: DKMS is not recreate the module

fumantsu, 2016-11-30:
Hi,

I have notice twice since a fresh install that every time that I have a new kernel, DKMS is removing the module but it doesn't recreate it
run-parts: executing /etc/kernel/prerm.d/dkms 4.4.32+ /boot/kernel.img
run-parts: executing /etc/kernel/prerm.d/dkms 4.4.32-v7+ /boot/kernel7.img
/etc/kernel/prerm.d/dkms:
dkms: removing: rtl8812AU 4.3.22-beta (4.4.32-v7+) (armv7l)

-------- Uninstall Beginning --------
Module:  rtl8812AU
Version: 4.3.22-beta
Kernel:  4.4.32-v7+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

8812au.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.4.32-v7+/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.....

DKMS: uninstall completed.

------------------------------
Deleting module version: 4.3.22-beta
completely from the DKMS tree.
------------------------------
Done.

dkms status is empty.
So as result, I need to do every time all the steps to add the module to DKMS, compile and load it.

jeremyb31, 2017-02-07:
Hi fumantsu,
After a new kernel is installed check 
modinfo 8812au
And see if the filename kernel version and the vermagic version match, if they don't
See if my fork with an updated dkms.conf file works for you
https://github.com/jeremyb31/rtl8812AU

fumantsu, 2017-02-07:
I will check it as soon a new kernel version is out. From my experience I try to avoid to do the major RPI firmware upgrade as I need to find the package or the tarball for the kernel headers. This sometimes is pain in the ass so I prefer official raspbian kernel updates

cooperarichardson, 2017-02-19:
The same issue happens on Kali (based on debian testing) when it updated from 4.8 to 4.9. Probably happens on any other debian-based distro.

fumantsu, 2017-03-01:
Ok, today I update the kernel. These are the logs.
Kernel before update: 
4.4.38-v7+
During upgrade:
run-parts: executing /etc/kernel/prerm.d/dkms 4.4.38+ /boot/kernel.img
/etc/kernel/prerm.d/dkms:
dkms: removing: rtl8812AU 4.3.22-beta (4.4.38+) (armv7l)

-------- Uninstall Beginning --------
Module:  rtl8812AU
Version: 4.3.22-beta
Kernel:  4.4.38+ (armv7l)
-------------------------------------
After:
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.4.48+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/dkms 4.4.48+ /boot/kernel.img
/etc/kernel/postinst.d/dkms:
Error! Bad return status for module build on kernel: 4.4.48+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log for more information.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.4.48+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 4.4.48-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/dkms 4.4.48-v7+ /boot/kernel7.img
/etc/kernel/postinst.d/dkms:
Error! Bad return status for module build on kernel: 4.4.48-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log for more information.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 4.4.48-v7+ /boot/kernel7.img
Setting up libraspberrypi-bin (1.20170215-1) ...
Setting up raspberrypi-kernel-headers (1.20170215-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.4.48+
Error! Bad return status for module build on kernel: 4.4.48+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log for more information.
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.4.48-v7+
Error! Bad return status for module build on kernel: 4.4.48-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log for more information.
root@pi-router:/tmp# cat /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log
DKMS make.log for rtl8812AU-4.3.22-beta for kernel 4.4.48-v7+ (armv7l)
Wed Mar  1 19:15:45 CET 2017
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.4.38-v7+/build M=/var/lib/dkms/rtl8812AU/4.3.22-beta/build  modules
make[1]: *** /lib/modules/4.4.38-v7+/build: No such file or directory.  Stop.
Makefile:1705: recipe for target 'modules' failed
make: *** [modules] Error 2

So looks that dkms tried to recompile modules to the 4.4.48-v7+ using libraries from the previous version. Maybe takes as parameter on the CROSS_COMPILE the current version from some 'uname -r'.

jeremyb31, 2017-03-08:
fumantsu
Where those results using the source from this github or mine?
Either way, it would be interesting to see

cat /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log 

if the original file from the failure still exists

fumantsu, 2017-03-08:
Well I'm not cloning the git. Just got the zip if I remember correctly some
time ago

cat /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log
DKMS make.log for rtl8812AU-4.3.22-beta for kernel 4.4.50-v7+ (armv7l)
Sun Mar  5 10:08:28 CET 2017
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.4.48-v7+/build
M=/var/lib/dkms/rtl8812AU/4.3.22-beta/build  modules
make[1]: *** /lib/modules/4.4.48-v7+/build: No such file or directory.
Stop.
Makefile:1705: recipe for target 'modules' failed
make: *** [modules] Error 2