diederikdehaas/rtl8812AU #53: Found no install instructions

gaining, 2017-01-21:
I'm would like to install this driver on mint 17 but there's no instructions at all. Not even for debian?

fumantsu, 2017-01-21:
README.md which is on the main page doesn't have instructions?

gaining, 2017-01-22:
Besides the dkms feature, I found no install instructions.

gaining, 2017-01-22:
I was able to find instructions from another repo though, it works for this one: https://github.com/abperiasamy/rtl8812AU_8821AU_linux/blob/master/README.md
Instructions:
# sudo make clean # sudo make # sudo make install # sudo modprobe -a rtl8812au

fumantsu, 2017-01-22:
The README.md contains the installation instructions
First you don't need to do make clean/make/make install. DKMS is doing for you and normally is doing it for you in every kernel change. Also if I'm not mistaken is doing also the modprobe in the end (If not, the you need to do only the modprobe) but in anyway I would suggest to search and learn what DKMS is doing.
 
If you need to go through to compile a module at least the point how to insert a module in to kernel you should be aware. It is not a everyday task and not for all.

diederikdehaas, 2017-01-22:
@fumantsu is completely right, DKMS is what you want.

Furthermore the instructions you found would probably work, but it's a typical case of *let's put sudo in front of everything*.
Normally, all `make` commands should be done as normal user, except `make install`

gaining, 2017-01-22:
Thanks, for the clarification, I've never used dkms to install anything before so I did not know about it. On that note, can you provide me with the uninstall instruction so that I can install it properly? It seems the driver only works right after I install it but after a reboot, it doesn't with my current method of install.

diederikdehaas, 2017-01-22:
`sudo make uninstall` should remove the module again.
In `/lib/modules/$(uname -r)/kernel/drivers/net/wireless/` you'll likely find the 8812au.ko module and after running `sudo make uninstall` it should be removed there.

DKMS stores them in a slightly different location, `/lib/modules/<kernel-version>/updates/dkms/`, but the system/kernel will be configured such that it'll find them there.

closed 2017-01-22 by gaining