diederikdehaas/rtl8812AU #56: Kernel Panics with "iw dev wlanX del"

atyg02, 2017-02-02:
I have compiled the source driver for [EDIMAX EW-7822UAC]USB wifi  in openwrt-chaos_calmer (15.05) and Ubunut (14.04).
when ever I issue " iw dev wlanX del" command then unplug the device, the kernel panics.

"iw dev wlan0 del"
`[ 665.916589] RTL871X: rtw_ndev_uninit(wlan0) if1`

If I unplugged the USB device it gives the the following the kernel mesg:

`[ 964.188648] usb 2-2: USB disconnect, device number 2
[ 964.193916] RTL871X: +rtw_dev_remove
[ 964.197633] RTL871X: [0x00100000,5]+dev_remove()
[ 964.202580] RTL871X: rtw_wdev_unregister(wdev=f60cf800)
[ 964.207968] RTL871X: rtw_cfg80211_indicate_scan_done without scan req
[ 964.214761] ------------[ cut here ]------------
[ 964.219504] Kernel BUG at c1391463 [verbose debug info unavailable]
[ 964.225927] invalid opcode: 0000 [#1] SMP
[ 964.230228] Modules linked in: 8812au tunprox pppoe ppp_async iptable_nat ath9k rt2500usb rndis_host pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE huawei_cy
[ 964.378478] CPU: 1 PID: 393 Comm: kworker/1:1 Not tainted 3.18.29 #30
[ 964.385025] Hardware name: PC Engines apu2/apu2, BIOS 88a4f96 03/07/2016
[ 964.391888] Workqueue: usb_hub_wq hub_event
[ 964.396280] task: f7285d50 ti: f71f8000 task.ti: f71f8000
[ 964.401845] EIP: 0060:[] EFLAGS: 00010202 CPU: 1
[ 964.407551] EIP is at rollback_registered_many+0xa3/0x230
[ 964.413075] EAX: 00000003 EBX: f54ef000 ECX: f7285d50 EDX: 00000000
[ 964.419532] ESI: f71f9d90 EDI: f71f9d50 EBP: f71f9d74 ESP: f71f9d64
[ 964.425989] DS: 007b ES: 007b FS: 00d8 GS: 0000 SS: 0068
[ 964.431559] CR0: 80050033 CR2: b7784558 CR3: 360ca000 CR4: 000407d0
[ 964.437923] Stack:
[ 964.439993] f607e81c c1463551 00000246 00000001 f71f9d74 f71f9d74 f54ef000 00000000
[ 964.448125] f607e800 f607e81c c139160a f54ef040 f54ef040 c1392bb7 f95cf7ce f962557e
[ 964.456318] f961b18f f54ef000 f54ef000 c1392c01 f96d4000 f95b5a03 00000000 f695f000
[ 964.464474] Call Trace:
[ 964.466989] [] ? printk+0x37/0x3b
[ 964.471476] [] ? rollback_registered+0x1a/0x30
[ 964.477048] [] ? unregister_netdevice_queue+0x67/0xa0
[ 964.483258] [] ? rtw_wdev_unregister+0x44/0xc2 [8812au]
[ 964.489705] [] ? unregister_netdev+0x11/0x20
[ 964.495084] [] ? rtw_os_ndev_unregister+0x2a/0x3f [8812au]
[ 964.501862] [] ? rtw_os_ndevs_unregister+0x1b/0x33 [8812au]
[ 964.508620] [] ? rtw_resume_common+0x3fd/0x4de [8812au]
[ 964.515025] [] ? usb_unbind_interface+0x54/0x170
[ 964.520717] [] ? __blocking_notifier_call_chain+0x40/0x50
[ 964.527217] [] ? __device_release_driver+0x4b/0x90
[ 964.533083] [] ? device_release_driver+0x15/0x20
[ 964.538855] [] ? bus_remove_device+0xbc/0xe0
[ 964.544246] [] ? device_del+0xeb/0x160
[ 964.549067] [] ? usb_disable_device+0x61/0x170
[ 964.554650] [] ? usb_disconnect+0x9a/0x1a0
[ 964.559916] [] ? usb_clear_port_feature+0x48/0x50
[ 964.565685] [] ? hub_event+0x459/0xdf0
[ 964.570608] [] ? vmstat_shepherd+0x9e/0xe0
[ 964.575797] [] ? process_one_work+0x15a/0x290
[ 964.581334] [] ? worker_thread+0x243/0x3d0
[ 964.586538] [] ? cancel_delayed_work_sync+0x10/0x10
[ 964.592574] [] ? kthread+0xaa/0xb0
[ 964.597003] [] ? ____call_usermodehelper+0xf7/0x110
[ 964.603011] [] ? ret_from_kernel_thread+0x21/0x30
[ 964.608801] [] ? kthread_create_on_node+0x110/0x110
[ 964.614814] Code: ca ff 8b 53 40 8b 43 44 89 42 04 89 10 c7 43 40 00 01 10 00 c7 43 44 00 02 20 00 eb 13 8d 76 00 3c 01 c6 83 a1 02 00 00 01 74 05 <0f> 0b 8d 76 00 8b 47 40 89 fb 8d 78 c0 8d 43 40 39 cb
[ 964.635896] EIP: [] rollback_registered_many+0xa3/0x230 SS:ESP 0068:f71f9d64
[ 964.644139] ---[ end trace cfa67b4a0a2dba10 ]---
[ 964.648912] Kernel panic - not syncing: Fatal exception
[ 964.654310] Kernel Offset: 0x0 from 0xc1000000 (relocation range: 0xc0000000-0xf878cfff)
[ 964.662686] Rebooting in 10 seconds..`

Do you have any suggestion how to fix it?
BR
Aman

weedy, 2017-02-02:
Try removing the module after you bring down the interface.

atyg02, 2017-02-02:
Hi weedy,
 thank you for your reply.
Here is the out after bring down the interface and then removing the module

#######################      In Openwrt   #########################################
        root@OpenWrt:/# 
	root@OpenWrt:/# iw dev wlan0 del
	[   97.512896] RTL871X: rtw_ndev_uninit(wlan0) if1
	root@OpenWrt:/# 
	root@OpenWrt:/# 
	root@OpenWrt:/# 
	root@OpenWrt:/# 
	root@OpenWrt:/# rmmod 8812au
	[  112.254706] RTL871X: module exit start
	[  112.258573] usbcore: deregistering interface driver rtl8821au
	[  112.264628] ------------[ cut here ]------------
	[  112.269441] Kernel BUG at ffffffff813291d3 [verbose debug info unavailable]
	[  112.276699] invalid opcode: 0000 [#1] SMP 
	[  112.280993] Modules linked in: 8812au(-) pppoe ppp_async iptable_nat ath9k rt2800usb rt2800lib rndis_host pppox ppp_generic nf_nat_ipv4 nf_coy
	[  112.396681] CPU: 2 PID: 1662 Comm: rmmod Not tainted 3.18.29 #10
	[  112.402885] task: ffff88011a33c380 ti: ffff8800d9d78000 task.ti: ffff8800d9d78000
	[  112.410486] RIP: 0010:[<ffffffff813291d3>]  [<ffffffff813291d3>] rollback_registered_many+0xc3/0x2b0
	[  112.419985] RSP: 0018:ffff8800d9d7bd38  EFLAGS: 00010202
	[  112.425409] RAX: 0000000000000003 RBX: ffff88011a1b5000 RCX: 00000000ffffffff
	[  112.432797] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff8800d9d7bd88
	[  112.440089] RBP: ffff8800d9d7bd88 R08: 0000000000000000 R09: 0000000000000000
	[  112.447418] R10: ffffffff81120279 R11: ffff88011de74a80 R12: ffff8800d9d7bd18
	[  112.454707] R13: ffff8800d9d7bd48 R14: 0000000000000000 R15: ffff880119a73400
	[  112.462058] FS:  00007f63f31ad6a0(0000) GS:ffff88011ed00000(0000) knlGS:0000000000000000
	[  112.470388] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
	[  112.476251] CR2: 00007f639a196fe8 CR3: 000000011a30c000 CR4: 00000000000407e0
	[  112.483603] Stack:
	[  112.485666]  ffffffff816ad7b8 000000018022001d ffff8800d9d7bd48 ffff8800d9d7bd48
	[  112.493350]  0000000000000000 ffff88011a1b5000 0000000000000000 0000000000000000
	[  112.501038]  ffff88011983b230 ffffffff813293e1 ffff88011a1b5070 ffff88011a1b5070
	[  112.508743] Call Trace:
	[  112.511249]  [<ffffffff813293e1>] ? rollback_registered+0x21/0x40
	[  112.517531]  [<ffffffff8132a580>] ? unregister_netdevice_queue+0x70/0xc0
	[  112.524395]  [<ffffffff8132a5e3>] ? unregister_netdev+0x13/0x20
	[  112.530627]  [<ffffffffa061f9db>] ? rtw_os_ndev_unregister+0x33/0x4d [8812au]
	[  112.538053]  [<ffffffffa06203bc>] ? rtw_os_ndevs_unregister+0x22/0x40 [8812au]
	[  112.545554]  [<ffffffffa0620fdb>] ? rtw_resume_common+0x254/0x2c3 [8812au]
	[  112.552637]  [<ffffffff812bf788>] ? usb_unbind_interface+0x68/0x190
	[  112.559091]  [<ffffffff81262d4e>] ? __device_release_driver+0x6e/0xe0
	[  112.565713]  [<ffffffff8126346e>] ? driver_detach+0x7e/0xb0
	[  112.571459]  [<ffffffff81262ac6>] ? bus_remove_driver+0x66/0xa0
	[  112.577499]  [<ffffffff812bfba2>] ? usb_deregister+0x62/0xd0
	[  112.583414]  [<ffffffffa0671ec4>] ? cleanup_module+0x31/0x5c [8812au]
	[  112.590092]  [<ffffffff81087058>] ? SyS_delete_module+0x118/0x1a0
	[  112.596310]  [<ffffffff81413540>] ? int_signal+0x12/0x17
	[  112.601799]  [<ffffffff814132c9>] ? system_call_fastpath+0x12/0x17
	[  112.608165] Code: 00 00 00 ad de 48 89 43 70 48 b8 00 02 20 00 00 00 ad de 48 89 43 78 eb 17 0f 1f 80 00 00 00 00 3c 01 c6 83 39 04 00 00 01  
	[  112.629272] RIP  [<ffffffff813291d3>] rollback_registered_many+0xc3/0x2b0
	[  112.636224]  RSP <ffff8800d9d7bd38>
	[  112.639899] ---[ end trace 1a4b645455aae5bb ]---
	Segmentation fault
	root@OpenWrt:/# 

##################### ###  Ubuntu #####################################
sudo iw dev wlan3 del
	[  906.040066] RTL871X: rtw_ndev_uninit(wlan3)

sudo rmmod 8812au

[ 1065.515784] usbcore: deregistering interface driver rtl8821au
[ 1065.516134] ------------[ cut here ]------------
[ 1065.516260] kernel BUG at /build/linux-lts-utopic-9LCANk/linux-lts-utopic-3.16.0/net/core/dev.c:5781!
[ 1065.516453] invalid opcode: 0000 [#1] SMP 
[ 1065.516573] Modules linked in: 8812au(OE-) pl2303 usbserial pf_ring(OE) nfnetlink_queue nfnetlink_log nfnetlink dummy ctr ccm rfcomm bnep bluetooth 6lowpan_iphc binfmt_misc nls_iso8859_1 uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common videodev media snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel snd_hda_controller snd_hda_codec snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer parport_pc ppdev snd lp parport arc4 hp_wmi sparse_keymap intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm crct10dif_pclmul crc32_pclmul cryptd rt2800pci rt2800mmio rt2800lib rt2x00pci rt2x00mmio rt2x00lib mac80211 joydev rtsx_pci_ms serio_raw memstick cfg80211 eeprom_93cx6 crc_ccitt i915 soundcore shpchp mei_me lpc_ich mei wmi drm_kms_helper video drm hp_wireless mac_hid i2c_algo_bit hid_generic usbhid hid rtsx_pci_sdmmc psmouse ahci r8169 rtsx_pci libahci mii
[ 1065.518951] CPU: 1 PID: 6320 Comm: rmmod Tainted: G           OE 3.16.0-67-generic #87~14.04.1-Ubuntu
[... 22 lines cut ...]
[ 1065.522142]  [<ffffffffc08f8f37>] rtw_dev_remove+0x25/0x77 [8812au]
[ 1065.522219]  [<ffffffff81572074>] usb_unbind_interface+0x74/0x2b0
[ 1065.522293]  [<ffffffff814bce5f>] __device_release_driver+0x7f/0xf0
[ 1065.522368]  [<ffffffff814bd828>] driver_detach+0xb8/0xc0
[ 1065.522436]  [<ffffffff814bcb15>] bus_remove_driver+0x55/0xd0
[ 1065.522507]  [<ffffffff814bdffc>] driver_unregister+0x2c/0x50
[... cut ...]
[ 1065.522981] Code: 70 48 8d 50 90 49 8d 44 24 70 49 39 c5 74 24 4c 89 e3 49 89 d4 0f b6 83 50 04 00 00 84 c0 74 93 3c 01 c6 83 51 04 00 00 01 74 d1 <0f> 0b 66 0f 1f 44 00 00 49 8b 55 00 49 39 d5 48 8d 42 90 74 30 
[ 1065.523558] RIP  [<ffffffff81662f88>] rollback_registered_many+0xd8/0x2e0
[ 1065.523733]  RSP <ffff88003668bcf0>
[ 1065.547116] ---[ end trace 2d4f9ee3828e5d4a ]---

still the problem exits on both distributions.

in Ubuntu if I run  "ifconfig"

it hangs up and prints nothing. seems the network manager is not there any more.

Thank you again.
BR
Aman

atyg02, 2017-02-02:
additional information, I have used this [https://github.com/dl12345/rtl8812au](url) Makefile for my openwrt-chaos_calmer (15.05) Kernel Ver= 3.18.29

In Ubuntu I have tried all the branches driver-4.3.14, driver-4.3.20 and driver-4.3.22-beta
and all of them have same problem.

Thank you
Aman

weedy, 2017-02-04:
Unrelated.
Do you have any idea what I messed up with the build system to cause the missing symbols?
https://forum.lede-project.org/t/rtl8812au-and-or-rtl8814au-drivers/67/7

atyg02, 2017-02-04:
I think first you need  first manually copy  compat-wireless "Module.symvers" located in build_dir/target.../linux-.../compat-wireless  into the folder "staging_dir/target.../usr/include/mac80211/Module.symvers" as to allow proper linking.

You can get the patches and the Makefile from here: https://github.com/dl12345/rtl8812au

atyg02, 2017-02-17:
@weedy @diederikdehaas 
do you know any repo for rtll8814au which supports VHT capability?

I have tested  the speed of  rtl8812au [EDIMAX EW-7822UAC]USB with VHT capability , I got around 78Mbps.

with rtl8814au [ EDIMAX EW-7833UAC]USB, default no VHT capability got 212 Mbps.

And regarding the issue, the problem is partially solved with following patch.
```
-- a/os_dep/linux/ioctl_cfg80211.c	2017-02-11 16:39:20.000000000 +0100
+++ b/os_dep/linux/ioctl_cfg80211.c	2017-02-12 23:45:35.177685046 +0100
@@ -4113,6 +4113,8 @@
 		DBG_871X(FUNC_NDEV_FMT" remove monitor interface\n", FUNC_NDEV_ARG(ndev));
 	}
 
+	if (adapter) adapter->pnetdev = NULL;
+
 exit:
 	return ret;
 }
--- a/os_dep/linux/os_intfs.c	2017-02-11 16:39:20.000000000 +0100
+++ b/os_dep/linux/os_intfs.c	2017-02-12 23:47:14.397684565 +0100
@@ -1127,8 +1127,10 @@
 	rtw_cfg80211_ndev_res_unregister(adapter);
 #endif
 
-	if ((adapter->DriverState != DRIVER_DISAPPEAR) && netdev)
+	if ((adapter->DriverState != DRIVER_DISAPPEAR) && netdev) {
 		unregister_netdev(netdev); /* will call netdev_close() */
+		adapter->pnetdev = NULL;
+	}
 
 #if defined(CONFIG_IOCTL_CFG80211) && !defined(RTW_SINGLE_WIPHY)
rtw_wiphy_unregister(adapter_to_wiphy(adapter));

```

uminokoe, 2017-02-18:
@atyg02 
Could you please try my driver which is available [here](https://github.com/uminokoe/rtl8812AU/tree/driver-4.3.21)?

I first imported the driver version 4.3.21 from EDIMAX website (which is targeted for RTL8814AU). Then I made some changes to add VHT capabilities (I don't know whether station mode works without adding the capabilities, but on my PC, I need them to let hostapd / AP mode work) and to allow it to work with my RTL8811AU/RTL8812AU/RTL8821AU series chipset.

To test my driver, you will need to:
1. Edit some files (e.g. Makefile, Kconfig, autoconf.h) so that you will get a module compiled for RTL8814AU
2. Add your USB device ID if necessary

Notes:
The driver is highly experimental. Speed, connection, and any other issues may exist. Some features may be badly or not implemented. **Please test and use it at your own risk.**

uminokoe, 2017-02-18:
For your reference, when I use RTL8811AU (1T1R, 80MHz) for AP mode, iperf3 (PC as server, Android as client) reported a speed of 180 Mbps to 200 Mbps. Sometimes when I do a reverse test in iperf3, the speed is only 60 Mbps to 80Mbps.

atyg02, 2017-02-18:
@uminokoe 
Thank you for you reply.
I have compiled your driver with the changes  (Makefile, Kconfig, autoconf.h) to switch it to 8814au, but still could not get the VHT capability.

The available channels and supported modes from rtl8814au are:
```
Available Channel Information on phy2
=====================================
Band 1:
Channel  Freq  Allowed HT Modes
0        0000  HT20 HT40 HT40+ HT40-
1        2412  HT20 HT40 HT40+
2        2417  HT20 HT40 HT40+
3        2422  HT20 HT40 HT40+
4        2427  HT20 HT40 HT40+
5        2432  HT20 HT40 HT40+ HT40-
6        2437  HT20 HT40 HT40+ HT40-
7        2442  HT20 HT40 HT40+ HT40-
8        2447  HT20 HT40 HT40+ HT40-
9        2452  HT20 HT40 HT40+ HT40-
10       2457  HT20 HT40 HT40-
11       2462  HT20 HT40 HT40-
12       2467  HT20 HT40 HT40-
13       2472  HT20 HT40 HT40-

Band 2:
Channel  Freq  Allowed HT Modes
0        0000  HT20 HT40 HT40+ HT40-
36       5180  HT20 HT40 HT40+
40       5200  HT20 HT40 HT40-
44       5220  HT20 HT40 HT40+
48       5240  HT20 HT40 HT40-
```
do you have  any idea where I missed in the config files  to enable VHT capability ?
BR
Aman

atyg02, 2017-02-18:
These are the changes i made to switch from 8812au to 8814au
```
--- a/include/autoconf.h        2017-02-18 12:03:23.941307887 +0100
+++ b/include/autoconf.h        2016-05-31 10:07:07.000000000 +0200
@@ -26,8 +26,8 @@
  * Public  General Config
  */
 #define AUTOCONF_INCLUDED
-#define RTL871X_MODULE_NAME "8812AU"
-#define DRV_NAME "rtl8812au"
+#define RTL871X_MODULE_NAME "8814AU"
+#define DRV_NAME "rtl8814au"

 #define CONFIG_USB_HCI 1
```

```
--- a/Makefile	2017-02-18 12:03:23.845305039 +0100
+++ b/Makefile	2017-02-18 11:47:26.663680000 +0100
@@ -14,7 +14,7 @@
 EXTRA_CFLAGS += -Wno-unused-function
 EXTRA_CFLAGS += -Wno-unused
 #EXTRA_CFLAGS += -Wno-uninitialized
-EXTRA_CFLAGS += -Wno-error=date-time	# Fix compile error on gcc 4.9 and later
+#EXTRA_CFLAGS += -Wno-error=date-time	# Fix compile error on gcc 4.9 and later
 
 EXTRA_CFLAGS += -I$(src)/include
 EXTRA_CFLAGS += -I$(src)/hal/phydm
@@ -26,11 +26,11 @@
 ########################## WIFI IC ############################
 CONFIG_MULTIDRV = n
 CONFIG_RTL8188E = n
-CONFIG_RTL8812A = y
-CONFIG_RTL8821A = y
+CONFIG_RTL8812A = n
+CONFIG_RTL8821A = n
 CONFIG_RTL8192E = n
 CONFIG_RTL8723B = n
-CONFIG_RTL8814A = n
+CONFIG_RTL8814A = y
 CONFIG_RTL8723C = n
 CONFIG_RTL8188F = n
 ######################### Interface ###########################
@@ -74,7 +74,7 @@
 ######### Notify SDIO Host Keep Power During Syspend ##########
 CONFIG_RTW_SDIO_PM_KEEP_POWER = y
 ###################### MP HW TX MODE FOR VHT #######################
-CONFIG_MP_VHT_HW_TX_MODE = n
+CONFIG_MP_VHT_HW_TX_MODE = y
 ###################### Platform Related #######################
 CONFIG_PLATFORM_I386_PC = y
 CONFIG_PLATFORM_ANDROID_X86 = n
@@ -528,8 +528,8 @@
 ########### HAL_RTL8814A #################################
 ifeq ($(CONFIG_RTL8814A), y)
 ## ADD NEW VHT MP HW TX MODE ##
-#EXTRA_CFLAGS += -DCONFIG_MP_VHT_HW_TX_MODE
-#CONFIG_MP_VHT_HW_TX_MODE = y
+EXTRA_CFLAGS += -DCONFIG_MP_VHT_HW_TX_MODE
+CONFIG_MP_VHT_HW_TX_MODE = y
 ##########################################
 RTL871X = rtl8814a
 ifeq ($(CONFIG_USB_HCI), y)
@@ -1690,11 +1690,11 @@
 $(MODULE_NAME)-$(CONFIG_MP_INCLUDED)+= core/rtw_bt_mp.o
 endif
 
-obj-$(CONFIG_RTL8812AU_8821AU) := $(MODULE_NAME).o
+obj-$(CONFIG_RTL8814AU) := $(MODULE_NAME).o
 
 else
 
-export CONFIG_RTL8812AU_8821AU = m
+export CONFIG_RTL8814AU = m
 
 all: modules
```

```
--- a/Kconfig	2017-02-18 12:03:23.845305039 +0100
+++ b/Kconfig	2016-05-31 10:07:07.000000000 +0200
@@ -1,6 +1,6 @@
-config RTL8821AU
-	tristate "Realtek 8821A USB WiFi"
+config RTL8814AU
+	tristate "Realtek 8814A USB WiFi"
 	depends on USB
 	---help---
-	  Help message of RTL8821AU
+	  Help message of RTL8814AU
```

uminokoe, 2017-02-18:
@atyg02 Do you have the latest version of `iw`, as well as the output of command `iw list`?
The following is part of the output of `iw list` for me. Notice that VHT capabilities shows up.
```
	Band 2:
		Capabilities: 0x1962
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		VHT Capabilities (0x03c03122):
			Max MPDU length: 11454
			Supported Channel Width: neither 160 nor 80+80
			short GI (80 MHz)
			SU Beamformee
			+HTC-VHT
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: not supported
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 434 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: not supported
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 434 Mbps
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5170 MHz [34] (disabled)
			* 5180 MHz [36] (17.0 dBm)
			* 5190 MHz [38] (disabled)
			* 5200 MHz [40] (17.0 dBm)
			* 5210 MHz [42] (disabled)
			* 5220 MHz [44] (17.0 dBm)
			* 5230 MHz [46] (disabled)
			* 5240 MHz [48] (17.0 dBm)
			* 5260 MHz [52] (24.0 dBm) (no IR)
			* 5280 MHz [56] (24.0 dBm) (no IR)
			* 5300 MHz [60] (24.0 dBm) (no IR)
			* 5320 MHz [64] (24.0 dBm) (no IR)
			* 5500 MHz [100] (24.0 dBm) (no IR)
			* 5520 MHz [104] (24.0 dBm) (no IR)
			* 5540 MHz [108] (24.0 dBm) (no IR)
			* 5560 MHz [112] (24.0 dBm) (no IR)
			* 5580 MHz [116] (24.0 dBm) (no IR)
			* 5600 MHz [120] (24.0 dBm) (no IR)
			* 5620 MHz [124] (24.0 dBm) (no IR)
			* 5640 MHz [128] (24.0 dBm) (no IR)
			* 5660 MHz [132] (24.0 dBm) (no IR)
			* 5680 MHz [136] (24.0 dBm) (no IR)
			* 5700 MHz [140] (24.0 dBm) (no IR)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5920 MHz [184] (disabled)
			* 5940 MHz [188] (disabled)
			* 5960 MHz [192] (disabled)
			* 5980 MHz [196] (disabled)
			* 6000 MHz [200] (disabled)
			* 6020 MHz [204] (disabled)
			* 6040 MHz [208] (disabled)
			* 6060 MHz [212] (disabled)
			* 6080 MHz [216] (disabled)
```

atyg02, 2017-02-18:
@uminokoe  Here is the output of iw list
```
 iw list
Wiphy phy1
	Band 1:
		Capabilities: 0x19e3
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT TX/RX MCS rate indexes supported: 0-23
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm)
			* 2472 MHz [13] (20.0 dBm)
			* 2484 MHz [14] (disabled)
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
	Band 2:
		Capabilities: 0x19e3
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT TX/RX MCS rate indexes supported: 0-23
		Frequencies:
			* 5170 MHz [34] (disabled)
			* 5180 MHz [36] (30.0 dBm)
			* 5190 MHz [38] (disabled)
			* 5200 MHz [40] (30.0 dBm)
			* 5210 MHz [42] (disabled)
			* 5220 MHz [44] (30.0 dBm)
			* 5230 MHz [46] (disabled)
			* 5240 MHz [48] (30.0 dBm)
			* 5260 MHz [52] (disabled)
			* 5280 MHz [56] (disabled)
			* 5300 MHz [60] (disabled)
			* 5320 MHz [64] (disabled)
			* 5500 MHz [100] (disabled)
			* 5520 MHz [104] (disabled)
			* 5540 MHz [108] (disabled)
			* 5560 MHz [112] (disabled)
			* 5580 MHz [116] (disabled)
			* 5600 MHz [120] (disabled)
			* 5620 MHz [124] (disabled)
			* 5640 MHz [128] (disabled)
			* 5660 MHz [132] (disabled)
			* 5680 MHz [136] (disabled)
			* 5700 MHz [140] (disabled)
			* 5745 MHz [149] (disabled)
			* 5765 MHz [153] (disabled)
			* 5785 MHz [157] (disabled)
			* 5805 MHz [161] (disabled)
			* 5825 MHz [165] (disabled)
			* 5920 MHz [184] (disabled)
			* 5940 MHz [188] (disabled)
			* 5960 MHz [192] (disabled)
			* 5980 MHz [196] (disabled)
			* 6000 MHz [200] (disabled)
			* 6020 MHz [204] (disabled)
			* 6040 MHz [208] (disabled)
			* 6060 MHz [212] (disabled)
			* 6080 MHz [216] (disabled)
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
	software interface modes (can always be added):
	interface combinations are not supported
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * new_beacon
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * action
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
```

atyg02, 2017-02-18:
@uminokoe  additional information.
I have two USB wifi one supports 8812au [EW-7822UAC] and another one supports 8814au [EW-7833UAC]

I have compiled both of them.
rtl8814au load ok and but not VHT capabilty.
but the rtl8812au simply loops on registering and unregistered the device.
here is the dmesg output:
```
[ 6083.088181] RTL871X: USB_SPEED_HIGH
[ 6083.088184] RTL871X: CHIP TYPE: RTL8812
[ 6083.088356] RTL871X: read_chip_version_8812a SYS_CFG(0xF0)=0x04411135 
[ 6083.088482] RTL871X: rtw_hal_config_rftype RF_Type is 2 TotalTxPath is 2 
[ 6083.088497] RTL871X: Chip Version Info: CHIP_8812_Normal_Chip_TSMC_C_CUT_2T2R_RomVer(0)
[ 6083.088502] RTL871X: _ConfigChipOutEP_8812 OutEpQueueSel(0x07), OutEpNumber(3) 
[ 6083.088601] RTL871X: Boot from EFUSE, Autoload OK !
[ 6083.427232] RTL871X: EEPROM ID=0x8129
[ 6083.427244] RTL871X: VID = 0x7392, PID = 0xA822
[ 6083.427250] RTL871X: Customer ID: 0x00, SubCustomer ID: 0xCD
[ 6083.427273] RTL871X: EEPROMRegulatory = 0x1 
[ 6083.427276] RTL871X: Board Type: 0x 0
[ 6083.427280] RTL871X: mlmepriv.ChannelPlan = 0x26
[ 6083.427283] RTL871X: CrystalCap: 0x e
[ 6083.427285] RTL871X: ThermalMeter = 0x23
[ 6083.427289] RTL871X: Hal_ReadRemoteWakeup_8812A...bSupportRemoteWakeup(0)
[ 6083.427293] RTL871X: SWAS: bHwAntDiv = 0, TRxAntDivType = 3
[ 6083.427298] RTL871X: pHalData->PAType_2G is 0x3, pHalData->ExternalPA_2G = 0
[ 6083.427303] RTL871X: pHalData->PAType_5G is 0x3, pHalData->ExternalPA_5G = 1
[ 6083.427308] RTL871X: pHalData->LNAType_2G is 0x0, pHalData->ExternalLNA_2G = 0
[ 6083.427314] RTL871X: pHalData->LNAType_5G is 0x88, pHalData->ExternalLNA_5G = 1
[ 6083.427320] RTL871X: pHalData->TypeGPA = 0x0
[ 6083.427322] RTL871X: pHalData->TypeAPA = 0x0
[ 6083.427325] RTL871X: pHalData->TypeGLNA = 0x0
[ 6083.427327] RTL871X: pHalData->TypeALNA = 0x0
[ 6083.427330] RTL871X: RFE Type: 0x 0
[ 6083.427334] RTL871X: PID= 0xa822, VID=  7392
[ 6083.427337] RTL871X: PID= 0xa822, VID=  7392
[ 6083.427340] RTL871X: Customer ID: 0x 0
[ 6083.429101] RTL871X: hal_ReadUsbType_8812AU: antenna=2, wmode=3
[ 6083.429107] RTL871X: hal_ReadUsbType_8812AU(): EFUSE_HIDDEN_812AU
[ 6083.429114] RTL871X: rtw_hal_read_chip_info in 340 ms
[ 6083.429194] RTL871X: init_channel_set((null)) ChannelPlan ID:0x26, ch num:32
[ 6083.429435] RTL871X: NR_RECVBUFF: 8
[ 6083.429442] RTL871X: MAX_RECVBUF_SZ: 32768
[ 6083.429456] RTL871X: NR_PREALLOC_RECV_SKB: 8
[ 6083.429562] RTL871X: rtw_alloc_macid((null)) if1, hwaddr:ff:ff:ff:ff:ff:ff macid:1
[ 6083.429585] RTL871X: can't get autopm: 
[ 6083.429592] RTL871X: rtw_macaddr_cfg mac addr:74:da:38:82:86:46
[ 6083.429597] RTL871X: bDriverStopped:True, bSurpriseRemoved:False, bup:0, hw_init_completed:0
[ 6083.430127] RTL871X: rtw_wiphy_alloc(phy470)
[ 6083.430134] RTL871X: rtw_wdev_alloc(padapter=ffffc900112e5000)
[ 6083.430141] RTL871X: rtw_wiphy_register(phy470)
[ 6083.430146] RTL871X: Register RTW cfg80211 vendor cmd(0x67) interface 
[ 6083.430889] RTL871X: rtw_ndev_init(wlan1) if1 mac_addr=74:da:38:82:86:46
[ 6083.459026] systemd-udevd[23268]: renamed network interface wlan1 to wlan46
[ 6083.476064] RTL871X: +871x_drv - drv_open, bup=0
[ 6083.480061] RTL871X: reg 0x9, usb read 1 fail, status:-71 value=0xc1363000, vendorreq_times:1
[ 6083.484312] RTL871X: reg 0x9, usb read 1 fail, status:-71 value=0xc1363000, vendorreq_times:2
[ 6083.487133] usb 2-1.2: USB disconnect, device number 102
[ 6083.487319] RTL871X: +rtw_dev_remove
[ 6083.487326] RTL871X: rtw_wdev_unregister(wdev=ffff880147c4a400)
[ 6083.488548] RTL871X: reg 0x9, usb read 1 fail, status:-71 value=0xc1363000, vendorreq_times:3
[ 6083.488559] RTL871X: reg 0x9, usb read 1 fail, status:-19 value=0xc1363000, vendorreq_times:4
[ 6083.488566] RTL871X: reg 0x9, usb read 1 fail, status:-19 value=0xc1363000, vendorreq_times:5
[ 6083.488571] RTL871X: [dvobj:ffff8800a755d000][ERROR] continual_io_error:5 > 4
[ 6083.488576] RTL871X:  power-on :REG_SYS_CLKR 0x09=0x00. REG_CR 0x100=0x00.
[ 6083.488586] RTL871X:  MAC has not been powered on yet.

```
have tested your driver with EW-7822AUC wifi ?
Thanks for your patience.
BR
Aman

atyg02, 2017-02-18:
@uminokoe 

The problem was from iw version I have used in my notebook is (3.4)
Then I have cross compiled to  my openwrt KV 3.18.45 and iw version 4.3
VHT is enabled

iw list out:
```
iw list
Wiphy phy0
        max # scan SSIDs: 9
        max scan IEs length: 2304 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x19e3
                        RX LDPC
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT Max RX data rate: 450 Mbps
                HT TX/RX MCS rate indexes supported: 0-23
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm)
                        * 2472 MHz [13] (20.0 dBm)
                        * 2484 MHz [14] (disabled)
        Band 2:
                Capabilities: 0x19e3
                        RX LDPC
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT Max RX data rate: 450 Mbps
                HT TX/RX MCS rate indexes supported: 0-23
                VHT Capabilities (0x03c050b2):
                        Max MPDU length: 11454
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                        SU Beamformee
                        +HTC-VHT
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 1300 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 1300 Mbps
                Frequencies:
                        * 5170 MHz [34] (disabled)
                        * 5180 MHz [36] (30.0 dBm)
                        * 5190 MHz [38] (disabled)
                        * 5200 MHz [40] (30.0 dBm)
                        * 5210 MHz [42] (disabled)
                        * 5220 MHz [44] (30.0 dBm)
                        * 5230 MHz [46] (disabled)
                        * 5240 MHz [48] (30.0 dBm)
                        * 5260 MHz [52] (disabled)
                        * 5280 MHz [56] (disabled)
                        * 5300 MHz [60] (disabled)
                        * 5320 MHz [64] (disabled)
                        * 5500 MHz [100] (disabled)
                        * 5520 MHz [104] (disabled)
                        * 5540 MHz [108] (disabled)
                        * 5560 MHz [112] (disabled)
                        * 5580 MHz [116] (disabled)
                        * 5600 MHz [120] (disabled)
                        * 5620 MHz [124] (disabled)
                        * 5640 MHz [128] (disabled)
                        * 5660 MHz [132] (disabled)
                        * 5680 MHz [136] (disabled)
                        * 5700 MHz [140] (disabled)
                        * 5745 MHz [149] (disabled)
                        * 5765 MHz [153] (disabled)
                        * 5785 MHz [157] (disabled)
                        * 5805 MHz [161] (disabled)
                        * 5825 MHz [165] (disabled)
                        * 5920 MHz [184] (disabled)
                        * 5940 MHz [188] (disabled)
                        * 5960 MHz [192] (disabled)
                        * 5980 MHz [196] (disabled)
                        * 6000 MHz [200] (disabled)
                        * 6020 MHz [204] (disabled)
                        * 6040 MHz [208] (disabled)
                        * 6060 MHz [212] (disabled)
                        * 6080 MHz [216] (disabled)
        interface combinations are not supported
```
Thank you again for your patience and of course your great work
BR
Aman

uminokoe, 2017-02-19:
@atyg02 
For the "loops on registering" problem you mentioned before, has the problem been fixed by using the newer version of `iw`?
If you have time and the driver works, do you mind doing an `iperf`/`iperf3` test (or other speed tests at your convenience)? I would like to know the speed (throughput) of 2T2R (and up) using that driver. My chipset is only 1T1R, 433 Mbps, and my `iperf3` test reports an average speed of ~180 Mbps.

atyg02, 2017-02-20:
@uminokoe version of iw doesn't solve the loop registration of rtl8812au.
it solves the VHT capability with rtl8814au.

Here is the measurement result I got from your driver with rtl8814au.
```
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 192.168.10.190 port 5001 connected with 192.168.10.1 port 59496
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0- 1.0 sec  29.4 MBytes   247 Mbits/sec
[  4]  1.0- 2.0 sec  33.7 MBytes   283 Mbits/sec
[  4]  2.0- 3.0 sec  33.7 MBytes   283 Mbits/sec
[  4]  3.0- 4.0 sec  31.6 MBytes   265 Mbits/sec
[  4]  4.0- 5.0 sec  33.8 MBytes   284 Mbits/sec
[  4]  5.0- 6.0 sec  30.7 MBytes   258 Mbits/sec
[  4]  6.0- 7.0 sec  24.8 MBytes   208 Mbits/sec
[  4]  7.0- 8.0 sec  34.1 MBytes   286 Mbits/sec
[  4]  8.0- 9.0 sec  30.5 MBytes   256 Mbits/sec
[  4]  9.0-10.0 sec  33.2 MBytes   279 Mbits/sec
[  4] 10.0-11.0 sec  25.8 MBytes   216 Mbits/sec
[  4] 11.0-12.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 12.0-13.0 sec  25.4 MBytes   213 Mbits/sec
[  4] 13.0-14.0 sec  32.7 MBytes   274 Mbits/sec
[  4] 14.0-15.0 sec  33.5 MBytes   281 Mbits/sec
[  4] 15.0-16.0 sec  31.8 MBytes   266 Mbits/sec
[  4] 16.0-17.0 sec  34.1 MBytes   286 Mbits/sec
[  4] 17.0-18.0 sec  26.7 MBytes   224 Mbits/sec
[  4] 18.0-19.0 sec  33.3 MBytes   279 Mbits/sec
[  4] 19.0-20.0 sec  31.9 MBytes   267 Mbits/sec
[  4] 20.0-21.0 sec  28.0 MBytes   235 Mbits/sec
[  4] 21.0-22.0 sec  34.1 MBytes   286 Mbits/sec
[  4] 22.0-23.0 sec  26.3 MBytes   220 Mbits/sec
[  4] 23.0-24.0 sec  34.0 MBytes   285 Mbits/sec
[  4] 24.0-25.0 sec  30.5 MBytes   256 Mbits/sec
[  4] 25.0-26.0 sec  29.9 MBytes   251 Mbits/sec
[  4] 26.0-27.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 27.0-28.0 sec  28.0 MBytes   235 Mbits/sec
[  4] 28.0-29.0 sec  34.3 MBytes   288 Mbits/sec
[  4] 29.0-30.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 30.0-31.0 sec  34.0 MBytes   285 Mbits/sec
[  4] 31.0-32.0 sec  34.3 MBytes   288 Mbits/sec
[  4] 32.0-33.0 sec  33.9 MBytes   284 Mbits/sec
[  4] 33.0-34.0 sec  33.2 MBytes   279 Mbits/sec
[  4] 34.0-35.0 sec  32.4 MBytes   272 Mbits/sec
[  4] 35.0-36.0 sec  34.3 MBytes   288 Mbits/sec
[  4] 36.0-37.0 sec  32.0 MBytes   268 Mbits/sec
[  4] 37.0-38.0 sec  23.6 MBytes   198 Mbits/sec
[  4] 38.0-39.0 sec  33.1 MBytes   278 Mbits/sec
[  4] 39.0-40.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 40.0-41.0 sec  32.1 MBytes   269 Mbits/sec
[  4] 41.0-42.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 42.0-43.0 sec  24.4 MBytes   205 Mbits/sec
[  4] 43.0-44.0 sec  34.1 MBytes   286 Mbits/sec
[  4] 44.0-45.0 sec  31.3 MBytes   262 Mbits/sec
[  4] 45.0-46.0 sec  27.2 MBytes   228 Mbits/sec
[  4] 46.0-47.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 47.0-48.0 sec  24.0 MBytes   202 Mbits/sec
[  4] 48.0-49.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 49.0-50.0 sec  29.3 MBytes   246 Mbits/sec
[  4] 50.0-51.0 sec  29.0 MBytes   243 Mbits/sec
[  4] 51.0-52.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 52.0-53.0 sec  26.1 MBytes   219 Mbits/sec
[  4] 53.0-54.0 sec  34.3 MBytes   287 Mbits/sec
[  4] 54.0-55.0 sec  28.5 MBytes   239 Mbits/sec
[  4] 55.0-56.0 sec  32.4 MBytes   271 Mbits/sec
[  4] 56.0-57.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 57.0-58.0 sec  34.3 MBytes   287 Mbits/sec
[  4] 58.0-59.0 sec  24.3 MBytes   204 Mbits/sec
[  4] 59.0-60.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 60.0-61.0 sec  30.8 MBytes   258 Mbits/sec
[  4] 61.0-62.0 sec  20.2 MBytes   170 Mbits/sec
[  4] 62.0-63.0 sec  31.2 MBytes   262 Mbits/sec
[  4] 63.0-64.0 sec  27.2 MBytes   228 Mbits/sec
[  4] 64.0-65.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 65.0-66.0 sec  26.2 MBytes   220 Mbits/sec
[  4] 66.0-67.0 sec  34.1 MBytes   286 Mbits/sec
[  4] 67.0-68.0 sec  29.5 MBytes   247 Mbits/sec
[  4] 68.0-69.0 sec  28.0 MBytes   235 Mbits/sec
[  4] 69.0-70.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 70.0-71.0 sec  24.2 MBytes   203 Mbits/sec
[  4] 71.0-72.0 sec  34.2 MBytes   286 Mbits/sec
[  4] 72.0-73.0 sec  32.1 MBytes   269 Mbits/sec
[  4] 73.0-74.0 sec  32.7 MBytes   274 Mbits/sec
[  4] 74.0-75.0 sec  28.8 MBytes   241 Mbits/sec
[  4] 75.0-76.0 sec  34.2 MBytes   287 Mbits/sec
[  4] 76.0-77.0 sec  26.7 MBytes   224 Mbits/sec
[  4]  0.0-77.9 sec  2.36 GBytes   260 Mbits/sec

```
I will do more measurement if I found an interesting result I will report you.
Thanks
Aman

atyg02, 2017-02-22:
@uminokoe , Here is another measurment with wget, down loading a big file from the internet.
the driver able to bundle LTE (Blue line)[100 Mbps ] + DSL (Yellow) [50 Mbps], in Total (Red line)around 150 Mbps.
[image]
BR
Aman

uminokoe, 2017-03-02:
@atyg02 
Thank you for your effort. I was expecting a speed of at least 300 Mbps actually. I do hope that the speed would be faster on Windows or in future Linux driver releases.

ulli-kroll, 2017-05-03:
Hi guys ..
it can't exceed 300 Mibps 

It only affects rtl8812au and rtl8813/rtl8814au device, these with USB3 support
If someone has a **sane** test setup for speed above 300 Mibps please contact !

I'm currently too lazy for this