diederikdehaas/rtl8812AU #58: Changing channel in monitor mode -> kernel oops

cooperarichardson, 2017-02-19:
When the card is in monitor mode:
```
ifconfig wlan0 up
iwconfig wlan0 channel 6
iwconfig wlan0 mode monitor
```

Running ```iwconfig wlan0 channel 1``` (or any other channel) results in a kernel oops and "Killed" being written in the terminal where iwconfig ran. Tested on Kali 4.8 and 4.9 kernel as well as Ubuntu 16.10 4.8 kernel.

Kernel OOPs (Ubuntu 16.10 with 4.8.0-37-generic 64 bit):
```
[ 1990.445860] RTL871X: rtw_cfg80211_indicate_connect(wlx00c0ca88c197) BSS not found !!
[ 1990.445881] BUG: unable to handle kernel paging request at fffffffffffffff8
[ 1990.445886] IP: [<ffffffff8d23f964>] memcpy_orig+0x54/0x110
[ 1990.445892] PGD 6d609067 PUD 6d60b067 PMD 0 
[ 1990.445896] Oops: 0000 [#1] SMP
[ 1990.445899] Modules linked in: 8812au(OE) nfnetlink_queue nfnetlink_log arc4 mac80211_hwsim mac80211 nfnetlink bluetooth cfg80211 vmw_vsock_vmci_transport vsock vmw_balloon coretemp crct10dif_pclmul crc32_pclmul ghash_clmulni_intel cryptd intel_rapl_perf input_leds joydev serio_raw nfit shpchp i2c_piix4 vmw_vmci mac_hid parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid psmouse vmwgfx ahci libahci e1000 mptspi mptscsih mptbase scsi_transport_spi ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm pata_acpi fjes [last unloaded: 8812au]
[ 1990.445928] CPU: 0 PID: 81919 Comm: iwconfig Tainted: G           OE   4.8.0-37-generic #39-Ubuntu
[ 1990.445930] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 07/02/2015
[ 1990.445931] task: ffffa07934069d80 task.stack: ffffa0790e550000
[ 1990.445932] RIP: 0010:[<ffffffff8d23f964>]  [<ffffffff8d23f964>] memcpy_orig+0x54/0x110
[ 1990.445935] RSP: 0018:ffffa0790e553c30  EFLAGS: 00010286
[ 1990.445936] RAX: ffffa07978438770 RBX: ffffa07978438720 RCX: 0000000000000000
[ 1990.445937] RDX: ffffffffffffffa6 RSI: 0000000000000000 RDI: ffffa07978438756
[ 1990.445938] RBP: ffffa0790e553c78 R08: ffffa0797b61c5e0 R09: 0000000000000000
[ 1990.445939] R10: ffffa07978438720 R11: 000000000000071b R12: ffffa07975d43aa0
[ 1990.445940] R13: 0000000000000000 R14: ffffa07922882800 R15: ffffffffffffffe6
[ 1990.445942] FS:  00007f37e197e700(0000) GS:ffffa0797b600000(0000) knlGS:0000000000000000
[ 1990.445943] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1990.445944] CR2: fffffffffffffff8 CR3: 000000002ba77000 CR4: 00000000000406f0
[ 1990.445998] Stack:
[ 1990.446000]  ffffffffc06898b5 000000000000001e 000000000000001a ffffc219010b50fc
[ 1990.446002]  ffffc219010b5000 ffffa07922882800 ffffa07919f65000 ffffffff8d8e5890
[ 1990.446004]  ffffa0790e553e00 ffffa0790e553cb8 ffffffffc0ab8857 ffffffffffffffe2
[ 1990.446006] Call Trace:
[ 1990.446071]  [<ffffffffc06898b5>] ? cfg80211_connect_bss+0x135/0x1a0 [cfg80211]
[ 1990.446103]  [<ffffffffc0ab8857>] rtw_cfg80211_indicate_connect+0x10c/0x227 [8812au]
[ 1990.446124]  [<ffffffffc0ab40c5>] rtw_os_indicate_connect+0x1d/0x5e [8812au]
[ 1990.446139]  [<ffffffffc0a73269>] rtw_indicate_connect+0x39/0x51 [8812au]
[ 1990.446160]  [<ffffffffc0aafe40>] rtw_wx_set_freq+0x103/0x16c [8812au]
[ 1990.446180]  [<ffffffff8d679402>] ioctl_standard_call+0x52/0xd0
[ 1990.446184]  [<ffffffff8d67a090>] ? iw_handler_get_private+0x60/0x60
[ 1990.446185]  [<ffffffff8d6793b0>] ? ioctl_standard_iw_point+0x3b0/0x3b0
[ 1990.446187]  [<ffffffff8d678a94>] wireless_process_ioctl+0x154/0x190
[ 1990.446188]  [<ffffffff8d679688>] wext_handle_ioctl+0x78/0xd0
[ 1990.446192]  [<ffffffff8d59c97d>] dev_ioctl+0x29d/0x5a0
[ 1990.446195]  [<ffffffff8d55f906>] sock_ioctl+0x126/0x290
[ 1990.446199]  [<ffffffff8d0485b3>] do_vfs_ioctl+0xa3/0x610
[ 1990.446202]  [<ffffffff8ce6b4d6>] ? __do_page_fault+0x266/0x4e0
[ 1990.446203]  [<ffffffff8d048b99>] SyS_ioctl+0x79/0x90
[ 1990.446207]  [<ffffffff8d69c4f6>] entry_SYSCALL_64_fastpath+0x1e/0xa8
[ 1990.446208] Code: 4c 89 07 4c 89 4f 08 4c 89 57 10 4c 89 5f 18 48 8d 7f 20 73 d4 83 c2 20 eb 44 48 01 d6 48 01 d7 48 83 ea 20 0f 1f 00 48 83 ea 20 <4c> 8b 46 f8 4c 8b 4e f0 4c 8b 56 e8 4c 8b 5e e0 48 8d 76 e0 4c 
[ 1990.446227] RIP  [<ffffffff8d23f964>] memcpy_orig+0x54/0x110
[ 1990.446229]  RSP <ffffa0790e553c30>
[ 1990.446230] CR2: fffffffffffffff8
[ 1990.446232] ---[ end trace bc4e82c6ab020840 ]---
```

uminokoe, 2017-02-19:
[astsam's driver](https://github.com/astsam/rtl8812au) claims that it has monitor mode. You may try **at your own risk** to see if it works.

dumper93, 2017-02-21:
It also does not work. It's the same result with astsam's driver.