diederikdehaas/rtl8812AU #61: No networks until connected.

Aboe646, 2017-02-26:
Hey,

I have an Alfa Network AWUS036ACH and use Fedora 25.
It work, it's fast and stable. But unless I'm connected to a Network, I can't see any networks at all in the NetworkManager from Gnome and only a handful when I use iwlist.
When I'm connected to a (known) network I can see a lot more networks!

Greeting, Aboe646

pnpavlov, 2017-04-15:
Hi @Aboe646 , @Turbito , 
I had exactly the same problem on Fedora25 - if i use "connect to hidden network" everything is okay and I can see other networks, but on boot I don't get connected right away and see no networks..

Are you sure you used the 22-beta? It worked for me.
Could you please cross check your commands with mine:
```
# removed the old driver
DRV_NAME=rtl8812AU
DRV_VERSION=4.3.14
dkms remove ${DRV_NAME}/${DRV_VERSION} --all

# installed the 22-beta
git clone -b driver-4.3.22-beta https://github.com/diederikdehaas/rtl8812AU.git
cd rtl8812AU/
DRV_VERSION=4.3.22-beta
mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
dkms install -m ${DRV_NAME} -v ${DRV_VERSION}

# reboot and check if wifi works
reboot 
```

Regards, 
Pavel

Turbito, 2017-04-15:
@pnpavlov I just tried the 22-beta and still the same
Arch Linux up to date, kernel 4.10.8

pnpavlov, 2017-04-16:
I noticed that after last kernel update it stopped working for me too... will share if i find solution... I noticed there are some repositories claiming there is version 5 of the driver... anyone tried it? Does it solve this problem?

pnpavlov, 2017-04-17:
@Turbito , sorry for the delay... I looked further in what's wrong for me and solved it ... again... hope you have same luck! 
Problem for me was that "DRV_NAME=rtl8812AU ; dkms status ${DRV_NAME}" returned more than one item... I had some left-overs from when I played around... Cleaning up  ALL available versions and re-install fixed it. For example
```
DRV_VERSION=(the old ones) 
dkms remove ${DRV_NAME}/${DRV_VERSION} --all
dkms status ${DRV_NAME}
(should return NO results) 
```
then followed the same procedure I described above and after reboot the wifi was OK again... the working setup  on my notebook now is:
```
DRV_NAME=rtl8812AU ; dkms status ${DRV_NAME}
rtl8812AU, 4.3.22-beta, 4.10.9-200.fc25.x86_64, x86_64: installed
```
Regards, Pavel

Aboe646, 2017-04-20:
THX!!!! That solved the problem! You're awesome!

Turbito, 2017-04-21:
Still didn't worked.
I'd notice that all drivers named "8812au" I test fails.
The only one that worked for me is https://github.com/abperiasamy/rtl8812AU_8821AU_linux 
It enable wifi but onyl work with "hidden networks"

Turbito, 2017-04-21:
Ok... was a problem with my model.
my model, archer T4U AC1300 has a bus id diferent from 1200, and that bus id isn't in the file that contain the usb bus id. Then added it and compiled. Now works like a charm.

pnpavlov, 2017-05-03:
@Aboe646 @Turbito  happy to hear that both of your devices work :)

smitgd, 2017-05-10:
dkms status rtl8812AU
rtl8812AU, 4.3.22-beta, 4.10.13-200.fc25.x86_64, x86_64: installed
Thanks here too. Only this working correctly with fedora 25 after trying several others!