diederikdehaas/rtl8812AU #63: New Version: 5.1.5

xxNull-lsk, 2017-03-24:
Download from Tenda.com.cn ,Support USB 3.0
And I modifiles its copy to support Tenda U12
More information at https://github.com/xxNull-lsk/rtl8812AU

uminokoe, 2017-04-02:
The package from Tenda does not contain the _hal_ files for RTL8811AU/RTL8821AU, and therefore can only be used for RTL8812AU.
To try out the 5.1.5 version on the aforementioned chipsets and/or on newer Linux kernels, you may use [my repository](https://github.com/uminokoe/rtl8812AU/tree/driver-5.1.5) **at your own risk**.
Note that the missing hal files are imported from 4.3.22-beta, and part of the code which addresses the "2.4G band switch" problem have to be commented out because of this.

ojnickel, 2017-04-05:
what about support for rpi with this version?

mk-fg, 2017-04-05:
@ojnickel 
Building on RPi is already supported in this repo, you just need to enable it, like e.g. https://aur.archlinux.org/cgit/aur.git/tree/rpi.diff?h=rtl8812au-rpi-dkms-git patch does.

Wrt 5.1.5 specifically - seem to work fine on RPi3 for me, with ~230 Mbits/s from iperf3/tcp between two boards using ALFA AWUS036AC dongles (hostapd + wpa_supplicant).
I've built and installed driver from https://github.com/mk-fg/rtl8812au, which is a fork of @uminokoe's driver-5.1.5 branch with some patches (e.g. for RPi build) cherry-picked on top, and probably won't be necessary if/when this stuff will be merged here.

mk-fg, 2017-04-05:
Btw, if you're looking to get 802.11ac bandwidth out of RPi3 with this chip, it probably won't happen due to USB there being ~35 MiB/s (~270 Mbits/s) bottleneck, at least when I tested it.

uminokoe, 2017-04-06:
@mk-fg 
Thanks for testing. I see that you have a [commit](https://github.com/mk-fg/rtl8812au/commit/f98af172d6dcac24dd2c18a35e72e08164794566) that disables regulatory domain adjustments.
Will that commit solve #35 ? I have always wanted to fix the problem while ensuring the driver complies with the laws and regulations regarding the use of wireless channels.

mk-fg, 2017-04-06:
@uminokoe
No, unfortunately, I don't think so.

It's based on your proposed fix with `CONFIG_DISABLE_REGD_C` in #22.
I've looked at the code in `wifi_regd.c`, saw that it only applies further restrictions to passed reg-domain and went on to disable the code entirely.

Was testing forwarding between multiple dongles yesterday, and was frustrated by channel selection, which was mostly caused by my initial lack of understanding of how `vht_oper_centr_freq_seg0_idx` option works in hostapd.conf - it's not well-documented, but essential for manual 802.11ac channel picks.

Did also patch db.txt to disable all no-IR/DFS flags, so wasn't worried about regulatory domain there at all, unfortunately, and that commit most certainly is not made with compliance in mind.
I don't know much about regdb stuff either, haven't seen #35, and don't really see why would driver want to change passed restrictions, but suspect more info can be found looking at mainline drivers if they do same thing (and if not, there's your answer too).

Thanks a lot for your work on this driver, btw!

uminokoe, 2017-04-06:
By the way, my adapter is RTL8811AU based which is 433 Mbps and USB 2.0. When I use [Zoltán Pallagi's app](https://play.google.com/store/apps/details?id=com.pzolee.android.localwifispeedtester&hl=en) to test the speed between my computer (Linux, x86_64) and my phone, KSysGuard shows speeds of around ~20 MiB/s.
[image]
But on my phone, the speeds shown by the app are always below 200 Mbit/s.
[image]
I wonder if it is a unit conversion issue. The speed shown by KSysGuard seems to be reasonable because it is close to the maximum actual speed of USB 2.0. When performing `dd` tests on a USB 3.0 flash drive connected to a USB 2.0 port, the speed is also around 23 MiB/s.

mk-fg, 2017-04-06:
I'm a bit confused about what's the issue with ~20 MiB/s vs ~160 Mbit/s, because 20 MiB * 8 = 160 Mbits/s (1 B = 8 bits), i.e. these are exactly same numbers, and there don't seem to be any difference there.

mk-fg, 2017-04-06:
> When performing dd tests on a USB 3.0 flash drive connected to a USB 2.0 port, the speed is also around 23 MiB/s.

I wouldn't take just that as an indication of USB bus speed, as it can be just a speed at which USB drive works with USB 2.0 interface.

My initial method of testing was plugging cardreaders with ~30 MiB/s speed each into RPi3 until speed on all of them starts to drop.
E.g. with RPi3, running `pv -rb -i3 </dev/sda >/dev/null` shows ~31 MiB/s on cr-1 and similar speed on cr-2, when they're used separately, but plugging in both at the same time gives only ~35 MiB/s sum, with each doing about half of max speed.
Also, `lsusb -t` shows that it's same USB host, and building IO Graph from e.g. `tcpdump -i usbmon1 -nn -s64 -w /tmp/usb.pcap` in wireshark shows that there are no free time-slots on USB bus, so no more throughput possible ("send" and "recv" are pretty much same on USB, as host initiates everything in a blocking manner).

With desktop, probably worth checking if there might be several different USB buses (as traffic on one wouldn't affect the other), and maybe doing something similar, i.e. find a point where bandwidth of individual devices drops, and make sure it's not CPU, disk or some other obvious bottleneck.

A bit unrelated to this issue though, my apologies.

kmaterka, 2017-04-13:
5.1.5 fixed really annoying issue for me:
https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/190
From time to time my Asus USB-AC56 hangs with error message in logs:
"RTL871X: There are some pkts to transmit
RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0"

@uminokoe I've used driver from your repository, it compiles fine, I had to change Makefile slightly to make it work correctly with dkms

sopparus, 2017-04-24:
amazing! thanks @uminokoe, first time my wifi is stable and does break becouse of https://github.com/Grawp/rtl8812au_rtl8821au/issues/39

smitgd, 2017-05-13:
Thanks also here. 4.3.22-beta was much better but still failed at 5GHz. This, so far, working find at both frequencies and seems faster.
Question: Do you know if Realtek provides public links to their driver? Or maybe they only provide the source to device makers that then release the gpl driver (such as Tenda). I couldn't find any such link for this driver.

xxNull-lsk, 2017-05-13:
This URL from Tengda:
http://down.tenda.com.cn/uploadfile/2017/Drive/U12_linux_v5.1.5_19247.20160830.rar
And, I don't known any URL from Realtek about their driver.

smitgd, 2017-05-13:
Yes, I saw that link that you provided before. At realtek site there is no download for the 5.1.5 driver AFAICT. Guess they keep it proprietary except for their chip customers, e.g., Tenda. Also, nothing on their ftp site. Also, I meant to say above that the 5.1.5 driver updated by @uminokoe is working well on both bands (lost track of which fork I was on, sorry).

uminokoe, 2017-05-16:
Realtek has not disclosed any links for the RTL8812AU driver (source code). However, on the Internet and through the use of search engines, I notice that there are results which contain a link to download the driver from Realtek. Some of the websites containing the link are wholesalers selling RTL8812AU chips. It turns out that the latest version is named "5.1.5.1_19247.20160830", but the source code is the same as the "5.1.5" by Tenda (, i.e. 5.1.5.1 == 5.1.5).

It seems that Tenda's source code package is up-to-date. Only when we find that the code is too old or buggy and we are unable to find a fix / newer version, we should consider this "extreme" method.

smitgd, 2017-05-17:
Thanks for the info. Also, your version of 5.1.5 is working better than any of the several forks on github. Would be nice if this was actually made part of the official linux kernel so it would be updated and tested as the kernel changes.

fumantsu, 2017-05-17:
Hi,

The 5.1.5 works flawless with dkms? I didn't see any dkms.conf.

mk-fg, 2017-05-17:
> The 5.1.5 works flawless with dkms? I didn't see any dkms.conf.

Yup, no changes there, can probably grab the conf from this repo.

kmaterka, 2017-05-17:
According to DKMS I used config file from:
https://github.com/abperiasamy/rtl8812AU_8821AU_linux
I also had to modify Makefile because it is not working with DKMS. I cherry-picked DKMS fixes from abperiasamy but it is not working flawlessly for me - probably my mistake.

kmaterka, 2017-05-17:
Thanks, I was able to fix my dmks.conf and now changes to Makefile are not needed. This works on Ubuntu 16.04. I've also changed module name from _8812au_ to _rtl8812au_.

Source in:
`/usr/src/rtl8812au-5.1.5`

To install:
```
dkms add -m rtl8812au -v 5.1.5
dkms build rtl8812au/5.1.5 
dkms install rtl8812au/5.1.5 --all
depmod
modprobe rtl8812au
```

**dkms.conf**
```
PACKAGE_NAME="rtl8812au"
PACKAGE_VERSION="5.1.5"
MAKE[0]="'make' KVER=${kernelver} KSRC=/lib/modules/${kernelver}/build USER_MODULE_NAME=$PACKAGE_NAME"
CLEAN="make clean"
BUILT_MODULE_NAME[0]="rtl8812au"
DEST_MODULE_LOCATION[0]="/kernel/drivers/net/wireless"
AUTOINSTALL="yes"
REMAKE_INITRD=no
```

xxNull-lsk, 2017-05-18:
@kmaterka Thanks. And, it works on Ubuntu 14.04 also.

fumantsu, 2017-05-19:
For RPI3 with kernel  4.9.24-v7+ doesn't work. The dkms.conf which was provided here gives this error:
Makefile:628: arch//Makefile: No such file or directory
The file indeed doesn't exist but doesn't exist in any headers so....

If I change the file and put from the different driver versions I get this:
make ARCH= CROSS_COMPILE= -C  M=/var/lib/dkms/rtl8812au/5.1.5/build  modules
make[1]: *** M=/var/lib/dkms/rtl8812au/5.1.5/build: No such file or directory.
which is weird because the directory exists....

everything is been done as root

mk-fg, 2017-08-31:
Fwiw, there was a next release of 5.2.9 from realtek ftp, which can be found at e.g. https://github.com/gordboy/rtl8812au with more direct download link mentioned in https://github.com/astsam/rtl8812au/issues/51#issuecomment-315117972

dernyn, 2017-11-09:
@mk-fg 
the current raspbian build of stretch 4.9.60+ makes a rpi compile, much harder...it's not just flags on the makefile.

mk-fg, 2017-11-09:
@dernyn what?

dernyn, 2017-11-09:
sorry for the weird way that came of on, I was doing too many things at the same time, I found this while doing some searches on search engines, I trying to create a better way to compile drivers for raspbian... I found your comment that all that was needed to compile on the rpi to provide support was the change or adaptation of y/n flags on the makefile....I was conveying that this is way much harder ,especially on current raspbian images like stretch 4.9.60+ , with the reduction of supporting kernel modules

mk-fg, 2017-11-09:
Ah, yeah, I haven't tried to build it for raspbian, so have no idea.

conrad10781, 2017-12-01:
This release resolves the same issue @kmaterka reported on https://github.com/diederikdehaas/rtl8812AU/issues/63#issuecomment-294019355 , that I had with a card I had purchased off of Amazon ( https://www.amazon.com/gp/product/B01IEU7UZ0 )

mk-fg, 2017-12-01:
It's a bit weird that @uminokoe, who did bulk of the work on 5.1.5 (fixing VHT and hal stuff there), had a bunch of comments here, which are now all gone, as well as his github account.
Then @gordboy found 5.2.9 and polished it into usable state, and now also gone from github, leaving only "ghost" comments behind.
Digging up new releases of rtl8812au breaks people, apparently :)