diederikdehaas/rtl8812AU #64: wireless discontinue after a while on opensuse 42.2 kernel 4.4.57 version 4.3.14 and 4.3.20

myljimcia, 2017-04-10:
Hi,

Thank you very much for providing the driver for rtl8812au. However, I met a problem with the driver. Now I use the version 4.3.14 on a kernel 4.4.57 system opensuse 42.2 computer. At first , the wireless works well.  After a while, it depends, rtl8812au can't scan any wireless net and I can see the device in the networkmanager. After I re-plugged the devices, it works again. Could you help me solve the problem? If you need anything, just tell me.

Thank you very much

myljimcia, 2017-04-10:
I find the cause. The driver is NOT compatible with GOOGLE CHROME 57. After I install Google Chrome, the wireless net will be unstable. I uninstall Google Chrome and it's all depends. The net work is normal

myljimcia, 2017-04-10:
It is might not be caused by google chrome, after I uninstall all the thing about google chrome, the condition happens again

kmaterka, 2017-04-13:
Probably the same issue as this one, I also had disconnects and and was unable to scan for networks:
https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/190
@myljimcia If you can try 5.1.5 version, more details in the link above.