diederikdehaas/rtl8812AU #67: Fedora 25 and Kernel 4.10.9 and higher, no 5Ghz only 2,4Ghz

przerada, 2017-04-22:
I have Asus RT68AC Dual-Band and I was using 5ghz network on kernel 4.10.8, after update kernel to 4.10.9 5ghz wasnt seen.
I was update kernel to 4.10.10 and remove driver and install again, it dont give me anything so I was update kernel to 4.10.11 and make this same, I can only use 2,4Ghz on my Edimax EW-7811UAC.
I was using driver 4.3.14 and 4.3.22-beta

smitgd, 2017-05-12:
For me, can connection with 5GHz but quickly locks and loses the network and can no longer choose 2.4GHz. This is Fedora 25 with kernal 4.10.14-200.fc25.x86_64. Haven't tried earlier kernels. rmmod 8812au ; modprobe 8812au brings it back so I can use 2.4GHz.

Id2ndR, 2017-06-07:
5Ghz works for me in Fedora 26 with 4.11.3-302.

closed 2017-06-08 by przerada