diederikdehaas/rtl8812AU #69: Module for CentOS 7 (RHEL)

chadjordan1, 2017-05-02:
Greetings!

I have been attempting to install your RTL8812AU module onto my CentOS 7 (host) which seems to work until reaching the compile portion, i.e., make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.10.0-514.10.2.el7.x86_64/build M=/root/rtl8812AU  modules.  I'm familiar enough with linux to figure things out, however, there appears to be an error associated with the compiling that may or may not be attributed to the linux flavor I'm using (e.g., CentOS 7) and/or a possible conflict with the latest kernel-devel version that I have installed.  With that said, I was hoping you could provide some suggestions on how I could troubleshoot this particular problem that may lead to a successful install on my CentOS 7 system.  Any assistance would be much appreciated!

Below lists some data you may want to know about before assisting my with this:

1.  Installation ran using 'root'
2. Began install using the git URL -
    
git clone https://github.com/diederikdehaas/rtl8812AU.git && cd rtl8812AU && make && sudo make install -y

3. PC's response...

Cloning into 'rtl8812AU'...
remote: Counting objects: 2142, done.
remote: Total 2142 (delta 0), reused 0 (delta 0), pack-reused 2142
Receiving objects: 100% (2142/2142), 7.57 MiB | 585.00 KiB/s, done.
Resolving deltas: 100% (1170/1170), done.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.10.0-514.10.2.el7.x86_64/build M=/root/rtl8812AU  modules
make: *** /lib/modules/3.10.0-514.10.2.el7.x86_64/build: No such file or directory.  Stop.
make: *** [modules] Error 2
[root@J-SERVER rtl8812AU]# make: *** /lib/modules/3.10.0-514.10.2.el7.x86_64/build: No such file or directory.  Stop.
-bash: make:: command not found

4. Verified which module was on my system -

[root@J-SERVER rtl8812AU]# ls -l /lib/modules/3.10.0-123.el7.x86_64
ls: cannot access /lib/modules/3.10.0-123.el7.x86_64: No such file or directory

[root@J-SERVER rtl8812AU]# ls -al
total 120
drwxr-xr-x. 8 root root  4096 May  2 02:10 .
dr-xr-x---. 6 root root  4096 May  2 02:09 ..
-rw-r--r--. 1 root root    64 May  2 02:10 clean
drwxr-xr-x. 3 root root  4096 May  2 02:10 core
-rw-r--r--. 1 root root   183 May  2 02:10 dkms.conf
drwxr-xr-x. 8 root root  4096 May  2 02:10 .git
-rw-r--r--. 1 root root   476 May  2 02:10 .gitignore
drwxr-xr-x. 8 root root  4096 May  2 02:10 hal
-rw-r--r--. 1 root root    51 May  2 02:10 ifcfg-wlan0
drwxr-xr-x. 4 root root  8192 May  2 02:10 include
-rw-r--r--. 1 root root   110 May  2 02:10 Kconfig
-rwxr-xr-x. 1 root root 49452 May  2 02:10 Makefile
drwxr-xr-x. 3 root root    40 May  2 02:10 os_dep
drwxr-xr-x. 2 root root  4096 May  2 02:10 platform
-rw-r--r--. 1 root root  2325 May  2 02:10 README.md
-rw-r--r--. 1 root root   423 May  2 02:10 runwpa
-rw-r--r--. 1 root root   294 May  2 02:10 wlan0dhcp

[root@J-SERVER rtl8812AU]# ls -l /lib/modules/3.10.0-514.10.2.el7.x86_64
total 2944
lrwxrwxrwx.  1 root root     43 Apr  2 20:47 build -> /usr/src/kernels/3.10.0-514.10.2.el7.x86_64
drwxr-xr-x.  2 root root      6 Mar  2 17:17 extra
drwxr-xr-x. 12 root root   4096 Apr  2 20:47 kernel
-rw-r--r--.  1 root root 762886 Apr  2 20:50 modules.alias
-rw-r--r--.  1 root root 735054 Apr  2 20:50 modules.alias.bin
-rw-r--r--.  1 root root   1326 Mar  2 17:17 modules.block
-rw-r--r--.  1 root root   6227 Mar  2 17:15 modules.builtin
-rw-r--r--.  1 root root   8035 Apr  2 20:50 modules.builtin.bin
-rw-r--r--.  1 root root 240071 Apr  2 20:50 modules.dep
-rw-r--r--.  1 root root 343333 Apr  2 20:50 modules.dep.bin
-rw-r--r--.  1 root root    361 Apr  2 20:50 modules.devname
-rw-r--r--.  1 root root    132 Mar  2 17:17 modules.drm
-rw-r--r--.  1 root root    110 Mar  2 17:17 modules.modesetting
-rw-r--r--.  1 root root   1580 Mar  2 17:17 modules.networking
-rw-r--r--.  1 root root  90643 Mar  2 17:15 modules.order
-rw-r--r--.  1 root root     89 Apr  2 20:50 modules.softdep
-rw-r--r--.  1 root root 350869 Apr  2 20:50 modules.symbols
-rw-r--r--.  1 root root 432777 Apr  2 20:50 modules.symbols.bin
lrwxrwxrwx.  1 root root      5 Apr  2 20:47 source -> build
drwxr-xr-x.  2 root root      6 Mar  2 17:17 updates
drwxr-xr-x.  2 root root     91 Apr  2 20:47 vdso
drwxr-xr-x.  2 root root      6 Mar  2 17:17 weak-updates

MrEngman, 2017-05-02:
Does the kernel source file that `build` points to, `/usr/src/kernels/3.10.0-514.10.2.el7.x86_64`, exist?

chadjordan1, 2017-05-02:
Yes.  I believe so.  I ran the following and it returned that path.

[root@J-SERVER] /]# ls /usr/src/kernels/3.10.0-514.10.2.el7.x86_64
/usr/src/kernels/3.10.0-514.10.2.el7.x86_64

chadjordan1, 2017-05-02:
I may have found the issue to the problem I’m having which most likely revolves around a compiling problem that the devs may have accidently caused and were not aware of when releasing new kernel versions ever since 3.10.0-123.X was surpassed.  According to a few forums, there’s been talk about Centos/RedHat developers, and even Ubuntu developers  screwing up the kernel headers which breaks the system in this case, and the best workaround would be to fix it for ourselves by editing “base.c” at line 472:

[FROM]

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 11, 0))

[TO]

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 10, 0))

URL Source: Here<https://github.com/lwfinger/rtlwifi_new/issues/40>

Anyway, I need a bit of sleep before making those changes, but in the meantime, if you have any feedback about this or suggestions then I’ll take a read at it when I return.  Thx.

Chad