diederikdehaas/rtl8812AU #70: Build error on latest RasPi kernel

austinmarkus, 2017-05-05:
Hi,

I'm getting a build error on the Raspbian 4.9.24-v7+ kernel. I've tried both the 4.3.14-9 and 4.3.22-beta-6 releases with the same errors. Log from build is below. Thanks for any help, as it used to work wonderfully prior to the latest update.

-Austin

----

  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.o
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:92:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:150:2: note: in expansion of macro ‘CHAN2G’
  CHAN2G(1, 2412, 0),
  ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:101:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:167:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34, 0),  CHAN5G(36, 0),
  ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:167:2: error: initializer element is not constant
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:167:2: error: (near initialization for ‘rtw_5ghz_a_channels[0].band’)
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:167:2: error: initializer element is not constant
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:167:2: error: (near initialization for ‘rtw_5ghz_a_channels[1].band’)
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:168:2: error: initializer element is not constant
  CHAN5G(38, 0),  CHAN5G(40, 0),
  ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:168:2: error: (near initialization for ‘rtw_5ghz_a_channels[2].band’)
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:168:2: error: initializer element is not constant
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:168:2: error: (near initialization for ‘rtw_5ghz_a_channels[3].band’)
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:169:2: error: initializer element is not constant
  CHAN5G(42, 0),  CHAN5G(44, 0),
  ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:169:2: error: (near initialization for ‘rtw_5ghz_a_channels[4].band’)
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:169:2: error: initializer element is not constant
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:169:2: error: (near initialization for ‘rtw_5ghz_a_channels[5].band’)
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:170:2: error: initializer element is not constant
  CHAN5G(46, 0),  CHAN5G(48, 0),
  ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:170:2: error: (near initialization for ‘rtw_5ghz_a_channels[6].band’)
[... 175 lines cut ...]
   wiphy->bands[IEEE80211_BAND_2GHZ] = NULL;
               ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:6559:18: error: array subscript is not an integer
  if (wiphy->bands[IEEE80211_BAND_5GHZ]) {
                  ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:6560:33: error: array subscript is not an integer
   rtw_spt_band_free(wiphy->bands[IEEE80211_BAND_5GHZ]);
                                 ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:6560:21: warning: passing argument 1 of ‘rtw_spt_band_free’ from incompatible pointer type
   rtw_spt_band_free(wiphy->bands[IEEE80211_BAND_5GHZ]);
                     ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:271:6: note: expected ‘struct ieee80211_supported_band *’ but argument is of type ‘struct ieee80211_channel *’
 void rtw_spt_band_free(struct ieee80211_supported_band *spt_band)
      ^
/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.c:6561:15: error: array subscript is not an integer
   wiphy->bands[IEEE80211_BAND_5GHZ] = NULL;
               ^
cc1: some warnings being treated as errors
scripts/Makefile.build:293: recipe for target '/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build/os_dep/linux/ioctl_cfg80211.o] Error 1
Makefile:1490: recipe for target '_module_/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8812AU/4.3.22-beta-6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.24-v7+'
Makefile:1693: recipe for target 'modules' failed
make: *** [modules] Error 2

MrEngman, 2017-05-05:
Compiles OK for me. I do get a couple of warnings but the compile is OK, however, I cross-compile on a PC.

The error referencing IEEE80211_BAND_2GHZ and IEEE80211_BAND_5GHZ would suggest to me your source code is not up to date. Check file `include/rtw_wifi_regd.h` and see if it includes
```#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 7, 0))
#define ieee80211_band nl80211_band
#define IEEE80211_BAND_2GHZ NL80211_BAND_2GHZ
#define IEEE80211_BAND_5GHZ NL80211_BAND_5GHZ
#define IEEE80211_NUM_BANDS NUM_NL80211_BANDS
#endif
```
If not your code needs updating.

austinmarkus, 2017-05-05:
Thank you for your quick response, I was using your release tarballs and not a git clone. Updating to the latest clone solved the issue. You might want to consider updating your releases to reflect the changes.

Thanks again,

Austin

closed 2017-05-05 by austinmarkus