diederikdehaas/rtl8812AU #71: Compiling on 4.11 kernel fails

godlikesonny, 2017-05-08:
Hello!

It looks like you're not able to compile the driver on 4.11 systems anymore. 4.10 systems work just fine, albeit a few warnings. Tried all versions from GitHub, neither of them seems to work. Same output. Tested on ubuntu 16.10 and 17.04. I would really appreciate some help. Output is as follows:

ubuntu@ubuntu:/media/ubuntu/USB/E200HA/rtl8812AU$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.0-041100-generic/build M=/media/ubuntu/USB/E200HA/rtl8812AU modules
make[1]: Entering directory '/usr/src/linux-headers-4.11.0-041100-generic'
CC [M] /media/ubuntu/USB/E200HA/rtl8812AU/core/rtw_cmd.o
In file included from /media/ubuntu/USB/E200HA/rtl8812AU/include/drv_types.h:32:0,
from /media/ubuntu/USB/E200HA/rtl8812AU/core/rtw_cmd.c:22:
/media/ubuntu/USB/E200HA/rtl8812AU/include/osdep_service.h: In function ‘thread_enter’:
/media/ubuntu/USB/E200HA/rtl8812AU/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
allow_signal(SIGTERM);
^~~~~~~~~~~~
/media/ubuntu/USB/E200HA/rtl8812AU/include/osdep_service.h: In function ‘flush_signals_thread’:
/media/ubuntu/USB/E200HA/rtl8812AU/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
if (signal_pending (current))
^~~~~~~~~~~~~~
/media/ubuntu/USB/E200HA/rtl8812AU/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
flush_signals(current);
^~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/media/ubuntu/USB/E200HA/rtl8812AU/core/rtw_cmd.o' failed
make[2]: *** [/media/ubuntu/USB/E200HA/rtl8812AU/core/rtw_cmd.o] Error 1
Makefile:1498: recipe for target '_module_/media/ubuntu/USB/E200HA/rtl8812AU' failed
make[1]: *** [_module_/media/ubuntu/USB/E200HA/rtl8812AU] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.11.0-041100-generic'
Makefile:1576: recipe for target 'modules' failed
make: *** [modules] Error 2

maurossi, 2017-05-13:
Build tested with kernel-4.12rc1

https://github.com/maurossi/rtl8812au/commits/kernel-4.12

godlikesonny, 2017-05-13:
First of all thanks for the help! Compiling now works, although connecting does not. system log says wrong key, even tho its correct. Here's th complete output:

May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8860] keyfile: update /etc/NetworkManager/system-connections/Sonny (2de61c5b-cab2-43a2-92aa-fc2ca016bd9d,"Sonny")
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8913] device (wlx00e04c3280e2): state change: need-auth -> prepare (reason 'none') [60 40 0]
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8926] device (wlx00e04c3280e2): state change: prepare -> config (reason 'none') [40 50 0]
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8932] device (wlx00e04c3280e2): Activation: (wifi) connection 'Sonny' has security, and secrets exist.  No new secrets needed.
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8934] Config: added 'ssid' value 'Sonny'
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8934] Config: added 'scan_ssid' value '1'
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8935] Config: added 'key_mgmt' value 'WPA-PSK'
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8935] Config: added 'auth_alg' value 'OPEN'
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.8935] Config: added 'psk' value '<omitted>'
May 13 23:04:51 ubuntu NetworkManager[1428]: <info>  [1494716691.9048] sup-iface[0x563feb891ce0,wlx00e04c3280e2]: config: set interface ap_scan to 1
May 13 23:04:52 ubuntu NetworkManager[1428]: <info>  [1494716692.3344] device (wlx00e04c3280e2): supplicant interface state: inactive -> scanning
May 13 23:04:52 ubuntu kernel: [  791.247496] RTL871X: nolinked power save leave
May 13 23:04:56 ubuntu wpa_supplicant[1922]: wlx00e04c3280e2: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
May 13 23:04:56 ubuntu wpa_supplicant[1922]: wlx00e04c3280e2: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
May 13 23:04:56 ubuntu wpa_supplicant[1922]: wlx00e04c3280e2: Trying to associate with c4:27:95:bb:8a:ae (SSID='Sonny' freq=2437 MHz)
May 13 23:04:56 ubuntu kernel: [  795.300101] RTL871X: rtw_set_802_11_connect(wlx00e04c3280e2)  fw_state=0x00000008
May 13 23:04:56 ubuntu NetworkManager[1428]: <info>  [1494716696.4101] device (wlx00e04c3280e2): supplicant interface state: scanning -> associating
May 13 23:04:56 ubuntu kernel: [  795.388653] RTL871X: start auth
May 13 23:04:56 ubuntu kernel: [  795.391407] RTL871X: auth success, start assoc
May 13 23:04:56 ubuntu kernel: [  795.397204] RTL871X: rtw_cfg80211_indicate_connect(wlx00e04c3280e2) BSS not found !!
May 13 23:04:56 ubuntu kernel: [  795.397222] RTL871X: assoc success
May 13 23:04:56 ubuntu wpa_supplicant[1922]: wlx00e04c3280e2: Associated with c4:27:95:bb:8a:ae
May 13 23:04:56 ubuntu NetworkManager[1428]: <info>  [1494716696.4898] device (wlx00e04c3280e2): supplicant interface state: associating -> associated
May 13 23:04:56 ubuntu kernel: [  795.409097] ath: EEPROM regdomain: 0x8114
May 13 23:04:56 ubuntu kernel: [  795.409104] ath: EEPROM indicates we should expect a country code
May 13 23:04:56 ubuntu kernel: [  795.409106] ath: doing EEPROM country->regdmn map search
May 13 23:04:56 ubuntu kernel: [  795.409108] ath: country maps to regdmn code: 0x37
May 13 23:04:56 ubuntu kernel: [  795.409110] ath: Country alpha2 being used: DE
May 13 23:04:56 ubuntu kernel: [  795.409111] ath: Regpair used: 0x37
May 13 23:04:56 ubuntu kernel: [  795.409115] ath: regdomain 0x8114 dynamically updated by country IE
May 13 23:04:56 ubuntu wpa_supplicant[1922]: wlx00e04c3280e2: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=DE
May 13 23:04:56 ubuntu kernel: [  795.414316] RTL871X: recv eapol packet
May 13 23:04:56 ubuntu kernel: [  795.414909] RTL871X: send eapol packet
May 13 23:04:56 ubuntu NetworkManager[1428]: <info>  [1494716696.5064] device (wlx00e04c3280e2): supplicant interface state: associated -> 4-way handshake
May 13 23:04:57 ubuntu kernel: [  796.251929] RTL871X: recv eapol packet
May 13 23:04:57 ubuntu kernel: [  796.252643] RTL871X: send eapol packet
May 13 23:04:58 ubuntu kernel: [  797.152673] RTL871X: recv eapol packet
May 13 23:04:58 ubuntu kernel: [  797.153368] RTL871X: send eapol packet
May 13 23:04:59 ubuntu kernel: [  798.052594] RTL871X: recv eapol packet
May 13 23:04:59 ubuntu kernel: [  798.053344] RTL871X: send eapol packet
May 13 23:05:00 ubuntu kernel: [  798.952400] RTL871X: recv eapol packet
May 13 23:05:00 ubuntu kernel: [  798.952931] RTL871X: send eapol packet
May 13 23:05:00 ubuntu kernel: [  799.852811] RTL871X: recv eapol packet
May 13 23:05:00 ubuntu kernel: [  799.853315] RTL871X: send eapol packet
May 13 23:05:01 ubuntu kernel: [  800.752436] RTL871X: recv eapol packet
May 13 23:05:01 ubuntu kernel: [  800.752923] RTL871X: send eapol packet
May 13 23:05:02 ubuntu kernel: [  801.652347] RTL871X: recv eapol packet
May 13 23:05:02 ubuntu kernel: [  801.653287] RTL871X: send eapol packet
May 13 23:05:03 ubuntu kernel: [  802.550398] RTL871X: OnDeAuth(wlx00e04c3280e2) reason=15, ta=c4:27:95:bb:8a:ae, ignore=0
May 13 23:05:03 ubuntu wpa_supplicant[1922]: wlx00e04c3280e2: CTRL-EVENT-DISCONNECTED bssid=c4:27:95:bb:8a:ae reason=0 locally_generated=1
May 13 23:05:03 ubuntu wpa_supplicant[1922]: wlx00e04c3280e2: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
[... cut ...]
May 13 23:05:03 ubuntu NetworkManager[1428]: <info>  [1494716703.6493] device (wlx00e04c3280e2): state change: config -> need-auth (reason 'supplicant-disconnect') [50 60 8]
May 13 23:05:03 ubuntu wpa_supplicant[1922]: wlx00e04c3280e2: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
May 13 23:05:03 ubuntu NetworkManager[1428]: <info>  [1494716703.7469] device (wlx00e04c3280e2): supplicant interface state: disconnected -> inactive
May 13 23:05:03 ubuntu nm-applet[2501]: No keyring secrets found for Sonny/802-11-wireless-security; asking user.

MarcosCC, 2017-10-29:
I have the same issue as godlikesonny in kernel 4.13: compiling works, but fails to connect:

Oct 29 14:21:14 localhost kernel: [156399.111913] RTL871X: start auth
Oct 29 14:21:15 localhost kernel: [156400.063048] RTL871X: auth success, start assoc
Oct 29 14:21:15 localhost kernel: [156400.066300] RTL871X: rtw_cfg80211_indicate_connect(wlx38d54706888f) BSS not found !!
Oct 29 14:21:15 localhost kernel: [156400.066331] RTL871X: assoc success
Oct 29 14:21:15 localhost wpa_supplicant[930]: wlx38d54706888f: Associated with e8:fc:af:f7:47:79
Oct 29 14:21:15 localhost NetworkManager[7613]: <info>  [1509258075.7897] device (wlx38d54706888f): supplicant interface state: associating -> associated
Oct 29 14:21:15 localhost kernel: [156400.088164] RTL871X: recv eapol packet
Oct 29 14:21:15 localhost kernel: [156400.088646] RTL871X: send eapol packet
Oct 29 14:21:15 localhost NetworkManager[7613]: <info>  [1509258075.8114] device (wlx38d54706888f): supplicant interface state: associated -> 4-way handshake
Oct 29 14:21:18 localhost kernel: [156403.064416] RTL871X: recv eapol packet
Oct 29 14:21:18 localhost kernel: [156403.064790] RTL871X: send eapol packet
Oct 29 14:21:19 localhost kernel: [156404.059692] RTL871X: recv eapol packet
Oct 29 14:21:19 localhost kernel: [156404.060060] RTL871X: send eapol packet
Oct 29 14:21:22 localhost kernel: [156407.035503] RTL871X: recv eapol packet
Oct 29 14:21:22 localhost kernel: [156407.035788] RTL871X: send eapol packet
Oct 29 14:21:23 localhost kernel: [156408.030722] RTL871X: OnDeAuth(wlx38d54706888f) reason=15, ta=e8:fc:af:f7:47:79, ignore=0
Oct 29 14:21:23 localhost wpa_supplicant[930]: wlx38d54706888f: CTRL-EVENT-DISCONNECTED bssid=e8:fc:af:f7:47:79 reason=0 locally_generated=1
Oct 29 14:21:23 localhost wpa_supplicant[930]: wlx38d54706888f: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
Oct 29 14:21:23 localhost wpa_supplicant[930]: wlx38d54706888f: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="helios2" auth_failures=1 duration=10 reason=WRONG_KEY
Oct 29 14:21:23 localhost wpa_supplicant[930]: wlx38d54706888f: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="helios2" auth_failures=2 duration=20 reason=CONN_FAILED

maurossi, 2017-10-29:
Hi, if you are trying the driver with Ubuntu 17.04 or other distribution with similar please check the following:

https://github.com/astsam/rtl8812au/issues/36

Please consider that I never tested the driver, I just added it to our android-x86 linux kernel rc cycle builds, so please provide feedback if the problem is solved.
Mauro

MarcosCC, 2017-10-29:
Hi, this is Debian buster... and the solution in that thread works!!!

NM tries to change MAC on every connect but something goes wrong.
Add the following lines
[device]
wifi.scan-rand-mac-address=no
at the end of the /etc/NetworkManager/NetworkManager.conf
and restart NetworkManager
sudo service NetworkManager restart

I am so happy!

Thanks for your work :)