diederikdehaas/rtl8812AU #72: DKMS doesn't rebuild rtl8812au when kernel is updated

Markussss, 2017-05-21:
Hi,
I'm not sure if this is a problem with this module or with something else, but I have no idea where I could ask instead of right here.

I'm using Fedora 25, and usually upgrade my kernel whenever there is a new version. Whenever I upgrade my kernel, I have to remove, recompile and re-add the module to the new kernel. I thought the reason to use DKMS was to not have to recompile kernel modules manually, so I'm confused about where I'm fucking up. Can anybody help me?

sirgrej, 2017-06-06:
I have the same problem

Id2ndR, 2017-06-07:
Solution is to add `KVER=${kernelver}` in make command in dkms.conf. Read comment 2 from [this answer](https://askubuntu.com/a/832372).

diederikdehaas, 2018-03-12:
@Markussss it may have been due to the driver not supporting a new(er) kernel version.
I just updated the 4.3.20 branch so it has support till kernel 4.15. Could you try that one?