diederikdehaas/rtl8812AU #73: The install instructions aren't clear at all

gaining, 2017-05-31:
When trying to install via dkms, 

I try: 
`git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}` but I get this output:
`fatal: Not a git repository (or any of the parent directories): .git
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors`

gaining, 2017-05-31:
Could you possibly write a script that automates the dkms install? I got past step 4 as I've realized I'm not using an archive but a clone of the repository but now I'm stuck on step 5.

sirgrej, 2017-06-06:
dont forget "sudo" in front of git und tar
try 
` sudo git archive driver-${DRV_VERSION} | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}`

diederikdehaas, 2018-03-12:
only on the `tar ...` part, but that's indeed a crucial thing

You do need to perform the `git archive` step in the (root) directory of the driver code.
Thus if you do `git clone https://github.com/diederikdehaas/rtl8812AU`, you then need to do `cd rtl8812AU` and then you do the `git archive` step

Teiby, 2018-07-10:
`root@raspberrypi:/home/pi/rtl8812AU# ls
clean  core  dkms.conf  hal  ifcfg-wlan0  include  Kconfig  LICENSE  Makefile  os_dep  platform  README.md  runwpa  wlan0dhcp
root@raspberrypi:/home/pi/rtl8812AU# echo $DRV_VERSION
4.3.14
root@raspberrypi:/home/pi/rtl8812AU# git archive driver-${DRV_VERSION}
fatal: Not a valid object name
`

diederikdehaas, 2019-08-24:
From the README:
_Whereby it is assumed you're in the cloned project directory_
so you need to be in the `rtl8812AU` directory.