diederikdehaas/rtl8812AU #75: build fails with implicit declaration of function ‘allow_signal’

pnpavlov, 2017-07-01:
Hi,

dkms build -m ${DRV_NAME} -v ${DRV_VERSION} fails on Fedora 25 with kernel 4.11.7-200.fc25.x86_64 . Any hint how to get the driver installed ? 

Works:
```
DRV_NAME=rtl8812AU
DRV_VERSION=4.3.22-beta
git clone -b driver-4.3.22-beta https://github.com/diederikdehaas/rtl8812AU.git
cd rtl8812AU/
mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
```

Fails:
`dkms build -m ${DRV_NAME} -v ${DRV_VERSION}`

Error: 

```
# dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make'...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.11.7-200.fc25.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log for more information.

# cat /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log
DKMS make.log for rtl8812AU-4.3.22-beta for kernel 4.11.7-200.fc25.x86_64 (x86_64)
Sat  1 Jul 14:42:32 CEST 2017
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.7-200.fc25.x86_64/build M=/var/lib/dkms/rtl8812AU/4.3.22-beta/build  modules
make[1]: Entering directory '/usr/src/kernels/4.11.7-200.fc25.x86_64'
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/drv_types.h:32:0,
                 from /var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h: In function ‘thread_enter’:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h: In function ‘flush_signals_thread’:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o] Error 1
Makefile:1496: recipe for target '_module_/var/lib/dkms/rtl8812AU/4.3.22-beta/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8812AU/4.3.22-beta/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.11.7-200.fc25.x86_64'
Makefile:1705: recipe for target 'modules' failed
make: *** [modules] Error 2
```

I tried manually modifying to have https://github.com/Id2ndR/rtl8812AU/commit/0ddf23c1c81f53db13954585ebddbbe4c5a6e6d5
# cat /usr/src/rtl8812AU-4.3.22-beta/dkms.conf | grep MAKE
MAKE="'make' KVER=${kernelver}"
# cat /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log
```
DKMS make.log for rtl8812AU-4.3.22-beta for kernel 4.11.7-200.fc25.x86_64 (x86_64)
Sat  1 Jul 14:48:13 CEST 2017
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.7-200.fc25.x86_64/build M=/var/lib/dkms/rtl8812AU/4.3.22-beta/build  modules
make[1]: Entering directory '/usr/src/kernels/4.11.7-200.fc25.x86_64'
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/drv_types.h:32:0,
                 from /var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h: In function ‘thread_enter’:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h: In function ‘flush_signals_thread’:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o] Error 1
Makefile:1496: recipe for target '_module_/var/lib/dkms/rtl8812AU/4.3.22-beta/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8812AU/4.3.22-beta/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.11.7-200.fc25.x86_64'
Makefile:1705: recipe for target 'modules' failed
make: *** [modules] Error 2
```

pl-buiquang, 2017-07-29:
I had the same problem and found this link : https://forum.manjaro.org/t/error-with-rtl8812au/24066/5
Seems that adding this :
`#include <linux/version.h>
#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 11, 0)
#include <linux/sched/signal.h>
#endif
`
to the file osdep_service.h corrected the build issue. At least it successfully did on Debian 9 with kernel 4.11.0.

Hope it helps !

robinhood2014, 2017-08-13:
No good compiling on Linux Mint with kernel 4.12.6.

What works:
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.12.6/build M=/home/brandon/rtl8812AU  modules
make[1]: Entering directory '/home/brandon/Downloads/linux-4.12.6'
  CC [M]  /home/brandon/rtl8812AU/core/rtw_cmd.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_security.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_debug.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_io.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_ioctl_query.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_ioctl_set.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_ieee80211.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_mlme.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_mlme_ext.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_wlan_util.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_vht.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_pwrctrl.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_rf.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_recv.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_sta_mgt.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_ap.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_xmit.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_p2p.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_tdls.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_br_ext.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_iol.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_sreset.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_btcoex.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_beamforming.o
  CC [M]  /home/brandon/rtl8812AU/core/rtw_odm.o
  CC [M]  /home/brandon/rtl8812AU/core/efuse/rtw_efuse.o
  CC [M]  /home/brandon/rtl8812AU/os_dep/osdep_service.o
  CC [M]  /home/brandon/rtl8812AU/os_dep/linux/os_intfs.o
  CC [M]  /home/brandon/rtl8812AU/os_dep/linux/usb_intf.o
  CC [M]  /home/brandon/rtl8812AU/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/brandon/rtl8812AU/os_dep/linux/ioctl_linux.o
  CC [M]  /home/brandon/rtl8812AU/os_dep/linux/xmit_linux.o
  CC [M]  /home/brandon/rtl8812AU/os_dep/linux/mlme_linux.o
  CC [M]  /home/brandon/rtl8812AU/os_dep/linux/recv_linux.o
```

What fails:
```
  CC [M]  /home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.o
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:824:6: error: passing argument 2 of ‘cfg80211_roamed’ from incompatible pointer type [-Werror=incompatible-pointer-types]
    , notify_channel
      ^
In file included from /home/brandon/rtl8812AU/include/osdep_service_linux.h:81:0,
                 from /home/brandon/rtl8812AU/include/osdep_service.h:41,
                 from /home/brandon/rtl8812AU/include/drv_types.h:32,
                 from /home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5435:6: note: expected ‘struct cfg80211_roam_info *’ but argument is of type ‘struct ieee80211_channel *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:826:6: warning: passing argument 3 of ‘cfg80211_roamed’ makes integer from pointer without a cast [-Wint-conversion]
    , cur_network->network.MacAddress
      ^
In file included from /home/brandon/rtl8812AU/include/osdep_service_linux.h:81:0,
                 from /home/brandon/rtl8812AU/include/osdep_service.h:41,
                 from /home/brandon/rtl8812AU/include/drv_types.h:32,
                 from /home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5435:6: note: expected ‘gfp_t {aka unsigned int}’ but argument is of type ‘unsigned char *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:822:3: error: too many arguments to function ‘cfg80211_roamed’
   cfg80211_roamed(padapter->pnetdev
   ^
In file included from /home/brandon/rtl8812AU/include/osdep_service_linux.h:81:0,
[... 8 lines cut ...]
  mon_ndev->destructor = rtw_ndev_destructor;
          ^
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_preinit_wiphy’:
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6205:18: error: ‘WIPHY_FLAG_SUPPORTS_SCHED_SCAN’ undeclared (first use in this function)
  wiphy->flags |= WIPHY_FLAG_SUPPORTS_SCHED_SCAN;
                  ^
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6205:18: note: each undeclared identifier is reported only once for each function it appears in
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c: At top level:
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6242:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = cfg80211_rtw_change_iface,
                         ^
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6242:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6262:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^
/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.c:6262:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
scripts/Makefile.build:302: recipe for target '/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/home/brandon/rtl8812AU/os_dep/linux/ioctl_cfg80211.o] Error 1
Makefile:1515: recipe for target '_module_/home/brandon/rtl8812AU' failed
make[1]: *** [_module_/home/brandon/rtl8812AU] Error 2
make[1]: Leaving directory '/home/brandon/Downloads/linux-4.12.6'
Makefile:1576: recipe for target 'modules' failed
make: *** [modules] Error 2
```

martinlindhe, 2017-08-21:
The solution from @pl-buiquang worked for me too, kernel 4.11.0, but i added the snippet to include/osdep_service_linux.h

heyarne, 2017-09-08:
Adding the suggested include of `linux/sched/signal.h` to `include/osdep_service_linux.h` (thx @martinlindhe) helped with the implicit function declaration error; it did however lead to a new error:

```
root@bananapi:~/rtl8812AU# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.11.6-sunxi/build M=/root/rtl8812AU  modules
make[1]: Entering directory '/usr/src/linux-headers-4.11.6-sunxi'
  CC [M]  /root/rtl8812AU/core/rtw_debug.o
/bin/sh: 1: ./scripts/recordmcount: not found
scripts/Makefile.build:294: recipe for target '/root/rtl8812AU/core/rtw_debug.o' failed
make[2]: *** [/root/rtl8812AU/core/rtw_debug.o] Error 127
Makefile:1492: recipe for target '_module_/root/rtl8812AU' failed
make[1]: *** [_module_/root/rtl8812AU] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.11.6-sunxi'
Makefile:1576: recipe for target 'modules' failed
make: *** [modules] Error 2
```

I'm trying to compile this on a bananapi with ARMBIAN 5.31 stable Debian GNU/Linux 8 (jessie) 4.11.6-sunxi. Can somebody tell me what's wrong?

martinlindhe, 2017-09-08:
@heyarne did you follow the dkms build instructions in the readme ?

https://github.com/diederikdehaas/rtl8812AU#dkms

heyarne, 2017-09-08:
@martinlindhe Yes. I made the changes, commited them, re-ran the `git archive` command and set up dkms anew, which leads to exactly the above error, same as when running `make` manually.

martinlindhe, 2017-09-08:
@heyarne
Okay, because the instructions I linked dont mention you running "make" directly, as your paste suggest.

Would you try 
```
# DRV_NAME=rtl8812AU
# DRV_VERSION=4.3.14
# mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
# git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
# dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
# dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
# dkms install -m ${DRV_NAME} -v ${DRV_VERSION}
```

instead?

pvetter, 2017-12-16:
Compilation on Fedora 27 kernel 4.14.5-300.fc27.x86_64 of git cloned version of 20171215 gives the same 3 implicit declaration errors as original post:

osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’?
osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? 
osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’?

is it safe to suppress these warnings, or do they represent a bug?

when I tried the suggested patch "include version.h ... signal.h" compilation generated a LOT of errors.

[make-errors.txt](https://github.com/diederikdehaas/rtl8812AU/files/1564454/make-errors.txt)

ak47suk1, 2017-12-18:
Successfully compiled this driver on Linux Mint kernel 4.8

pvetter, 2017-12-18:
Compilation on Fedora 27 kernel 4.14.5-300.fc27.x86_64 of git cloned version of 20171215 gives the same 3 implicit declaration errors as original post even after I commented out the lines in Makefile:
EXTRA_CFLAGS += -Wall
EXTRA_CFLAGS += -Wextra

snwokenk, 2018-01-25:
has this been solved. I all of a sudden started getting this error after a 9 months of using this driver

daqqad, 2018-01-25:
Same issue as @snwokenk on 4.14.14 on OpenSUSE.
```
/var/lib/dkms/rtl8812AU/4.3.14/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/var/lib/dkms/rtl8812AU/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:824:6: error: passing argument 2 of ‘cfg80211_roamed’ from incompatible pointer type [-Werror=incompatible-pointer-types]
```

snwokenk, 2018-01-26:
@daqqad i just reverted back to kernel 4.10. And I'm able to connect. I'll wait till this is fixed.

daqqad, 2018-01-26:
I am able to build an older version of the driver (4.2.2) and connect to most networks, but I'm hoping newer driver will resolve WPA2 Enterprise PEAP connection issue I'm having. I'm getting an error from wpa_supplicant that claims driver does not support association.

I'm on a desktop with reliable wired connection so getting my wireless USB card to work is just something to play with @ lunch.

kaustubhcs, 2018-02-17:
Just waiting till this is fixed!

snwokenk, 2018-02-17:
Anyone having this problem  @VitalKoshalew beta-mod branch was able to compile and install and connect for me. on 4.13+ kernel

snwokenk, 2018-02-17:
Don't know why it hasn't been merged?

snwokenk, 2018-02-17:
https://github.com/VitalKoshalew/rtl8812AU/tree/driver-4.3.22-beta-mod

Tabestan, 2018-02-25:
I have the same error than OP.
`cat /var/lib/dkms/rtl8812AU/4.3.22-beta/build/make.log 
DKMS make.log for rtl8812AU-4.3.22-beta for kernel 4.13.0-36-generic (x86_64)
Sun Feb 25 12:51:49 PST 2018
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-36-generic/build M=/var/lib/dkms/rtl8812AU/4.3.22-beta/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-36-generic'
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/drv_types.h:32:0,
                 from /var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h: In function ‘thread_enter’:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h: In function ‘flush_signals_thread’:
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current)) 
      ^
/var/lib/dkms/rtl8812AU/4.3.22-beta/build/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl8812AU/4.3.22-beta/build/core/rtw_cmd.o] Error 1
Makefile:1550: recipe for target '_module_/var/lib/dkms/rtl8812AU/4.3.22-beta/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8812AU/4.3.22-beta/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-36-generic'
Makefile:1705: recipe for target 'modules' failed
make: *** [modules] Error 2
`

I get the same error after @pl-buiquang  suggestion.

Snerken, 2018-03-03:
Who would ever have thought installing a USB Wifi adapter would be so difficult. Been at this for days and no closer to figuring it out... Some consolation that I'm not the only one struggling with this.

```
snerx@snerx:~/rtl8812AU_8821AU_linux$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-32-generic/build M=/home/snerx/rtl8812AU_8821AU_linux  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-32-generic'
  CC [M]  /home/snerx/rtl8812AU_8821AU_linux/core/rtw_cmd.o
In file included from /home/snerx/rtl8812AU_8821AU_linux/include/drv_types.h:32:0,
                 from /home/snerx/rtl8812AU_8821AU_linux/core/rtw_cmd.c:22:
/home/snerx/rtl8812AU_8821AU_linux/include/osdep_service.h: In function ‘thread_enter’:
/home/snerx/rtl8812AU_8821AU_linux/include/osdep_service.h:343:2: **error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^**
/home/snerx/rtl8812AU_8821AU_linux/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/snerx/rtl8812AU_8821AU_linux/include/osdep_service.h:353:6: **error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current)) {
      ^**
/home/snerx/rtl8812AU_8821AU_linux/include/osdep_service.h:354:3: **error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^**
cc1: all warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/home/snerx/rtl8812AU_8821AU_linux/core/rtw_cmd.o' failed
make[2]: *** [/home/snerx/rtl8812AU_8821AU_linux/core/rtw_cmd.o] Error 1
Makefile:1550: recipe for target '_module_/home/snerx/rtl8812AU_8821AU_linux' failed
make[1]: *** [_module_/home/snerx/rtl8812AU_8821AU_linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-32-generic'
Makefile:1570: recipe for target 'modules' failed
make: *** [modules] Error 2
snerx@snerx:~/rtl8812AU_8821AU_linux$ 
```

bmenyuk, 2018-03-03:
Same issue as @snwokenk / @daqqad. Reverted to driver-4.2.2 which compiles and works.

diederikdehaas, 2018-03-12:
Can you retest this with the latest from the 4.3.20 branch?

ronald-jaepel, 2018-03-23:
On a fresh Ubuntu 16.04 install the current 4.3.20 branch compiles without errors.

diederikdehaas, 2018-03-24:
> commented out the lines in Makefile:
EXTRA_CFLAGS += -Wall
EXTRA_CFLAGS += -Wextra

That hides the compilation issue, but does not fix it.

Anyway, the issue was caused by a kernel (header) change in the 4.12.x+ kernels and should be fixed in commit https://github.com/diederikdehaas/rtl8812AU/commit/523c38e09c9590029bf3066bb23970c9ad03d2d7 for the `driver-4.22-beta` branch and commit https://github.com/diederikdehaas/rtl8812AU/commit/c377d1902d5c6a5c763c4dac59de2d91301e97ea for the `driver-4.3.20` branch.

If you still have issues, please report the exact kernel version (`uname -a`) and driver version/branch you tried it with (and OS)

efikatec, 2018-03-25:
I followed the instructions in https://github.com/tbsdtv/linux_media/wiki and it seems to compile for Linux server 4.13.0-37-generic
But for now I cannot test it as the card is for now 'locked' in another machine. This I will do this week.

shlomiuziel, 2018-03-26:
@diederikdehaas Thanks for the fix, can this be patched in the driver-4.3.14 branch also? According to this issue https://github.com/diederikdehaas/rtl8812AU/issues/55 it's the only version working with 5 Ghz networks.

jeremysouthard, 2018-04-29:
When I run lspci I see the USB 3.0 hub.
When I run lsusb I see the network adapter.
When I run ifconfig the network adapter is not listed.

My google searching led me here to install the 8812ua driver, but I am getting the same errors when attempting to `make`...

`$ uname -a
Linux machine1 4.13.0-32-generic #35~16.04.1-Ubuntu SMP Thu Jan 25 10:13:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

`systemuser@machine1:~/rtl8812AU_8821AU_linux$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-32-generic/build M=/home/systemuser/rtl8812AU_8821AU_linux  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-32-generic'
  CC [M]  /home/systemuser/rtl8812AU_8821AU_linux/core/rtw_cmd.o
In file included from /home/systemuser/rtl8812AU_8821AU_linux/include/drv_types.h:32:0,
                 from /home/systemuser/rtl8812AU_8821AU_linux/core/rtw_cmd.c:22:
/home/systemuser/rtl8812AU_8821AU_linux/include/osdep_service.h: In function ‘thread_enter’:
/home/systemuser/rtl8812AU_8821AU_linux/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^
/home/systemuser/rtl8812AU_8821AU_linux/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/systemuser/rtl8812AU_8821AU_linux/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current)) {
      ^
/home/systemuser/rtl8812AU_8821AU_linux/include/osdep_service.h:354:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^
cc1: all warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/home/systemuser/rtl8812AU_8821AU_linux/core/rtw_cmd.o' failed
make[2]: *** [/home/systemuser/rtl8812AU_8821AU_linux/core/rtw_cmd.o] Error 1
Makefile:1550: recipe for target '_module_/home/systemuser/rtl8812AU_8821AU_linux' failed
make[1]: *** [_module_/home/systemuser/rtl8812AU_8821AU_linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-32-generic'
Makefile:1570: recipe for target 'modules' failed
make: *** [modules] Error 2
`

KES777, 2018-06-06:
I clone repository and then follow [instruction](https://github.com/diederikdehaas/rtl8812AU#dkms)

```
work rtl8812AU # DRV_NAME=rtl8812AU
work rtl8812AU # DRV_VERSION=4.3.20
work rtl8812AU # mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
work rtl8812AU # git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
work rtl8812AU # dkms add -m ${DRV_NAME} -v ${DRV_VERSION}

Creating symlink /var/lib/dkms/rtl8812AU/4.3.20/source ->
                 /usr/src/rtl8812AU-4.3.20

DKMS: add completed.
work rtl8812AU # dkms build -m ${DRV_NAME} -v ${DRV_VERSION}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' KVER=4.13.0-26-generic.....................
cleaning build area....

DKMS: build completed.
work rtl8812AU # dkms install -m ${DRV_NAME} -v ${DRV_VERSION}

8812au:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.13.0-26-generic/updates/dkms/

depmod.....

DKMS: install completed.

```

All steps succeed

mridah, 2018-10-05:
> I had the same problem and found this link : https://forum.manjaro.org/t/error-with-rtl8812au/24066/5
> Seems that adding this :
> 
> ```c++
> #include <linux/version.h>
> #if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 11, 0)
> #include <linux/sched/signal.h>
> #endif
> ```
> 
> to the file osdep_service.h corrected the build issue. At least it successfully did on Debian 9 with kernel 4.11.0.
> 
> Hope it helps !

where do I find this file ?? I'm using Ubuntu

diederikdehaas, 2019-08-23:
I'm assuming this issue is fixed. If not, feel free to respond. Include output of `uname -a` and the driver version used.

closed 2019-08-23 by diederikdehaas