diederikdehaas/rtl8812AU #76: confusing install instructions

kevinkey619, 2017-07-17:
Why am I getting the below errors? The install instructions are very confusing.

root@raspberrypi3:/usr/src/rtl8812AU-4.3.14# dkms add -m ${DRV_NAME} -v ${DRV_VERSION}

Creating symlink /var/lib/dkms/rtl8812AU/4.3.14/source ->
                 /usr/src/rtl8812AU-4.3.14

DKMS: add completed.
root@raspberrypi3:/usr/src/rtl8812AU-4.3.14# dkms build -m ${DRV_NAME} -v ${DRV_VERSION}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make'....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.9.35-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.14/build/make.log for more information.
root@raspberrypi3:/usr/src/rtl8812AU-4.3.14# dkms install -m ${DRV_NAME} -v ${DRV_VERSION}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make'....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.9.35-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.14/build/make.log for more information.
root@raspberrypi3:/usr/src/rtl8812AU-4.3.14#

PolarJunction, 2017-12-17:
This is a build error with your current kernel version, you may find this thread of use: https://github.com/diederikdehaas/rtl8812AU/issues/75

diederikdehaas, 2018-03-12:
Correct. The latest from the 4.3.20 branch should fix that