diederikdehaas/rtl8812AU #78: odroid xu4: modinfo: missing alias (0bda:0811)

ekeyser, 2017-08-10:
Previously I have been able to compile, install and bring up 8812AU adapters on raspberry pi devices. Recently I purchased an Odroid XU4 and tried to perform the same procedure. Compile  works just fine but I've noticed that modinfo lists a smaller set of aliases which presumably doesn't trigger activation of the adapter.

Specifically, I was expecting modinfo to indicate 0bda:0811 as an alias. That's what shows up when I run modinfo on my raspberry pi. lsusb is Bus 003 Device 005 for this device (below).

Now, I realize maybe that this isn't the right place for posting but maybe it is. I'm not really a C coder but I plan on digging into the code a little today (I see some interesting stuff in os_dep/linux/usb_intf.c) but thought I'd also post here in the meantime. I've got some output below as reference. Wondering if you happen to know why an alias would not show up or under what circumstances a compiler would decide not to include an alias. Like I said I plan on looking at the code and digging into device driver stuff to try to figure this out on my own - and if I do I'll paste my findings :-)

Thanks for the work you do.

ref:

insertion of device on USB:
```
Aug 10 14:37:59 odroid kernel: [ 1005.801640] usb 3-1.2: new high-speed USB device number 5 using xhci-hcd
Aug 10 14:37:59 odroid rsyslogd-2007: action 'action 9' suspended, next retry is Thu Aug 10 14:38:29 2017 [v8.16.0 try http://www.rsyslog.com/e/2007 ]
Aug 10 14:37:59 odroid kernel: [ 1005.902375] usb 3-1.2: New USB device found, idVendor=0bda, idProduct=0811
Aug 10 14:37:59 odroid kernel: [ 1005.902382] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 10 14:37:59 odroid kernel: [ 1005.902388] usb 3-1.2: Product: 802.11ac WLAN Adapter 
Aug 10 14:37:59 odroid kernel: [ 1005.902392] usb 3-1.2: Manufacturer: Realtek 
Aug 10 14:37:59 odroid kernel: [ 1005.902397] usb 3-1.2: SerialNumber: 00e04c000001
```

```
root@odroid:~/_rtl# lsusb
Bus 006 Device 002: ID 0bda:8153 Realtek Semiconductor Corp. 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0bda:0811 Realtek Semiconductor Corp.
Bus 003 Device 004: ID 413c:2006 Dell Computer Corp. 
Bus 003 Device 003: ID 413c:1004 Dell Computer Corp. 
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
root@odroid:~/_rtl# modinfo 8812au
filename:       /lib/modules/4.9.28-38/kernel/drivers/net/wireless/rtl8812au/8812au.ko
version:        v5.1.5_19247.20160830
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     BCDFD29A8A9C3113C5A0B53
alias:          usb:v2001p3316d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3315d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB30d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1740p0100d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1058p0632d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3313d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0586p3426d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0022d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p17D2d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0409p0408d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0789p016Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0952d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0074d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v050Dp1106d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8812d*dc*dsc*dp*ic*isc*ip*in*
depends:        cfg80211
intree:         Y
vermagic:       4.9.28-38 SMP preempt mod_unload ARMv7 p2v8 
parm:           rtw_ips_mode:The default IPS mode (int)
[... 60 lines cut ...]
parm:           rtw_powertracking_type:default init value:64 (uint)
parm:           rtw_GLNA_type:default init value:0 (uint)
parm:           rtw_TxBBSwing_2G:default init value:0xFF (uint)
parm:           rtw_TxBBSwing_5G:default init value:0xFF (uint)
parm:           rtw_OffEfuseMask:default open Efuse Mask value:0 (uint)
parm:           rtw_FileMaskEfuse:default drv Mask Efuse value:0 (uint)
parm:           rtw_rxgain_offset_2g:default RF Gain 2G Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gl:default RF Gain 5GL Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_lmt_enable:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_a:5G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_b:5G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_c:5G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_d:5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_phy_file_path:The path of phy parameter (charp)
parm:           rtw_load_phy_file:PHY File Bit Map (int)
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)
```

diederikdehaas, 2018-03-12:
According to https://wikidevi.com/wiki/0bda, the `0bda:0811` is a Realtek RTL8811AU device.
Note that it says 881**1**, not 881**2**.
When looking in `os_dep/linux/usb_intf.c` I See the `0x0811` device identifier in the `CONFIG_RTL8821A` list, not `CONFIG_RTL8812A`.
And that would indeed explain the shorter alias list.

(I'm not a C coder either ;-))

dachshund-digital, 2018-04-11:
Is this not the alias expected? 

# modinfo 8812au | grep alias | grep 0BDA | grep A811
alias:          usb:v0BDApA811d * dc * dsc * dp * ic * isc * ip * in *

Sorry for the multiple greps... was just walking the list.  Did this against driver module I found from a different source... 'install-wifi' solution, from FARS Robotics, i.e. http://www.fars-robotics.net/install-wifi

diederikdehaas, 2018-04-11:
that looks to be `0xA811` not `0x0811`.
There are 2 other possible issues at play here:
1) different driver versions support different (single/multiple) chipsets
2) it looks like I expanded the driver ID list for one version, but not all of them