diederikdehaas/rtl8812AU #83: kernel 4.15rc1 build fails..

oscarbg, 2017-12-04:
has compilation issues

DaveC49, 2018-01-16:
also kernel 4.13.0-26-generic  generating -Weeror=implicit-function-declaration error on the following calls:
allow_signal(SIGTERM)
if (signal_pending (current)) 
flush_signals(current)

DaveC49, 2018-01-16:
I've tried several other repositories with rt8812au drivers with the same result whether compiling and installing directly or using dkms

Jarrux, 2018-01-19:
I have the same problem due to a kernel update from 4.10.0-28-generic to 4.13.-26-generic

Unable to compile under the new kernel. I came back to older version, and everything was fine.

snwokenk, 2018-01-25:
@Jarrux  I reverted and it works. Didn't even need to recompile. I guess until they make something stable then i'll stick to 4.10

DaveC49, 2018-01-25:
You may find it safer to revert to 4.4 rather than 4.10. Ubuntu and others have announced that 4.10 will not be patched for the meltdown and Sceptre vulnerabilities whereas 4.4 and 4.13 have been already.

fbelzile, 2018-03-06:
This commit seems to be fixing issues with the new timer interface found in 4.15 (worked for me):
https://github.com/aircrack-ng/rtl8812au/commit/f221a169f281dab9756a176ec2abd91e0eba7d19

diederikdehaas, 2018-03-12:
Support for 4.15 kernel should now work (or at least compile) with commits I added to the 4.3.20 branch.
Can you test it?

fbelzile, 2018-03-23:
Just tested it on 4.15.10-300.fc27.x86_64. It builds with no errors but after trying to install it with "modprobe -a 8812au" it didn't detect my wireless adapter (D-Link DWA-171 A1). Reverting to this driver worked: https://github.com/aircrack-ng/rtl8812au

diederikdehaas, 2018-03-23:
What's the USB ID (`lsusb`)

diederikdehaas, 2018-03-23:
According to https://wikidevi.com/wiki/D-Link_DWA-171_rev_A1 it's `2001:3314` and it's already part of ` os_dep/linux/usb_intf.c`, but it's in the `CONFIG_RTL8821A` section.
In the aircrack-ng repo, they have `CONFIG_RTL8821A = y`, whereas it's `CONFIG_RTL8821A = n` in my repo.
I'm 99% certain that's the cause of it.

diederikdehaas, 2018-03-23:
Changing the `n` to a `y` will result in a compile error though as my `hal/phydm/rtl8821a` folder is present, but empty.

TBF: you're probably better off using their driver/repo.
They actually seem to know what they're doing ;-)