diederikdehaas/rtl8812AU #85: Error while git archive

etudor, 2018-01-22:
Hello, 

I get the following errors while runnig this command ```sudo git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION} ```: 
```
sudo git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION} 
tar: .gitignore: Cannot open: Permission denied
tar: Kconfig: Cannot open: Permission denied
tar: Makefile: Cannot open: Permission denied
tar: README.md: Cannot open: Permission denied
tar: clean: Cannot open: Permission denied
tar: core: Cannot mkdir: Permission denied
tar: core: Cannot mkdir: Permission denied
tar: core/efuse: Cannot mkdir: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/efuse/rtw_efuse.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_ap.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_beamforming.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_br_ext.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_bt_mp.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_btcoex.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_cmd.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_debug.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_eeprom.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_ieee80211.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_io.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_ioctl_query.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_ioctl_rtl.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_ioctl_set.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_iol.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_mem.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_mlme.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_mlme_ext.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_mp.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_mp_ioctl.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_odm.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_p2p.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_pwrctrl.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_recv.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_rf.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_security.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_sreset.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
tar: core/rtw_sta_mgt.c: Cannot open: No such file or directory
tar: core: Cannot mkdir: Permission denied
[... 748 lines cut ...]
tar: os_dep: Cannot mkdir: Permission denied
tar: os_dep/osdep_service.c: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_ARM_SUNnI_sdio.c: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_ARM_SUNxI_sdio.c: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_ARM_SUNxI_usb.c: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_ARM_WMT_sdio.c: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_RTK_DMP_usb.c: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_arm_act_sdio.c: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_ops.c: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_ops.h: Cannot open: No such file or directory
tar: platform: Cannot mkdir: Permission denied
tar: platform/platform_sprd_sdio.c: Cannot open: No such file or directory
tar: runwpa: Cannot open: Permission denied
tar: wlan0dhcp: Cannot open: Permission denied
tar: Exiting with failure status due to previous errors
```

darkdrgn2k, 2018-01-23:
| tar is no running as sudo

try 

sudo git archive driver-${DRV_VERSION} | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}

diederikdehaas, 2018-03-12:
You can leave out the first sudo, so `git archive` is run as normal user, but your conclusion is absolutely right. You *must* run `tar -x ...` as sudo as you're trying to write into `/usr/src/`.