diederikdehaas/rtl8812AU #89: Exec format error

HornedReaper, 2018-02-22:
Each time my system update the kernel, I must re-install the module because when I type "sudo modprobe 8812au" I got this line as all result :
"modprobe: ERROR: could not insert '8812au': Exec format error"
Watch if there is no bug nor backdoor that could disable nor destroy or make a script to re-install with the kernel update the module because it's very boring to have to use my phone to get network to get the module that give me the network at each update/upgrade of my system... :-/

dizzi90, 2018-02-24:
Same error. Won't compile either.
`magnus@pop-os:~/Source/rtl8812AU$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-36-generic/build M=/home/magnus/Source/rtl8812AU  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-36-generic'
  CC [M]  /home/magnus/Source/rtl8812AU/core/rtw_cmd.o
In file included from /home/magnus/Source/rtl8812AU/include/drv_types.h:32:0,
                 from /home/magnus/Source/rtl8812AU/core/rtw_cmd.c:22:
/home/magnus/Source/rtl8812AU/include/osdep_service.h: In function ‘thread_enter’:
/home/magnus/Source/rtl8812AU/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/home/magnus/Source/rtl8812AU/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/magnus/Source/rtl8812AU/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
      timer_pending
/home/magnus/Source/rtl8812AU/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/home/magnus/Source/rtl8812AU/core/rtw_cmd.o' failed
make[2]: *** [/home/magnus/Source/rtl8812AU/core/rtw_cmd.o] Error 1
Makefile:1550: recipe for target '_module_/home/magnus/Source/rtl8812AU' failed
make[1]: *** [_module_/home/magnus/Source/rtl8812AU] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-36-generic'
Makefile:1576: recipe for target 'modules' failed
make: *** [modules] Error 2
`

diederikdehaas, 2018-03-12:
I've added some commits to the 4.3.20 branch, which among other things adds support for newer kernels.
Can you test that and see whether that fixes your issue?

daapp, 2019-09-11:
Same error in Devuan/buster on amd64:
 % modprobe -f -v 8812au
insmod /lib/modules/4.19.0-5-amd64/updates/dkms/8812au.ko 
modprobe: ERROR: could not insert '8812au': Operation not permitted
Was installed using dkms.