diederikdehaas/rtl8812AU #90: Build with DKMS fail Kernel 4.13.0-36-generic (x86-64)

DaveC49, 2018-02-26:
make.log
DKMS make.log for rtl8812AU-4.3.14 for kernel 4.13.0-36-generic (x86_64)
Monday 26 February  11:56:21 AEST 2018
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-36-generic/build M=/var/lib/dkms/rtl8812AU/4.3.14/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-36-generic'
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.14/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8812AU/4.3.14/build/include/drv_types.h:32:0,
                 from /var/lib/dkms/rtl8812AU/4.3.14/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h: In function ‘thread_enter’:
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h: In function ‘flush_signals_thread’:
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current)) 
      ^
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/var/lib/dkms/rtl8812AU/4.3.14/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl8812AU/4.3.14/build/core/rtw_cmd.o] Error 1
Makefile:1550: recipe for target '_module_/var/lib/dkms/rtl8812AU/4.3.14/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8812AU/4.3.14/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-36-generic'
Makefile:1576: recipe for target 'modules' failed
make: *** [modules] Error 2

DaveC49, 2018-03-05:
Found a solution which worked for me. It is documented here
https://forums.linuxmint.com/viewtopic.php?f=53&t=264675&p=1436132#p1436132

diederikdehaas, 2018-03-12:
Both repos are good one :-)

I have just pushed several commits to the 4.3.20 branch. Can you test whether that fixes your issues?

EvelienSnel, 2018-04-02:
I got the same errors as DaveC49 when building with DKMS from the official Ubuntu repository.
Today I took a clone from here, built the 4.3.20 branch, and did not have a problem.
`evelien@BombadilU16:~/Projects/rtl8812AU$ sudo dkms build -m ${DRV_NAME} -v ${DRV_VERSION}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' KVER=4.13.0-37-generic............................
cleaning build area....

DKMS: build completed.
evelien@BombadilU16:~/Projects/rtl8812AU$ 
`
Hope this helps. :)

diederikdehaas, 2018-04-02:
It does, thanks :)

closed 2018-04-11 by DaveC49