diederikdehaas/rtl8812AU #91: Tp-link ac1300 Archer T4U

mavridis92, 2018-03-08:
I have already installed the driver 4.3.22-beta and it's run perfect with D-link dwa-171 and Edimax Ew-7811utc, but unfortunately, it just isn't working  for my new  adapter Tplink Archer ac1300.

diederikdehaas, 2018-03-12:
What is the Bus ID (`lsusb`)?

mavridis92, 2018-04-05:
I found a solution [here] (https://unix.stackexchange.com/questions/345375/issue-while-installing-driver-for-archer-t4u-ac1300-on-mint-17) that works for me. 
Update /rtl8812au/os_dep/linux/usb_intf.c-> find the original bus id for the archer T4U AC1200 (2357:0101) and add a line with the bus id for the AC1300 (2357:010d) right after:

{USB_DEVICE(0x2357, 0x0101),.driver_info = RTL8812},    
**{USB_DEVICE(0x2357, 0x010d),.driver_info = RTL8812},**

PR #105 (2019-08-11): Add TP-LINK T4U AC1300 vendor/device ids.
resolves #91
closed 2019-08-11 by diederikdehaas