diederikdehaas/rtl8812AU #98: 5.2.20 Driver

cariad61, 2018-04-17:
Your notes suggest you may be interested to hear of new drivers, I understand there is version 5.2.20 - which I found at : https://github.com/gordboy/rtl8812au

My interest is to see if any more power can be unlocked from the AWUS036AH Alfa unit as I am using an external antenna on a 2m mast with the associated attenuation in the coax between the adapter and antenna. 

I have complied drivers that suggest they allow the Tx setting to be unlocked and whilst they report an increase that appears to make no difference of the actual output  (checked via setting the unit up as a hotspot).  Moreover I am having problems with these drivers working with wpa_supplicant,  I keep getting disconnected as the 4 way handshake fails when using in managed mode.

Any thoughts welcomed.

heroxbd, 2018-05-01:
I was about to report the same.  So +1 for this.

doa379, 2018-06-02:
This driver appears to be good. With the driver in the repo abperiasamy/rtl8812AU_8821AU_linux I could only get 150Mbps (reported 300Mbps in iw list). Now I get 952Mbps reported in iw list.

diederikdehaas, 2019-08-22:
Added this version to the repo.

closed 2019-08-22 by diederikdehaas