diederikdehaas/rtl8812AU #100: Setup on Ubuntu 18.04 4.15.0-20-generic

siddharth99, 2018-05-13:
I am having a little difficulty getting a USB WiFi setup; I am able to compile the module and insert into the kernel but I don't see the lights lit up on the USB dongle and neither do I see anything in ifconfig. I'm not sure what I am doing wrong; any help would be greatly appreciated.

doa379, 2018-06-02:
Yes I too have this problem. The network interface wlan0 isn't showing up for some reason.

doa379, 2018-06-02:
I tried another driver and this works:
https://github.com/gordboy/rtl8812au

ljschumacher, 2018-08-01:
I'm having trouble with the installation under that link. Specifically, the instructions for the dkms installation say " (as root, or sudo) copy source folder contents to /usr/src/rtl8812au-5.2.20"
but where is the source folder?

ItsTehStory, 2018-08-04:
@ljschumacher the source folder, unless I'm wrong, is the actual git repository. I copied the whole thing, it compiled, but my adapter still won't show up tho.