diederikdehaas/rtl8812AU #102: TP Link Archer T4U 4.16.12-300

joeminicucci, 2018-05-30:
I just built and installed successfully on Fedora 28, kernel 4.16.12-300 with no problems, except the interface is not coming up and appear in `ifconfig` like it usually does after the `dkms` build/install. Any ideas? Thanks!

rburcham, 2018-06-01:
This version of the driver hasn't been updated for this specific device yet....  apply the following patch and it'll work.

--- os_dep/linux/usb_intf.c     2018-06-01 09:06:40.794566280 -0500
+++ os_dep/linux/usb_intf.c_patch       2018-06-01 09:07:31.945858230 -0500
@@ -168,6 +168,7 @@
        {USB_DEVICE(0x2019, 0xAB30),.driver_info = RTL8812}, /* Planex - Abocom */
        {USB_DEVICE(0x20F4, 0x805B),.driver_info = RTL8812}, /* TRENDnet - */
        {USB_DEVICE(0x2357, 0x0101),.driver_info = RTL8812}, /* TP-Link - Archer T4U */
+       {USB_DEVICE(0x2357, 0x010D),.driver_info = RTL8812}, /* TP-Link - Archer T4U */
        {USB_DEVICE(0x2357, 0x0103),.driver_info = RTL8812}, /* TP-Link - T4UH */
        {USB_DEVICE(0x7392, 0xA822),.driver_info = RTL8812}, /* Edimax - EW-7822UAC */

Question for you:  Have you had difficulty with drops?  and kernel WARNING in dmesg on cfg80211  and/or 8812au?

joeminicucci, 2018-06-23:
Sorry about the delayed response. I couldnt get the patch working, perhaps because I edited os_dep/linux/usb_intf.c instead of deleting it and creating os_dep/linux/usb_intf.c_patch in place of it. Even so I was unable to build the DKMS mod, i was running into the exact same problems as the fellow over at [ubuntu forms](urlhttps://forum.ubuntu-it.org/viewtopic.php?p=5045535). My solution was to use the repository at [github] (https://github.com/abperiasamy/rtl8812AU_8821AU_linux) which is based on your 4.3.14 port. To answer your question, I havent been experiencing any drops and dmesg appears otherwise business as usual