diederikdehaas/rtl8812AU #103: `dkms build` looks for 'armv7l' folder instead of 'arm' folder

necromuralist, 2018-06-03:
I don't know if this is a dkms problem or the configuration problem, but when I run:

```
sudo dkms install -m $DRV_NAME -v $DRV_VERSION
```

I get the error:

```
'make' KVER=4.4.38-v7+....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812AU: 4.3.20 not found
Error! Bad return status for module build on kernel: 4.4.38-v7+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.20/build/make.log for more information.
```

Unfortunately I've successfully built it since then so I don't have the errors from the 'make.log'. But the solution for me was what I saw in [this Stack Overflow post](https://stackoverflow.com/questions/39775844/how-to-make-dkms-pass-the-right-arch-to-make), which was to make a symlink to the arm folder:

```
sudo ln -s /usr/src/linux-headers-4.4.38-v7+/arch/arm/ /usr/src/linux-headers-4.4.38-v7+/arch/armv7l
```

So, once you know about it the fix is relatively easy.

juliogonzalez, 2018-07-10:
make.log with the latest Raspbian 9.4 kernel:
```
DKMS make.log for rtl8812AU-4.3.20 for kernel 4.14.52-v7+ (armv7l)
Tue Jul 10 03:28:28 CEST 2018
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.14.52-v7+/build M=/var/lib/dkms/rtl8812AU/4.3.20/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.52-v7+'
Makefile:638: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.14.52-v7+'
Makefile:1687: recipe for target 'modules' failed
make: *** [modules] Error 2
```

juliogonzalez, 2018-07-10:
And to bypass the error without creating the link (it's dirty, since you would need to do the same for subsequent kernel updates):
```
ARCH=arm dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
dkms install -m ${DRV_NAME} -v ${DRV_VERSION}
```
It seems to me that the problem could be at https://github.com/diederikdehaas/rtl8812AU/blob/driver-4.3.20/Makefile#L910

I wonder if Rasbian changed the way they compile the kernel, as this was working for kernel ```4.9.59-v7+```

juliogonzalez, 2018-07-10:
In fact there is no need to use ```ARCH=arm```, as I just noticed at 4.3.20 README the following: https://github.com/diederikdehaas/rtl8812AU/blob/driver-4.3.20/README.md#raspberry-pi

This is also required for 4.3.14 (I just checked) and most probably f or 4.3.8 as well, but the README for those versions does not have it.

JackGrinningCat, 2018-12-13:
Hi @juliogonzalez,

I actually did change the makefile but anyway got the error. 
I tried also to change the makefile, that it is setting the ARCH := arm . Did not helped for the makefile in the linux-headers folders.
I tried to call dkms with the -a arm option but that didn't helped either.

So in my case the best option for Beagle Bone Black with the newest DEBIAN stable image is making the link.
`sudo ARCH=arm dkms build -m ${DRV_NAME} -v ${DRV_VERSION}`

juliogonzalez, 2018-12-18:
@JackGrinningCat, what driver version are you trying to compile?

You don't need to change `ARCH` and neither need to change nothing at the linux headers.

If you use >= 4.3.14, you need to following the instructions at https://github.com/diederikdehaas/rtl8812AU/blob/driver-4.3.20/README.md#raspberry-pi

The only Makefile you need to change is the one from rtl8812AU

CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_ARM_RPI = n
to

CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = y

That is, disabling building for i386, and enabling building for ARM on a Raspberry Pi.

If that still fails for you, please share the exact steps you are following, and paste the exact error (maybe I will be able to submit a fix, even if I don't own a BeagleBone :-)

PR #106 (2019-08-23): Add Raspberry Pi instructions for 4.3.14
This is required for 4.3.14 as well (I tested myself).

Most probably this commit should be cherrypicked to 4.3.8 as well.

It should fix #103,