diederikdehaas/rtl8812AU #118: Cannot compile on the BeagleBoard X15

bitreiver, 2019-03-12:
I'm installing an Alfa AC usb wifi adapter on a BeagleBoard X15. The Alfa website specifies that the device uses the RTL8812AU drivers.

After pulling the repo, updating the Makefile to use the 'Raspberry Pi' settings (ie. CONFIG_PLATFORM_ARM_RPI = y), and running through the commands provided in the readme, I got the following error on the '`dkms build...`' step:

Error! Your kernel headers for kernel 4.14.49-ti-r54 cannot be found.

...the fix for that was to install the BeagleBoard kernel headers with the following command:
`sudo apt-get install linux-headers-$(uname -r)`

Once the headers were installed (and verified with: "`dpkg -L linux-headers-$(uname -r)`"), I ran the build command again...
`dkms build -m ${DRV_NAME} -v ${DRV_VERSION}`
...and got the following error:
```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.14.49-ti-r54...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.14.49-ti-r54 (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.20/build/make.log for more information.
```

Turns out, on the BeagleBoard, the linux headers create a 
`/usr/src/linux-headers.../arch/arm` folder, but the drivers here expect a `/usr/src/.../arch/armv7l` folder.

If this happens, you can get around the issue by simply creating a symlink:
`ln -s /usr/src/linux-headers-4.14.49-ti-r54/arch/arm /usr/src/linux-headers-4.14.49-ti-r54/arch/armv7l` 

Re-run the build step and it should work fine! Once the install command is run, log out and log back in and run `iwconfig` to see the wifi device show up as an interface.

bitreiver, 2019-03-12:
Would it be possible to implement a fix so that the BeagleBoard is a supported platform?

diederikdehaas, 2019-08-23:
> the drivers here expect a /usr/src/.../arch/armv7l folder

I've recently pushed some updates, could you try again if the problem still exists?
When I do `make` on a RPi3B with kernel `Linux rasppi-3b 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux`, it works.

bitreiver, 2020-03-12:
Thanks! Works great.

closed 2020-03-12 by bitreiver