diederikdehaas/rtl8812AU #119: Wi-Fi Direct not advertising

visbs, 2019-04-05:
Hi,

I am using RTL8812AU based chipset USB Wi-Fi Adapter (Like ASUS AC56 , Linksys WUSB6300) in ubuntu (tried in different OS and Kernel version).
And I am trying to make use of WI-Fi Direct using wpa_supplicant (P2P Enabled) with user app wpa_cli.
In case of Intel Wireless Driver, whenever I do P2P_FIND or any scan command, from wpa_cli, my device is visible in any Wi-Fi Direct supported devices like Android, Windows etc.
But in case of Realtek driver (rtl8812au), my device is not visible or listed in Wi-Fi Direct device list, whereas I can get the nearby peers available in my device. As well I am able to connect to any SSID available.
So, am I missing something additional setup required for driver to make my device visible to other peer?
Any help would be very helpful?

Thank you,

diederikdehaas, 2019-08-23:
See https://github.com/aircrack-ng/rtl8812au/issues/314 and https://github.com/aircrack-ng/rtl8812au/commit/619d992fe43f637715ad6a8e2230d21de02b50ac (in branch v5.6.4.1)

The code in that repo is substantially modified from the original code by Realtek, so even if I wanted to, I'd probably fail in copying that code over and making it work.

kimocoder, 2019-08-23:
For Wi-Fi direct, private ioctl code needs to be added.

diederikdehaas, 2019-08-23:
Thanks :)

The problem for me would be that I'd copying code which I don't understand (at all).
I recently added a note (https://github.com/diederikdehaas/rtl8812AU/commit/c49c75e1b0bd7142e8d22822aa5c069158d00690) that I'm actually clueless wrt C and thus this driver. I've recently made updates to most branches which makes it compile until kernel 5.2.
That's pretty much the extend to what I'm capable of.

But then it still didn't **work**. Spend days researching `Association request to the driver failed`, but I'm also pretty much clueless when it comes to wireless in general when things don't work right away.
So I tried your/aircrack-ng driver and that did work :+1:  It did lose at some point the connection to my 5G network, but seems rock-solid when connected to my 2.4G network.

I started this repo as I found a newer version of the driver that no one else had (or at least had shared). That is no longer the case. Now there are several (too many imo) repos with regularly updated driver versions for my (one) wireless device, maintained by people who actually know what they're doing.

So now I'm mostly contemplating 'shutting down' this repo and refer people to your/aircrack-ng repo. It would be great if all the people interested in this driver would work together on one repo instead of lots of people maintaining (i.e. copying each other's patches) their own copy/fork of a 8812AU repo.

kimocoder, 2019-08-23:
@diederikdehaas 

ftp3.realtek.com
user: RTL8812A
password: J2e8Fg

There you have the officials

diederikdehaas, 2019-08-23:
Yeah, I had already found those and that's where I got the latest versions from and is what I used to publish newer driver versions. As do you and some other people.
Sure, it's cool that apparently so many people like/use this repo, but the real added value it once had, is no longer true. With previous updates I could use my wireless device again, but even that is no longer true. With good/great alternatives, what are the reasons to keep working on this repo.
Vanity? Doesn't sound like a good reason to me :thinking:

diederikdehaas, 2019-08-24:
> `Association request to the driver failed`

Looks a lot like https://github.com/aircrack-ng/rtl8812au/issues/265. Also with my 5.2.20 and iirc also with 5.6.4.

kimocoder, 2019-08-24:
Only working om 5.6.4.1. Branches below gets... Retired