diederikdehaas/rtl8812AU #120: build issues raspbian buster pi4 b

supaplextor, 2019-07-10:
```
pi@raspberrypi:~/rtl8812AU $ vim Makefile 
pi@raspberrypi:~/rtl8812AU $ sudo apt-get install dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dkms is already the newest version (2.6.1-4).
0 upgraded, 0 newly installed, 0 to remove and 23 not upgraded.
pi@raspberrypi:~/rtl8812AU $ sudo bash -l
root@raspberrypi:/home/pi/rtl8812AU# 
root@raspberrypi:/home/pi/rtl8812AU# 
root@raspberrypi:/home/pi/rtl8812AU# 
root@raspberrypi:/home/pi/rtl8812AU# cd
root@raspberrypi:~# 
root@raspberrypi:~# 
root@raspberrypi:~# cd ~pi/rtl8812AU/
root@raspberrypi:/home/pi/rtl8812AU# DRV_NAME=rtl8812AU
root@raspberrypi:/home/pi/rtl8812AU# DRV_VERSION=4.3.20
root@raspberrypi:/home/pi/rtl8812AU# mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
root@raspberrypi:/home/pi/rtl8812AU# git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
root@raspberrypi:/home/pi/rtl8812AU# dkms add -m ${DRV_NAME} -v ${DRV_VERSION}

Creating symlink /var/lib/dkms/rtl8812AU/4.3.20/source ->
                 /usr/src/rtl8812AU-4.3.20

DKMS: add completed.
root@raspberrypi:/home/pi/rtl8812AU# dkms build -m ${DRV_NAME} -v ${DRV_VERSION}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.19.50-v7l+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.50-v7l+ (armv7l)
Consult /var/lib/dkms/rtl8812AU/4.3.20/build/make.log for more information.
Inline here:
DKMS make.log for rtl8812AU-4.3.20 for kernel 4.19.50-v7l+ (armv7l)
Wed 10 Jul 2019 10:55:47 AM MST
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.19.50-v7l+/build M=/var/lib/dkms/rtl8812AU/4.3.20/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.50-v7l+'
Makefile:612: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.19.50-v7l+'
make: *** [Makefile:1687: modules] Error 2
root@raspberrypi:/home/pi/rtl8812AU# grep -i arm Makefile | head
CONFIG_PLATFORM_ARM_RPI = y
CONFIG_PLATFORM_ARM64 = n
```

Juansbulbisful, 2020-09-20:
_juan

Juansbulbisful, 2020-09-20:
_(*:_*)