diederikdehaas/rtl8812AU #122: New kernel incompatibility

shawnhcorey, 2020-01-17:
I recently update my kernel to 5.3.0-26-generic (#28~18.04.1-Ubuntu) from Linux Mint. My WiFi doesn't always start at boot and sometimes I can't get it to start at all.

Is there any diagnostic files I should look at?

I am currently running 5.0.0.37-generic (#40~18.04.1-Ubuntu) from Linux Mint. rtl8812AU version is 4.3.20

shawnhcorey, 2020-01-17:
I have also found a new version of libdrm, 2.4.99-1ubuntu1~18.04.1 which was installed with the new kernel.

Currently I am running 2.4.97-1ubuntu1~18.04.1

shawnhcorey, 2020-01-30:
Well, I tried again and version 5.3.x does not work. I rebuild the driver but when i tried `sudo modprobe -v 8812au` it just said `insmod /lib/modules/5.3.0-28-generic/updates/dkms/8812au.ko` and froze.

So I tried this:
`$ sudo modprobe -v -f 8812au
insmod /lib/modules/5.3.0-28-generic/updates/dkms/8812au.ko 
modprobe: ERROR: could not insert '8812au': Exec format error`

Seems there something different between Linux kernel versions 5.3.x and previous ones.

SecondeJK, 2020-02-06:
I now have this.
I really, really, really wish I hadn't bought a realtek chipset device, despite the box saying "linux support" on it.

3vi1, 2020-06-23:
I can confirm the same results (kernel 5.4.0-31, groovy gorilla alpha).  I've tried with 4.3.20, 4.3.22-beta (process crashes), 5.2.20, and 5.6.4.

My wifi light simply never comes on at all using these versions, which is the same as what I see using Ubuntu's 4.3.8 dkms package, as well as the 4.3.14 that is built into Ubuntu 22.10 at the moment.

If I revert the driver all the way back to 4.2.3 - who's source I found in another git, it works 'fine'.  I say 'fine', because it works great for a day or two, then specific apps begin to hang on networking operations related to security (logins and such).  If I unload/reload it, it's fine for another day.

SecondeJK, 2020-06-26:
Personally, gave up, threw chipset in bin and got something that's plug and play with Linux. Problem solved

3vi1, 2020-06-28:
You're probably coming out ahead.  I compiled the 5.6.4.2 drivers from the aircrackng repo.  They work, but have the same stability issue where certain apps start failing if I don't uload/reload them every couple of days.